helgeerbe/picframe #259: New Buster issue: "Can't roll back Pillow; was not uninstalled"

sapnho, 2022-04-08:
I got two comments this last week that apparently there has been an OS Buster update on April 4 which messes up things:

```
...and I'm stuck at sudo pip3 install picframe
I get:
Can't roll back Pillow; was not uninstalled
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-w5vo6_9a/Pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-ng8l3s4c/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-w5vo6_9a/Pillow/

And if I then reboot and do picframe -i .
I get this:
-bash: picframe: command not found

```
This is with OS Buster, latest release on a Pi3. Did anything change with Pillow?

helgeerbe, 2022-04-08:
Try this:

apt-get install libjpeg-dev zlib1g-dev
and then

pip3 install Pillow

Does this work?

sapnho, 2022-04-08:
I had to add sudo for the first command.

> pi@pi3test:~ $ apt-get install libjpeg-dev zlib1g-dev
E: Could not open lock file /var/lib/dpkg/lock-frontend - open (13: Permission denied)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), are you root?
pi@pi3test:~ $ sudo apt-get install libjpeg-dev zlib1g-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
zlib1g-dev is already the newest version (1:1.2.11.dfsg-1+deb10u1).
zlib1g-dev set to manually installed.
The following NEW packages will be installed:
  libjpeg-dev libjpeg62-turbo-dev
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 239 kB of archives.
After this operation, 546 kB of additional disk space will be used.
Get:1 http://mirror.de.leaseweb.net/raspbian/raspbian buster/main armhf libjpeg62-turbo-dev armhf 1:1.5.2-2+deb10u1 [181 kB]
Get:2 http://mirror.de.leaseweb.net/raspbian/raspbian buster/main armhf libjpeg-dev all 1:1.5.2-2+deb10u1 [57.7 kB]
Fetched 239 kB in 1s (289 kB/s)   
Selecting previously unselected package libjpeg62-turbo-dev:armhf.
(Reading database ... 99929 files and directories currently installed.)
Preparing to unpack .../libjpeg62-turbo-dev_1%3a1.5.2-2+deb10u1_armhf.deb ...
Unpacking libjpeg62-turbo-dev:armhf (1:1.5.2-2+deb10u1) ...
Selecting previously unselected package libjpeg-dev.
Preparing to unpack .../libjpeg-dev_1%3a1.5.2-2+deb10u1_all.deb ...
Unpacking libjpeg-dev (1:1.5.2-2+deb10u1) ...
Setting up libjpeg62-turbo-dev:armhf (1:1.5.2-2+deb10u1) ...
Setting up libjpeg-dev (1:1.5.2-2+deb10u1) ...
pi@pi3test:~ $ pip3 install Pillow
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: Pillow in /usr/lib/python3/dist-packages (5.4.1)

sapnho, 2022-04-08:
Can we update the `sudo pip3 install picframe` script to account for this?

helgeerbe, 2022-04-08:
Does it work now?

paddywwoof, 2022-04-08:
Or change the line in `setup.py` from `Pillow>=9.0.0` to `Pillow==9.0.1`. Until the pillow devs fix it! Using dependencies >= might not be as safe as using == then waiting for one of use to test with a later version!

That was basically the error I got the other day (2nd of Apr!!) which I referred to as [a complaint about a missing jpeg header for pillow](https://github.com/helgeerbe/picframe/discussions/257#discussioncomment-2491774) As I said, I assumed it was because I had missed out or mistyped one of @thomasedoff console inputs.

To patch over it I just used `apt install python3-pillow` (also more correctly known as `python3-pil` I think the version in bullseye is 8.1.2). Presumably we can't include a command to apt install a header file in the `setup.py` file but we could add it to the manual installation requirements, or using @thomasedoff instructions add it to [here](https://gist.github.com/thomasedoff/3973846101da1e7496524be075f1c02f#install-necessary-software-and-dependencies).

sapnho, 2022-04-08:
It would be nice to include it in the `setup.py` file so that installation is as easy as before. Could we do that or would it have other consequences?

paddywwoof, 2022-04-08:
Looks like there is this https://github.com/python-pillow/Pillow/issues/5797#issuecomment-1093054519 issue with v9.1.0 so we might have to force setup to use v9.0.1 until we clear up both issues.

thomasedoff, 2022-04-08:
Actually, looking closer at piwheels.org t[here does not seem to be a precompiled package of Pillow 9.1.0](https://www.piwheels.org/simple/pillow/) for any platform.

So if you're trying to install 9.1.0 at this point in time, you will need the headers, regardless of the architecture.

paddywwoof, 2022-04-08:
@thomasedoff you are probably right about the piwheels being offline, and also about the aarch64 support, but I think I got the same multi-page error text about missing headers when I tried to setup bullseye with piOS32, which I did before testing the 64 bit, and I'm sure the headers didn't need to be installed previously. Of course they might have been a requirement of some other package that is no longer there in the latest piOS lite setup!

helgeerbe, 2022-04-11:
Issue could be solved now. I just checked pinwheels.org and there are precompiled packages for Pillow 9.1.0 now available.

@paddywwoof I installed `Pillow 9.1.0.` yesterday manually and I can't confirm that our old memory issue popped up again. Seems to be a different issue. My system logs look fine.

helgeerbe, 2022-04-11:
So we can close this issue now?

sapnho, 2022-04-11:
Let me do a fresh install later today so that I can confirm that it works.