helgeerbe/picframe #260: KeyError: 'peripherals' error when running.

MDanihy, 2022-04-08:
INFO:start.py:starting ['/usr/local/bin/picframe', '/home/pi/picframe_data/config/configuration2.yaml']
INFO:model.Model:Open config file: /home/pi/picframe_data/config/configuration2.yaml:
Traceback (most recent call last):
  File "/usr/local/bin/picframe", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/dist-packages/picframe/start.py", line 130, in main
    m = model.Model(args.configfile)
  File "/usr/local/lib/python3.7/dist-packages/picframe/model.py", line 153, in __init__
    self.__config[section] = {**DEFAULT_CONFIG[section], **conf[section]}
KeyError: 'peripherals'

jgodfrey, 2022-04-08:
Hmmm... That seems to indicate that your config file (`configuration2.yaml`) doesn't contain a `peripherals` section, which it should. Is that section missing (or misspelled or mis-formatted)? You can look at the `config/configuration_example.yaml` file for reference.

MDanihy, 2022-04-08:
I have two RPIs setup, one with a touch-screen, the one with the error has a monitor connected to the HDMI port.  Neither one has a peripherals section in the configuration file.

MDanihy, 2022-04-08:
Found it in the original config, not sure why it is missing.

Will take care of and see if that helps.

Thanks,

closed 2022-04-11 by helgeerbe