helgeerbe/picframe #261: Picframe stalls due to missing images

simontrimmer, 2022-04-15:
I'd been tinkering over the last few months with various combinations of Pi hardware and base OSes with mixed success, in an attempt to get a debuggable system I created a new instance based off Thomasedoff's recent gist (https://gist.github.com/thomasedoff/3973846101da1e7496524be075f1c02f) and Wolfgang's HEIC notes. Testing that I found that it was proving reliable until it stalled this evening, digging a little revealed this exception

/var/log/daemon.log:
Apr 15 23:25:56 picframe xinit[21262]: Traceback (most recent call last):
Apr 15 23:25:56 picframe xinit[21262]:   File "/home/pi/.local/bin/picframe", line 8, in <module>
Apr 15 23:25:56 picframe xinit[21262]:     sys.exit(main())
Apr 15 23:25:56 picframe xinit[21262]:   File "/home/pi/.local/lib/python3.9/site-packages/picframe/start.py", line 154, in main
Apr 15 23:25:56 picframe xinit[21262]:     c.loop()
Apr 15 23:25:56 picframe xinit[21262]:   File "/home/pi/.local/lib/python3.9/site-packages/picframe/controller.py", line 301, in loop
Apr 15 23:25:56 picframe xinit[21262]:     pics = self.__model.get_next_file()
Apr 15 23:25:56 picframe xinit[21262]:   File "/home/pi/.local/lib/python3.9/site-packages/picframe/model.py", line 335, in get_next_file
Apr 15 23:25:56 picframe xinit[21262]:     pic_row = self.__image_cache.get_file_info(file_ids[0])
Apr 15 23:25:56 picframe xinit[21262]:   File "/home/pi/.local/lib/python3.9/site-packages/picframe/image_cache.py", line 158, in get_file_info
Apr 15 23:25:56 picframe xinit[21262]:     if row is not None and row['last_modified']  != os.path.getmtime(row['fname']):
Apr 15 23:25:56 picframe xinit[21262]:   File "/usr/lib/python3.9/genericpath.py", line 55, in getmtime
Apr 15 23:25:56 picframe xinit[21262]:     return os.stat(filename).st_mtime
Apr 15 23:25:56 picframe xinit[21262]: FileNotFoundError: [Errno 2] No such file or directory: '/photo/picframe_photos/simons_cameraphone_photos/20151118_124658.jpg'

The service is still running, though I think it's compromised - It is still showing the last image with the frozen clock and I can switch the screen off/on using the picframe_display MQTT mechanism but the prev/pause/next/shuffle mechanisms don't work. My logfile is set to WARNING and doesn't contain any details of the error so I think this is likely the root cause of the issues I'd been having in my past experiments.

That directory is a network share and contains a few thousand images from various devices so rebuilding the database takes a while! Adding, moving and deleting images is a common maintenance activity so finding missing files is likely to be relatively likely in my setup so working out some graceful error handling sounds like a good idea.

Python isn't my usual language but the os.path.getmtime() isn't in a try or perhaps os.path.exists() wouldn't throw an exception and would make for more manageable code?

Anyway... Thoughts?

pi@picframe:~ $ picframe --version
INFO:start.py:starting ['/home/pi/.local/bin/picframe', '--version']
picframe version:  2022.03.24

Checking required packages......
PIL :  9.1.0
exifread :  2.3.2
pi3d :  2.49
yaml :  6.0
paho.mqtt :  1.6.1
iptcinfo3 :  2.1.4
numpy :  1.22.3
ninepatch : installed, but no version info

Checking optional packages......
pyheif :  0.7.0

simontrimmer, 2022-04-15:
Should also say that this system started as the extremely recent (latest) 32bit 2022-04-04-raspios-bullseye-armhf-lite.img

paddywwoof, 2022-04-16:
Not looked at the code but I thought all the os calls that might fail with missing file were inside try blocks, but it looks like that one isn't. It might be better to check for file existence although using try seems to be regarded as 'pythonic'

simontrimmer, 2022-04-16:
Just having a quick scan through the code again today, it does look like there is a check for the file existing in model.py but by that that the cache will already have attempted the getmtime() and raised OSError. 

It might not be pythonic but I'd be tempted just to add a specific check into get_file_info() before the getmtime() something like:

...
        row = self.__db.execute(sql).fetchone()
>        if row is not None and not os.path.isfile(row['fname']):
>           self.__logger.debug('Cache miss: File %s missing on disk', row['fname'])
>           return None
        if row is not None and row['last_modified']  != os.path.getmtime(row['fname']):
...

As it looks like the code has the appropriate None checks to do the right thing, I think there is a separate maintenance operation that hunts for missing files and that should clean up the db entry

helgeerbe, 2022-04-18:
Thanks for finding that. Indeed I missed a check for file existence. I will prepare a new release.

````python
try:
    if row is not None and row['last_modified']  != os.path.getmtime(row['fname']):
        self.__logger.debug('Cache miss: File %s changed on disk', row['fname'])
        self.__insert_file(row['fname'], file_id)
        row = self.__db.execute(sql).fetchone() # description inserted in table
except OSError:
    self.__logger.warning("Image '%s' does not exists or is inaccessible" %row['fname'])
````

helgeerbe, 2022-04-18:
I just commited a fix to dev.

simontrimmer, 2022-04-19:
Cool thanks for that, will do some image maintenance when it's released :)

closed 2022-04-19 by helgeerbe

simontrimmer, 2022-04-19:
Argh wrong tab.

Just checked and it seems to be working fine (I get the expected warning message in the patch)

Is the prune of missing files triggered once per startup or is it a regular background or triggered operation?

jgodfrey, 2022-04-19:
I don't think missing files are ever purged _automatically_. That's intentional, as we didn't want to clear the cached data for a drive or mount that was temporarily offline. Instead, they're marked as missing and can be cleaned up manually by triggering the `purge_files()` command (via the MQTT or HTML interfaces)

simontrimmer, 2022-04-19:
That makes sense - I'd not seen the purge_files topic, I think I'll have to have a read over all the files now

Cheers!