helgeerbe/picframe #262: Request-new MQTT function- update caption

idcline, 2022-04-16:
I’ve been using the caption field IPTC tag to capture what I want displayed on my photos.  I’ve used several editors to add/edit the tags on my photos but I haven’t gone through my entire collection and often I see a caption that I want to change.  

I noticed that iptcinfo3 is in the dependencies for pic frame and that it has an `Add/change an attribute info['caption/abstract'] = 'Witty caption here'` function. Other IPTC/EXIF services exist too.  

It would be nice if there was an MQTT function that acted on the current image. Input is a string caption.  The code would then edit the current image using an IPTC service.  

Operationally it would make sense for a user to pause the show before issuing this command to avoid missing the target. 

Is this reasonable?

jgodfrey, 2022-04-16:
Hmmm... While I understand the reason for the request, it doesn't feel like a good fit for a project that's designed to *view* images. Performing any kind of edit on (what could be) a User's original images is a slippery slope I think, and greatly increases the risk for the project.

Related, I'd think it'd be a poor environment to do caption editing in anyway.

My vote would be to do those edits externally, in software designed to do just that. But, again, I understand the reasoning - you're looking at the image in the slideshow and notice something you'd like to change "right now".

I'm curious what others think...

idcline, 2022-04-16:
Good point about not wanting to mix up a display system with a photo editor. 

Perhaps a middle ground would be to create a “comment log” that would allow a MQTT message to make a text entry in the log, along with time/date and current photo filename and path.  That could extend to notes like “go back to original and adjust contrast and recrop” photo.  User can later work off the comments/notes.   

The log file could be used by an independent program or script to make caption changes.