helgeerbe/picframe #268: Sighting on dev - sqlite3.OperationalError: cannot start a transaction within a transaction

simontrimmer, 2022-05-18:
This is mostly capturing the information so I can look into it myself another day - I had a quick look over but the root cause isn't obvious to me.

I'd noticed my main picframe had hung earlier this evening, there are no entries in the log (set at warning) but looking in the daemon.log I found

`May 18 18:27:09 picframe xinit[15018]: display_power=1
May 18 18:31:18 picframe xinit[10334]: Traceback (most recent call last):
May 18 18:31:18 picframe xinit[10334]:   File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
May 18 18:31:18 picframe xinit[10334]:     return _run_code(code, main_globals, None,
May 18 18:31:18 picframe xinit[10334]:   File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
May 18 18:31:18 picframe xinit[10334]:     exec(code, run_globals)
May 18 18:31:18 picframe xinit[10334]:   File "/home/pi/git/picframe/picframe/start.py", line 169, in <module>
May 18 18:31:18 picframe xinit[10334]:     main()
May 18 18:31:18 picframe xinit[10334]:   File "/home/pi/git/picframe/picframe/start.py", line 160, in main
May 18 18:31:18 picframe xinit[10334]:     c.loop()
May 18 18:31:18 picframe xinit[10334]:   File "/home/pi/git/picframe/picframe/controller.py", line 304, in loop
May 18 18:31:18 picframe xinit[10334]:     pics = self.__model.get_next_file()
May 18 18:31:18 picframe xinit[10334]:   File "/home/pi/git/picframe/picframe/model.py", line 338, in get_next_file
May 18 18:31:18 picframe xinit[10334]:     pic_row = self.__image_cache.get_file_info(file_ids[0])
May 18 18:31:18 picframe xinit[10334]:   File "/home/pi/git/picframe/picframe/image_cache.py", line 166, in get_file_info
May 18 18:31:18 picframe xinit[10334]:     if self.__get_geo_location(row['latitude'], row['longitude']):
May 18 18:31:18 picframe xinit[10334]:   File "/home/pi/git/picframe/picframe/image_cache.py", line 199, in __get_geo_location
May 18 18:31:18 picframe xinit[10334]:     self.__db.execute(sql, (lat, lon, location))
May 18 18:31:18 picframe xinit[10334]: sqlite3.OperationalError: cannot start a transaction within a transaction`

This is on the tip of dev with my opacity and suppression PRs pulled in on top, which looking at the stack seem unrelated. 

commit 10a696120147c8d91072a0fc882d64cbb520fe68 (HEAD -> 17may_dev_opacity_suppress2)
Author: Simon Trimmer <github@urbanmyth.org>
Date:   Sat Apr 23 21:56:54 2022 +0100

    Introduce geo_suppress_list to omit strings from location text
    
    Allows you to remove text from the location text, for instance when
    most of your photos are from one location or the reverse geo results
    for a location is extremely verbose.

commit 61a5f0ace95942b6d44b55e4b59f2e77083cf74c (origin/basic_text_opacity)
Author: Simon Trimmer <github@urbanmyth.org>
Date:   Fri Apr 22 17:53:46 2022 +0100

    Add opacity option to text and clock

commit 77760b130f0693aa7e2df97d0fef9f9bc4df94dd (origin/dev)
Merge: 96d81c2 f481b50
Author: helgeerbe <59169507+helgeerbe@users.noreply.github.com>
Date:   Sun May 15 15:07:22 2022 +0200

    Merge pull request #267 from helgeerbe/split_mqtt_notify
    
    mqtt notify improvements

helgeerbe, 2022-05-19:
I have never seen this error before. 

Looks like different sql statements are executed in parallel and are not wrapped in something like db.serialize(). My guess is, that we are not thread safe somewhere.

@jgodfrey do you have an idea?

simontrimmer, 2022-05-19:
@helgeerbe I've not seen it before either and was thinking along the same lines, I knew there is some element of multi-threading as the mqtt interface survives after seeing a trace such as this and the screen can be switched on/off, but I've not got an architectural view to know what things in the image cache could be running in parallel. 

That install had been started the evening before to test out the dev with the PRs, so it started with an empty database and ~2000 images some of which have geolocation.  As the picture frame is still novelty in the household we have the time between picture changes relatively short at 45 seconds, so there is more activity than one with the default 200 seconds.

greensum, 2022-12-05:
I also got a sqlite3.OperationalError a couple of weeks back.  I'm using 2022.06.03.  Don't know if it is the same bug.

```
ERROR:geo_reverse.GeoReverse:lat=14.782600, lon=-71.887700 -> 'features'
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.7/dist-packages/picframe/image_cache.py", line 55, in __loop
    self.update_cache()
  File "/usr/local/lib/python3.7/dist-packages/picframe/image_cache.py", line 84, in update_cache
    self.__update_file_stats()
  File "/usr/local/lib/python3.7/dist-packages/picframe/image_cache.py", line 190, in __update_file_stats
    self.__db.execute(sql, (timestamp, file_id))
sqlite3.OperationalError: cannot start a transaction within a transaction
```

simontrimmer, 2022-12-05:
Related at least!

I’d not been able to reproduce this since the initial sighting, though I’ve not dropped several thousand images on the picframe since. Did you recently add a number of images?

greensum, 2022-12-05:
This hasn't happened again even though I have been adding images from time to time, like every other day or so.

helgeerbe, 2022-12-06:
Hi, Thanks for reporting. It seems to me, that we have some timing issues here. I will check the code and see, if we have to serialize the DB writes. Not sure, if transactions also include db reads.

jgodfrey, 2022-12-06:
This seems likely to be the same underlying issue I noted in my above comment from May 19. That is, the geo location info is written to the DB "on the fly" - outside of the image caching thread. With some unlucky timing, I think this update is likely colliding with normal db updates from the image caching thread.

As mentioned above, there was never any intention of writing to the DB outside of the image caching thread, because of these kinds of dangers. So, I think the fix here will be to eliminate this particular [db insertion](https://github.com/helgeerbe/picframe/blob/3930e4e6bdc7558dc4f004be13583b18478864b7/picframe/image_cache.py#L199).

I ran into a similar issue when attempting to update the image statistics a while back. Originally that update was running on the main thread too and occasionally caused similar issues. To fix that, I ended up caching the image stat updates to a temporary collection in the main thread and then writing them to the db in the caching thread. That seems to have fixed that issue. While there are probably better ways to fix this (architecturally), something similar might be the simplest way forward with the currently offending insertion.

That is, instead of inserting it directly into the db, cache it in a temporary collection and let the image caching thread do the insertion in its own time.

Thoughts?

Assuming that seems like a reasonable way forward, I can try to look at it sometime soon - unless someone beats me to it (no promises on timing though).

helgeerbe, 2022-12-06:
Hi @jgodfrey I googled a little bit. It's a multithread issue of corse. But I couldn't find the one and only solution. This error could happen if you have more than one thread. It doesn't matter if it is a reading or writing command.

It depends how sqlite is compiled and configured. Compiled thread safe, auto commit configured, .....

My ideas:
- serialize  with mutex. Should be easy to acquire for each db access a lock.
- Have one db connection for reading and one for writing (serialize only the writer thread). 
- catch sqlite3.OperationalError and try agin after a short sleep

I would vote for the serialize with mutex. Except you know how to configure sqlite, that it can handle this on its own.

jgodfrey, 2022-12-06:
@paddywwoof originally added the threading code to this class, so maybe he'll have some input here.

In the meantime though, see (paddy's) comment [here](https://github.com/helgeerbe/picframe/blob/3930e4e6bdc7558dc4f004be13583b18478864b7/picframe/image_cache.py#L299).

So, I'm confident the intent was to only ever WRITE to the db in the caching thread - however that intent is being violated by the direct update that's being done in by the geo location code. I do think that info could just be cached similar to how the [file stats](https://github.com/helgeerbe/picframe/blob/3930e4e6bdc7558dc4f004be13583b18478864b7/picframe/image_cache.py#L30) are currently being handled (and then written in the caching thread), but it's not the most elegant solution.

helgeerbe, 2022-12-07:
I created a test branch [db-threading](https://github.com/helgeerbe/picframe/compare/main...db-threading)

I removed the cached file stats handling and added just simple mutex locks before the db write calls. On my Mac it takes around 1 ms to update file stats or location on sqlite. This seems to be fast enough that we don't need a lazy write in the caching thread.

I tested it with 517 images without an issue. Just one image crashed Pillow while it tried to extract xmp tags. But this is a different issue.

paddywwoof, 2022-12-07:
Thread locking round all the db action seems a very sensible approach, especially if we no longer need the stat cache system. I suppose on a slower RPi, when putting on an initial bulk amount of images there might be more delay but I'm not sure it will impact the slide transitions.

helgeerbe, 2022-12-07:
I only put the locking around update, insert, commit calls (write), not select (read). Seems to work fine. I just put it on my living room frame (pi 3B+), and deleted the database.

Is see that updating file stats take around 1ms (one measurement shows 17ms). Sometimes there is a wait for 1ms after aquire the lock.

Fore geo updates it varies between 14-19 ms. Up to now, I don't have seen any wait for lock there.

Up to now, everything looks fine for me. Feel free to test the branch.

helgeerbe, 2022-12-13:
Should be fixed with release 2022.12.13

closed 2022-12-13 by helgeerbe