helgeerbe/picframe #271: Home Assistant Buttons for "Today", "Last Week", or "Last Month"

sapnho, 2022-06-12:
@helgeerbe Your integration of Pi3D PictureFrame into Home Assistant last year was another milestone for this project. I was wondering about a use case that I actually have quite often and how to make it happen.

I often want to show images only of "Today", "Last Week", or "Last Month". So in Home Assistant, I manually set the FROM date. Is it possible to have buttons that send a FROM DATE MQTT message which is calculated automatically? How would that work?

Thanks!

helgeerbe, 2022-06-13:
Hi @sapnho,

that should doable. 

One could create 3 buttons triggering an automation. Based on the entity-Id of the buttons an automation could calculate the mqtt message.

To calculate today
````yaml
{{ (now().strftime("%Y/%m/%d")) }}
````

To calculate last week one could use timetelda()
1. Get today
2. get weekday (%w) weekday as a decimal number [0(Sunday),6].
3. get first day of last week->today - timedelta(day=weekday, week=1)
4. get last day of week-> first day of week + 6 days

To calculate last month:
1. Get today
2. Get last day of last week = timedelta(day=day of the month +1)
3. Get first day of last month = set day of month to 1

helgeerbe, 2022-06-13:
Just tested the calculation:

````YAML
Today
{% set today = now() %}
{{ (today.strftime("%Y/%m/%d")) }}

Last week:
{% set first_day_of_week = (today.strftime("%w")) | int %}
  day of week: {{ first_day_of_week }}
  first day of week: {{ as_timestamp(today - timedelta(days=first_day_of_week, weeks=1)) | timestamp_custom("%Y/%m/%d")}}
  last day of week: {{ as_timestamp(today - timedelta(days=first_day_of_week - 6, weeks=1)) | timestamp_custom("%Y/%m/%d")}}

Last month:
{% set day_of_month = (today.strftime("%d")) | int %}
  day of month: {{ day_of_month }}
  last day of last month: {{ as_timestamp(today - timedelta(days=day_of_month)) | timestamp_custom("%Y/%m/%d")}}
  first day of last month: {{ as_timestamp(today - timedelta(days=day_of_month)) | timestamp_custom("%Y/%m/01")}}
````

output:

````txt
Today

2022/06/13

Last week:

  day of week: 1
  first day of week: 2022/06/05
  last day of week: 2022/06/11

Last month:

  day of month: 13
  last day of last month: 2022/05/31
  first day of last month: 2022/05/01
````

helgeerbe, 2022-06-13:
Found a bug. The date filters are not updated correctly. But setting the dates works. I have to check this.

sapnho, 2022-06-13:
Thanks, @helgeerbe! Will also give it a try and let us know, once you have found the bug.

sapnho, 2022-06-13:
When I paste the automation I get the error "Message malformed: template value should be a string for dictionary value @ data['action'][0]['else'][0]['else'][1]['data']".
Maybe that helps to trace the bug?

helgeerbe, 2022-06-13:
I had to edit the home assistant blueprint code. The easiest way to update is to delete the automation and to delete the blueprint. Then import the blueprint again and create the automation from the blueprint.

Since release 2021.10 Home Assistant requires default values in some time manipulation to create date objects from string. Without default values an exception is thrown.