helgeerbe/picframe #274: The "bump" transition is a little confusing

djkoelkast, 2022-06-20:
I was hoping for a transition that "bumps" the picture from left to right or top to bottom, but bump is something totally different.
Is there an option to use different transition, like the one I'm looking for?

paddywwoof, 2022-06-20:
The transition effect is controlled by the 'shader' see [here](https://github.com/helgeerbe/picframe/blob/main/picframe/data/shaders/blend_new.fs), and you can use whatever suits your needs.

*However* it's not trivial by any means as the info gets passed from the python picframe code to the GPU shader language via various arrays and vectors with opaque names such as `unif[18][1]`. The shader also does scaling to make the images fit or overlap the screen according to your settings.

When you say bump do you imagine that one image would slide into view while the existing one would slide out?

djkoelkast, 2022-06-20:
That's correct, that's the type of transition I'm looking for.