helgeerbe/picframe #276: Some MQTT values not appearing?

brynyneuadd, 2022-07-15:
**Issue**
I've followed the instructions on https://www.thedigitalpictureframe.com/integrate-raspberry-pi-picture-frame-home-assistant/ to integrate into HA, and it appears from Home Assistant that it's 'sort of' working. I can see the values coming through, but they're null. Also it's missing some of the elements (eg picframe_time_delay doesn't appear). 

**Couple of possible causes:**
- I installed this before on home assistant and then had to rebuild my RPI4 install, can't think why this'd be a problem, but just in case. I wonder if some of these sensors/entities could be actually just from previous install and it's not actually working any more?
- As I plan to have more than one frame, i changed the name in the configuration.yaml to HallPhotoFrame (vs picframe)

**Screenshots**

[image]
[image]

**What are your OS, Python and picframe versions(please complete the following information):?**

- Hardware: RPi 4 (two separate ones for both home assistant and photo frame)
- OS: Latest versions of Raspian, HA, HAOS etc
- picframe (run `picframe -v` if possible): 

```
viewer:
  blur_amount: 12                         # default=12,  larger values than 12 will increase processing load quite a bit
  blur_zoom: 1.0                          # default=1.0, must be >= 1.0 which expands the backgorund to just fill the space around the image
  blur_edges: False                       # default=False, use blurred version of image to fill edges - will override FIT = False
  edge_alpha: 0.5                         # default=0.5, background colour at edge. 1.0 would show reflection of image
  fps: 20.0                               # default=20.0
  background: [0.2, 0.2, 0.3, 1.0]        # default=[0.2, 0.2, 0.3, 1.0], RGBA to fill edges when fitting
  blend_type: "blend"                     # default="blend", choices={"blend", "burn", "bump"}, type of blend the shader can do
  font_file: "/home/admin/picframe_data/data/fonts/NotoSans-Regular.ttf"
  shader: "/home/admin/picframe_data/data/shaders/blend_new"
  show_text_fm: "%b %d, %Y"               # default "%b %d, %Y", format to show date over the image
  show_text_tm: 20.0                      # default=20.0, time to show text over image with file name
  show_text_sz: 40                        # default=40, text character size
  show_text: ""  # default="title caption name date folder location", show text, include combination of words: title, caption name, date, location, folder
  text_justify: "L"                       # text justification L, C or R
  text_bkg_hgt: 0.25                      # default=0.25 (0.0-1.0), percentage of screen height for text background texture
  text_opacity: 1.0                       # default=1.0 (0.0-1.0), alpha value of text overlay
  fit: False                              # default=False, True => scale image so all visible and leave 'gaps'
                                          #                False => crop image so no 'gaps'
  kenburns: False                         # default=False, will set fit->False and blur_edges->False
  display_x: 0                            # offset from left of screen (can be negative)
  display_y: 0                            # offset from top of screen (can be negative)
  display_w: null                         # width of display surface (null->None will use max returned by hardware)
  display_h: null                         # height of display surface
  use_glx: False                          # default=False. Set to True on linux with xserver running

  mat_images: False                        # default=0.01, True, automatically mat all images. False, don't automatically mat any images. Real value, auto-mat all images with aspect ratio difference > than value
  mat_type: float_polaroid                          # default=null, A string containing the mat types to choose from when matting images. It can consist of any or
                                          # all of 'float float_polaroid float_color_wrap single_bevel double_bevel double_flat' (null or '' will use all mat types)
  outer_mat_color: [255, 255, 255]                   # default=null, Color of the outer mat as an RGB list. null will auto-select a reasonable color based on the image.
  inner_mat_color: null                   # default=null, Color of the inner mat as an RGB list. null will auto-select a reasonable color based on the image.
  outer_mat_border: 75                    # default=75, Minimum outer mat border in pixels
  inner_mat_border: 40                    # default=40, Minimum inner mat border in pixels (for styles that use it)
  outer_mat_use_texture: False             # default=True, True uses a texture for the outer mat. False creates a solid-color outer mat.
  inner_mat_use_texture: False            # default=False, True uses a texture for the inner mat. False creates a solid-color inner mat.
  mat_resource_folder: "/home/admin/picframe_data/data/mat" # Folder containing mat image files

  show_clock: False                       # default=False, True shows clock overlay. False does not show clock overlay
  clock_justify: "R"                      # default="R", clock justification L, C, or R
  clock_text_sz: 120                      # default=120, clock character size
  clock_format: "%I:%M"                   # default="%I:%M", strftime format for clock string
  clock_opacity: 1.0                      # default=1.0 (0.0-1.0), alpha value of clock overlay

  menu_text_sz: 40                        # default=40, menu character size
  menu_autohide_tm: 10.0                  # default=10.0, time in seconds to show menu before auto hiding (0 disables auto hiding)
  geo_suppress_list: []                   # default=None, substrings to remove from the location text

model:
  pic_dir: "/home/admin/Pictures"                   # default="/home/admin/Pictures", root folder for images
  deleted_pictures: "/home/admin/DeletedPictures"   # move deleted pictures here
  follow_links: False                     # default=False, By default, picframe will not walk down into symbolic links that resolve to directories. Set follow_links to True to visit directories pointed to by symlinks, on systems that support them.
[... 59 lines cut ...]
peripherals:
  input_type: keyboard                        # default=null, valid options: {null, "keyboard", "touch", "mouse"}
  buttons:
    pause:                                # pause/unpause the show
      enable: True                        # default=True
      label: "Pause"                      # default="Pause"
      shortcut: " "                       # default=" "
    display_off:                          # turn off the display (when off, any input from selected peripheral will turn it back on)
      enable: True                        # default=True
      label: "Display off"                # default="Display off"
      shortcut: "o"                       # default="o"
    location:                             # shows or hides location information
      enable: False                       # default=False
      label: "Location"                   # default="Location"
      shortcut: "l"                       # default="l"
    exit:                                 # exit PictureFrame
      enable: True                       # default=False
      label: "Exit"                       # default="Exit"
      shortcut: "e"                       # default="e"
    power_down:                           # power down the device, uses sudo
      enable: True                       # default=False
      label: "Power down"                 # default="Power down"
      shortcut: "p"                       # default="p"

```

helgeerbe, 2022-07-15:
Hi @brynyneuadd,

first of all I would recommend do upgrade picframe. Ttere are some breaking changes in the home_assistant integration (https://github.com/helgeerbe/picframe/wiki/Changelog).

Then you should delete the old picframe setup in home assistant. In the device info in home assistant you see three dots (...). Delete the old setup. Also delete the old automation and the old blueprint.

Follow the instructions here (https://github.com/helgeerbe/picframe/wiki/Homeassistant)

Have in mind that you have to change the deviceId in the blueprint setup to "hallartframe" as you configured it in your configuration.yaml.  And for every picframe you have to create its own helpers. In the blueprint setup set the correct entity_ids.

Hope that helps.

brynyneuadd, 2022-07-15:
Hi @helgeerbe,

Thanks for your reply!

Ok - have checked and picframe already latest version, so should be ok:
```
picframe version:  2022.06.03

Checking required packages......
PIL :  9.1.1
exifread :  3.0.0
pi3d :  2.49
yaml :  6.0
paho.mqtt :  1.6.1
iptcinfo3 :  2.1.4
numpy :  1.16.2
ninepatch : installed, but no version info
```

I've deleted everything as you describe, however nothing's now coming through on the MQTT broker in HA as the new device (have other things that are working fine, but nothing from picframe)

sudo systemctl status mosquitto gives

```
 mosquitto.service - Mosquitto MQTT v3.1/v3.1.1 Broker
   Loaded: loaded (/lib/systemd/system/mosquitto.service; enabled; vendor preset
   Active: active (running) since Fri 2022-07-15 09:24:36 BST; 7min ago
     Docs: man:mosquitto.conf(5)
           man:mosquitto(8)
 Main PID: 461 (mosquitto)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/mosquitto.service
           └─461 /usr/sbin/mosquitto -c /etc/mosquitto/mosquitto.conf

Jul 15 09:24:35 HallFrame systemd[1]: Starting Mosquitto MQTT v3.1/v3.1.1 Broker
Jul 15 09:24:36 HallFrame systemd[1]: Started Mosquitto MQTT v3.1/v3.1.1 Broker.
```
This all seems fine?

Any ideas on anything else I can test? (I'm not sure I can get on with the next steps in the instructions (eg blueprints, helpers etc) until this part is working.

helgeerbe, 2022-07-15:
Can you show pic frames log file? Stop service and run it from the console. Just to check if it sends mqtt messages successfully. Maybe set Debug Level in configuration.yaml to Debug.

brynyneuadd, 2022-07-15:
Thanks! Have attached, this line looks supicious:
2022-07-15 10:00:29,058 - interface_mqtt.InterfaceMQTT - WARNING - Can't connect with mqtt broker. Reason = 5

```
2022-07-15 09:59:41,502 - image_cache.ImageCache - DEBUG - Creating an instance of ImageCache
2022-07-15 09:59:41,538 - image_cache.ImageCache - DEBUG - Updating cache
2022-07-15 09:59:41,539 - controller.Controller - INFO - creating an instance of Controller
2022-07-15 09:59:41,540 - image_cache.ImageCache - DEBUG - No unprocessed files in memory, checking disk
2022-07-15 09:59:41,542 - pi3d.Display - DEBUG - 

  Pi3D module - version 2.49

  Copyright (c) Tim Skillman, 2012-2022
  Copyright (c) Patrick Gaunt, 2012-2022
  Copyright (c) Tom Ritchford, 2012-2022

  Updates available from www.github.com/tipam/pi3d

2022-07-15 09:59:41,544 - pi3d.Display - DEBUG - Display size is w=3840, h=2160
2022-07-15 09:59:41,545 - image_cache.ImageCache - DEBUG - Found 0 new files on disk
2022-07-15 09:59:42,166 - pi3d.Shader - DEBUG - no file shaders//home/admin/picframe_data/data/shaders/blend_new.vs in pkg_resources trying
2022-07-15 09:59:42,190 - pi3d.Shader - DEBUG - no file shaders//home/admin/picframe_data/data/shaders/blend_new.fs in pkg_resources trying
2022-07-15 09:59:42,202 - pi3d.Shader - DEBUG - saving shader /home/admin/picframe_data/data/shaders/blend_new
2022-07-15 09:59:42,203 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.shape.Sprite.Sprite object at 0xb45687b0>
2022-07-15 09:59:42,204 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.Buffer.Buffer object at 0xb45689d0>
2022-07-15 09:59:42,217 - pi3d.Shader - DEBUG - saving shader uv_flat
2022-07-15 09:59:42,218 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.Texture.Texture object at 0xb4568b90>
2022-07-15 09:59:42,218 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.shape.Sprite.Sprite object at 0xb4568c10>
2022-07-15 09:59:42,219 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.Buffer.Buffer object at 0xb4568c70>
2022-07-15 09:59:42,220 - picframe.interface_peripherals - INFO - creating an instance of InterfacePeripherals
2022-07-15 09:59:42,220 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.util.Font.Font object at 0xb4568bb0>
2022-07-15 09:59:42,951 - pi3d.Shader - DEBUG - re-using shader uv_flat
2022-07-15 09:59:42,953 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.util.String.String object at 0xb4568bd0>
2022-07-15 09:59:42,954 - pi3d.util.String - DEBUG - Creating string ...
2022-07-15 09:59:42,954 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.Buffer.Buffer object at 0xb456e250>
2022-07-15 09:59:42,956 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.util.String.String object at 0xb456e2b0>
2022-07-15 09:59:42,957 - pi3d.util.String - DEBUG - Creating string ...
2022-07-15 09:59:42,958 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.Buffer.Buffer object at 0xb456e270>
2022-07-15 09:59:42,959 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.util.String.String object at 0xb456e230>
2022-07-15 09:59:42,960 - pi3d.util.String - DEBUG - Creating string ...
2022-07-15 09:59:42,961 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.Buffer.Buffer object at 0xb456e2f0>
2022-07-15 09:59:42,962 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.util.String.String object at 0xb456e210>
2022-07-15 09:59:42,963 - pi3d.util.String - DEBUG - Creating string ...
2022-07-15 09:59:42,964 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.Buffer.Buffer object at 0xb456e430>
2022-07-15 09:59:42,966 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.Texture.Texture object at 0xb4568c30>
2022-07-15 09:59:42,967 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.shape.Sprite.ImageSprite object at 0xb456e3b0>
2022-07-15 09:59:42,967 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.Buffer.Buffer object at 0xb456e330>
2022-07-15 09:59:42,969 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.Texture.Texture object at 0xb456e1b0>
2022-07-15 09:59:42,969 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.shape.Sprite.ImageSprite object at 0xb456e1f0>
2022-07-15 09:59:42,970 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.Buffer.Buffer object at 0xb456e1d0>
2022-07-15 09:59:42,971 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.shape.Sprite.ImageSprite object at 0xb456e130>
2022-07-15 09:59:42,971 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.Buffer.Buffer object at 0xb456e2d0>
2022-07-15 09:59:42,973 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.Texture.Texture object at 0xb4568ab0>
2022-07-15 09:59:42,973 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.shape.Sprite.ImageSprite object at 0xb456e450>
2022-07-15 09:59:42,974 - pi3d.util.Loadable - DEBUG - __init__: <pi3d.Buffer.Buffer object at 0xb456e3d0>
2022-07-15 09:59:43,001 - interface_mqtt.InterfaceMQTT - INFO - creating an instance of InterfaceMQTT
2022-07-15 09:59:43,013 - simple_server.InterfaceHttp - INFO - creating an instance of InterfaceHttp
2022-07-15 09:59:43,015 - interface_mqtt.InterfaceMQTT - WARNING - Can't connect with mqtt broker. Reason = 5
2022-07-15 09:59:43,018 - interface_mqtt.InterfaceMQTT - DEBUG - Send image attributes: {'latitude': None, 'longitude': None, 'location': None, 'EXIF FNumber': 0.0, 'EXIF ExposureTime': None, 'EXIF ISOSpeedRatings': 0.0, 'EXIF FocalLength': None, 'EXIF DateTimeOriginal': 1657470591.7937217, 'Image Model': None, 'Image Make': None, 'IPTC Caption/Abstract': None, 'IPTC Object Name': None, 'IPTC Keywords': None}
2022-07-15 09:59:43,019 - interface_mqtt.InterfaceMQTT - INFO - Send image state: {'image': 'Mimi-Rainbow.jpg'}
[... cut ...]
2022-07-15 10:00:57,085 - image_cache.ImageCache - DEBUG - Updating cache
2022-07-15 10:00:57,086 - image_cache.ImageCache - DEBUG - No unprocessed files in memory, checking disk
2022-07-15 10:00:57,088 - image_cache.ImageCache - DEBUG - Found 0 new files on disk
2022-07-15 10:00:59,100 - image_cache.ImageCache - DEBUG - Updating cache
2022-07-15 10:00:59,101 - image_cache.ImageCache - DEBUG - No unprocessed files in memory, checking disk
2022-07-15 10:00:59,106 - image_cache.ImageCache - DEBUG - Found 0 new files on disk
```

helgeerbe, 2022-07-15:
That means:

Reason code: 5 - Not authorized to perform the requested operation.

Check your settings in configuration.yaml

helgeerbe, 2022-07-15:
Glad that it works now. Nice frame. What kind of screen are you using?

brynyneuadd, 2022-07-15:
Thanks - few piece of walnut left over from another project, so made easy frame!

Using a Samsung 43" 4k TV:
https://www.currys.co.uk/products/samsung-ue43au8000kxxu-43-smart-4k-ultra-hd-hdr-led-tv-with-bixby-alexa-and-google-assistant-10221998.html

Pretty cheap, 4k, has lots of stuff from Samsung's photo frame range now built in and only 25mm thick, so ideal for things like this!

closed 2022-07-15 by brynyneuadd