helgeerbe/picframe #285: Non-MQTT brightness control?

Judman, 2022-10-03:
Hello,

I am working on building multiple standalone units to distribute to family members. They will not be connected to a home assistant instance and therefore cannot be easily controlled remotely.

I see image brightness is an option to be set by MQTT. Is there a way to set it outside of MQTT commands? I would love a brightness sensor and cronjob to make these changes every so often.

My current testing monitor is DVI so I cannot try the ddcutil as most suggest. Picframe seems to be the next best option in the meantime.

Thanks!

helgeerbe, 2022-10-04:
Hi @Judman
- activate pic frames internal Webserver and use 'curl' on localhost in a script to set the brightness or
- install mosquito (mqtt broker) locally on the raspi. Use mqqt messages in a script.

paddywwoof, 2022-10-04:
I think @sapnho wrote something about doing this with a background python script sending MQTT messages. I think this would be the easiest way to go if you wanted the brightness sensor to work in conjunction with the timing.

However I can't see the article offhand, just a general help page https://www.thedigitalpictureframe.com/

sapnho, 2022-10-04:
I only tried this here so far: https://www.thedigitalpictureframe.com/how-to-automatically-control-the-display-brightness-of-your-raspberry-pi-photo-frame/

Judman, 2022-10-04:
So it is super easy to turn on the HTTP server and use `curl "localhost:9000/?brightness=1"` to change the brightness.  Having messed around with MQTT in the past it did not seem worth the effort to set it up just for brightness when the HTTP server works well. I haven't worked on a Photocell (or LDR) script. 

However, I figured out ddcutil brightness control on the Pi 3 through HDMI. I used my camera to compare changing the brightness in PicFrame vs changing the monitors to similar levels. Turns out the pictures stay truer to color by adjusting the monitors' native brightness.

All that to say I think using DDC to adjust brightness is the current path I am going to pursue.

Thanks all for the amazing work on this project and being so quick to help with issues of a new user trying to learn all this!

closed 2022-10-04 by Judman