helgeerbe/picframe #286: Problems with HEIC images even with pyheif installed

TopDown71, 2022-10-24:
**Describe the bug**
picframe throws warning when opening HEIC images and fails if a HEIC is the first image. 
NOTE: username changed from 'pi' to 'myname' when building image using Raspberry Pi Imager

**To Reproduce**
build picframe, launch from terminal window, and confirm okay with .jpg images

add HEIC images

launch from terminal window and HEIC images display but:
   WARNING:get_image_meta.GetImageMeta:Failed attempt to convert /home/myname/Pictures/IMG_7823.HEIC 
  ** Have you installed pyheif? **

Install pyheif using steps by @luksan in https://github.com/pi3d/pi3d_demos/issues/24 since steps by @vinyli85 and @sapnho fail due to a version issue with Autoconf requiring 2.71 and I can't find how to update.

launch from terminal window - images display with same warning messages.

If this doesn't cause a problem when run from systemd then it would be okay - BUT

if the HEIC image is the first displayed by picframe the result in the terminal window is:
picframe ~/picframe_data/config/configuration.yaml
INFO:start.py:starting ['/home/myname/.local/bin/picframe', '/home/myname/picframe_data/config/configuration.yaml']
INFO:model.Model:Open config file: /home/myname/picframe_data/config/configuration.yaml:
WARNING:get_image_meta.GetImageMeta:Failed attempt to convert /home/myname/Pictures/IMG_7823.HEIC 
** Have you installed pyheif? **
Traceback (most recent call last):
  File "/home/myname/.local/bin/picframe", line 10, in <module>
    sys.exit(main())
  File "/home/myname/.local/lib/python3.7/site-packages/picframe/start.py", line 160, in main
    c.loop()
  File "/home/myname/.local/lib/python3.7/site-packages/picframe/controller.py", line 321, in loop
    (loop_running, skip_image) = self.__viewer.slideshow_is_running(pics, time_delay, fade_time, self.__paused)
  File "/home/myname/.local/lib/python3.7/site-packages/picframe/viewer_display.py", line 496, in slideshow_is_running
    wh_rat = (self.__display.width * self.__sfg.iy) / (self.__display.height * self.__sfg.ix)

There is no picframe display and no exit to terminal. Pi GUI is still functional. ctrl-C in terminal flickers the GUI once and terminal displays:
AttributeError: 'NoneType' object has no attribute 'iy'
^CYou pressed Ctrl-c!

Terminal window returns to normal.

**Expected behavior**
No errors, warnings, or terminal session freeze

**Screenshots**
see above

**What are your OS, Python and picframe versions(please complete the following information):?**

- Hardware: Raspberry Pi 3B+
- OS: Raspbian GNU/Linux 10 (buster)
- Python: 2.7.16
- picframe (run `picframe -v` if possible): 
$ picframe -v
INFO:start.py:starting ['/home/myname/.local/bin/picframe', '-v']
picframe version:  2022.06.03

Checking required packages......
PIL :  9.2.0
exifread :  3.0.0
pi3d :  2.49
yaml :  6.0
paho.mqtt :  1.6.1
iptcinfo3 :  2.1.4
numpy :  1.16.2
ninepatch : installed, but no version info

Checking optional packages......
pyheif : Not found!                                       _This seems strange considering I had what looked like a successful install_

[... 118 lines cut ...]
      shortcut: " "                       # default=" "
    display_off:                          # turn off the display (when off, any input from selected peripheral will turn it back on)
      enable: True                        # default=True
      label: "Display off"                # default="Display off"
      shortcut: "o"                       # default="o"
    location:                             # shows or hides location information
      enable: False                       # default=False
      label: "Location"                   # default="Location"
      shortcut: "l"                       # default="l"
    exit:                                 # exit PictureFrame
      enable: False                       # default=False
      label: "Exit"                       # default="Exit"
      shortcut: "e"                       # default="e"
    power_down:                           # power down the device, uses sudo
      enable: False                       # default=False
      label: "Power down"                 # default="Power down"
      shortcut: "p"                       # default="p"
```

````txt
log output goes here
````

**Additional context**
Add any other context about the problem here.

paddywwoof, 2022-10-24:
Hmm. Firstly thanks for that super explicit bug report. I remember HEIC being a bit of a nightmare to install so I suspect your problems are linked to whatever prevented you from following @sapnho steps. I have only quickly read your notes on my phone (I'm away from home at the moment) but do you have to use python 2?

TopDown71, 2022-10-24:
Thanks for the quick reply! I’ve been ‘hacking’ code for about 44 years now - mainly as a way to get more done in my job. So it’s very much utility FORTRAN and VBA with a scattering of C. I’m somewhat familiar with python, but don’t get into it all that often. My history and trying to fix things for people I shared code with explains my level of detail 😀 

I just typed python —version so I guess it defaulted to 2. I do have python 3.7.3 on the Pi as well. I guess it’s part of the standard Raspberry PI Imager build unless it came along for the ride with the picframe install.

My Pi shows no updates or upgrades available so I don’t see how to update autoconf. When I check the version I get:

apt show autoconf
Package: autoconf
Version: 2.69-11
Priority: optional
Section: devel
Maintainer: Ben Pfaff ***@***.***>
Installed-Size: 1,913 kB
Depends: perl (>> 5.005), m4 (>= 1.4.13), debianutils (>= 1.8)
Recommends: automake | automaken
Suggests: autoconf-archive, gnu-standards, autoconf-doc, libtool, gettext
Breaks: gettext (<< 0.10.39), pkg-config (<< 0.25-1.1)
Homepage: http://www.gnu.org/software/autoconf/
Download-Size: 341 kB
APT-Sources: http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages
Description: automatic configure script builder
 The standard for FSF source packages.  This is only useful if you
 write your own programs or if you extensively modify other people's
 programs.
 .
 For an extensive library of additional Autoconf macros, install the
 `autoconf-archive' package.
 .
 This version of autoconf is not compatible with scripts meant for
 Autoconf 2.13 or earlier.

bigcat88, 2022-10-24:
maybe a bit brazen, but wouldn't it be easier to just use `pillow_heif` instead of `pyheif`?

Pros, to do that:
1. It has working arm7 wheels(that will probably work on Rasberry).
2. It supports EXIF and other standard Pillow's meta data.
3. It is simplier to use, just adding it as a Pillow plugin.

paddywwoof, 2022-10-25:
@bigcat88 has a good point. I think pillow_heif was in an early state of development or didn't work when the how-to and module were written. It could well be that they would be a tidier approach now.

helgeerbe, 2022-10-25:
@bigcat88 Thanks for the hint. If I have some time left, I will give it a try. Looks like it could easily integrated.

TopDown71, 2022-10-25:
Thank you both for looking into this! As a followup, I was trying to confirm the heic image displayed, but it does not. The heic file name and date appear, but picframe displays what must be the next displayable image. If it's just this small bit of code that has to be touched it's worth a shot.

helgeerbe, 2022-11-28:
I created a branch pi_heif and pulled #288 from @bigcat88 .
I did some changes to the original pr.
- make pi_heif a mandatory lib (>= 0.8.0)
- make use of `register_heif_opener()`to simplify the code. 

On my Mac, everything seems to work fine. I will merge it to dev for others to test.

bigcat88, 2022-11-28:
Just a note: you can call `register_heif_opener` once anywhere in project.
For example in `start.py`, and it will be enougth. Just to simplify code even more :)

helgeerbe, 2022-11-28:
I tried that. But `get_image_object(fname):` is a static function. I have to call `register_heif_opener()` in its context.

helgeerbe, 2022-11-28:
Actual I have some issues with the meta data. They are read properly, but don't show up on home assistant. I have to find out why.

helgeerbe, 2022-11-28:
There seems to be a problem with exifread lib 3.0.0 on raspberry (https://github.com/ianare/exif-py/issues/160). Downgrading to a version < 3 solves the problem.

helgeerbe, 2022-11-29:
@bigcat88 I think about to remove exifread lib from picframe,  with your lib Pillow supports heic/heif images.

I wrote a small test to compare the values for a heic image with the exifread tool. And I found that I got the wrong orientation.
1 instead of 6. The rest looks fine. Any ideas?

````python
import pytest
import logging

from PIL import Image
from PIL.Image import Exif
from PIL.ExifTags import TAGS, GPSTAGS
from pi_heif import register_heif_opener
from fractions import Fraction

logger = logging.getLogger("test_get_exifs_by_pillow")
logger.setLevel(logging.DEBUG)

register_heif_opener()

def get_labeled_exif(exif):
    return {
        TAGS.get(key, key): value
        for key, value in exif.items()
    }

def get_exif_ifd(exif):
    for key, value in TAGS.items():
        if value == "ExifOffset":
            break
    info = exif.get_ifd(key)
    return {
        TAGS.get(key, key): value
        for key, value in info.items()
    }

def get_geo(exif):
    for key, value in TAGS.items():
        if value == "GPSInfo":
            break
    gps_info = exif.get_ifd(key)
    return {
        GPSTAGS.get(key, key): value
        for key, value in gps_info.items()
    }

def convert_to_degrees(value):
    (deg, min, sec) = value
    d = float(deg.numerator) / float(deg.denominator if deg.denominator > 0 else 1) #TODO better catching?
    m = float(min.numerator) / float(min.denominator if min.denominator > 0 else 1)
    s = float(sec.numerator) / float(sec.denominator if sec.denominator > 0 else 1)
    return d + (m / 60.0) + (s / 3600.0)

def test_pillow_exifs_heic():
    try:
        image = Image.open("test/images/test3.HEIC")
   
        exifs = image.getexif()
        labled_exif = get_labeled_exif(exifs)
        labled_exif_ifd = get_exif_ifd(exifs)
        labled_exif_geo = get_geo(exifs)

        print(labled_exif)
        print(labled_exif_ifd)
        print(labled_exif_geo)

        orientation = labled_exif['Orientation']
        assert  orientation == 6
        assert labled_exif['Make'] == "Apple"
        assert labled_exif['Model']  == "iPhone 8"

        assert labled_exif_ifd['ExifImageWidth'] == 4032
        assert labled_exif_ifd['ExifImageHeight'] == 3024
        assert labled_exif_ifd['FNumber']== 1.8
        assert str(Fraction(labled_exif_ifd['ExposureTime']))== "1/5"
        assert labled_exif_ifd['ISOSpeedRatings']== 100
        assert labled_exif_ifd['FocalLength']== 3.99
        assert labled_exif_ifd['LensModel']== "iPhone 8 back camera 3.99mm f/1.8"
        assert labled_exif_ifd['DateTimeOriginal']== "2021:05:14 20:27:14"

        # GPS
        lat = convert_to_degrees(labled_exif_geo['GPSLatitude'])
        if len(labled_exif_geo['GPSLatitudeRef']) > 0 and labled_exif_geo['GPSLatitudeRef'][0] == 'S':
            # assume zero length string means N
            lat = 0 - lat
        assert lat == 38.71365
      
        lon = convert_to_degrees(labled_exif_geo['GPSLongitude'])
        if len(labled_exif_geo['GPSLongitudeRef']) > 0 and labled_exif_geo['GPSLongitudeRef'][0] == 'W':
            # assume zero length string means N
            lon = 0 - lon
        assert lon == -78.15960555555556

    except:
        pytest.fail("Unexpected exception")
```

bigcat88, 2022-11-29:
The problem with orientation is described in docs. It is a complex problem for HEIF...

helgeerbe, 2022-11-30:
@bigcat88  Thanks for the hint. I removed also the "exifread" lib. Will test this dev branch on my frame. If everything works well I will make a new release, which supports heif/heic natively.

TopDown71, 2022-12-01:
Thank you both for following up on what was obviously more than just a super simple issue. I’ll install a new version when I see that it is available.

Regards,
Dan

helgeerbe, 2022-12-04:
Should be fixed with release 2022.12.04

closed 2022-12-04 by helgeerbe

TopDown71, 2023-01-10:
I had a family situation come up over the holidays and finally got to build a new frame over the last few days. Everything seems to be working perfectly now. Thanks for all of the joint effort it took to get there!

Happy new year and kind regards