helgeerbe/picframe #297: Bug: Install errors using "pip3 install picframe"

nyarrow, 2022-12-08:
Install fails when installing on Raspian Debian Buster 2022-09-22 (with current security updates).

Trace (trim previous successful steps):
Collecting ninepatch (from picframe)
  Using cached https://files.pythonhosted.org/packages/3b/02/614abc019f18e4f04975fbb3b2eceb63082754afa2b96f02b2aab7699eb7/ninepatch-0.1.22-py3-none-any.whl
Collecting pi-heif>=0.8.0 (from picframe)
  Using cached https://files.pythonhosted.org/packages/ec/75/a55256740a9564ab742fc9d3332c6140229b40e25374c05a052a42ae93ff/pi_heif-0.8.0.tar.gz
Exception:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/_internal/cli/base_command.py", line 143, in main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/_internal/commands/install.py", line 338, in run
    resolver.resolve(requirement_set)
  File "/usr/lib/python3/dist-packages/pip/_internal/resolve.py", line 102, in resolve
    self._resolve_one(requirement_set, req)
  File "/usr/lib/python3/dist-packages/pip/_internal/resolve.py", line 256, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "/usr/lib/python3/dist-packages/pip/_internal/resolve.py", line 209, in _get_abstract_dist_for
    self.require_hashes
  File "/usr/lib/python3/dist-packages/pip/_internal/operations/prepare.py", line 298, in prepare_linked_requirement
    abstract_dist.prep_for_dist(finder, self.build_isolation)
  File "/usr/lib/python3/dist-packages/pip/_internal/operations/prepare.py", line 100, in prep_for_dist
    self.req.load_pyproject_toml()
  File "/usr/lib/python3/dist-packages/pip/_internal/req/req_install.py", line 428, in load_pyproject_toml
    str(self)
  File "/usr/lib/python3/dist-packages/pip/_internal/pyproject.py", line 43, in load_pyproject_toml
    pp_toml = pytoml.load(f)
  File "/usr/share/python-wheels/pytoml-0.1.2-py2.py3-none-any.whl/pytoml/parser.py", line 303, in load
    filename=fin.name)
  File "/usr/share/python-wheels/pytoml-0.1.2-py2.py3-none-any.whl/pytoml/parser.py", line 370, in loads
    toks.expect('=', 'expected_equals')
  File "/usr/share/python-wheels/pytoml-0.1.2-py2.py3-none-any.whl/pytoml/parser.py", line 250, in expect
    self.error(error_text)
  File "/usr/share/python-wheels/pytoml-0.1.2-py2.py3-none-any.whl/pytoml/parser.py", line 253, in error
    raise TomlError(message, self.pos[0][0], self.pos[0][1], self._filename)
pytoml.core.TomlError: /tmp/pip-install-a_s_6d2j/pi-heif/pyproject.toml(31, 7): expected_equals

helgeerbe, 2022-12-08:
Hi, there seems indeed a problem with installing pi_heif. See #295 
Can you tell me which version you are using `pip -V`?
Mine is `pip 22.3.1 from /opt/homebrew/lib/python3.10/site-packages/pip (python 3.10)`

What happens if you upgrade pip?

dougc84, 2022-12-08:
@helgeerbe I had the same issue. My version of `pip` from RasPi Buster showed 17. Doing the suggestion here:

https://github.com/helgeerbe/picframe/discussions/295#discussioncomment-4332837

fixed the problem.

nyarrow, 2022-12-08:
Prior to this, I had a clean image prepared per "TheDigitalPictureFrame" instructions: https://www.thedigitalpictureframe.com/how-to-set-up-your-raspberry-pi-for-your-digital-picture-frame/

I was at:
> pip 18.1 from /usr/lib/python2.7/dist-packages/pip (python 2.7)

Per dougc's suggestion, I ran:

> pip3 install --upgrade pip and also
> pip3 install --upgrade pi3d

I still see:
> pip 18.1 from /usr/lib/python2.7/dist-packages/pip (python 2.7)

I still get similar errors when executing:
> sudo pip3 install picframe

As such, I updated Python with instructions here: https://www.folkstalk.com/tech/debian-upgrade-python-with-code-examples/

Using the tarball from here: https://www.python.org/ftp/python/3.11.1/Python-3.11.1.tgz

After that, versions are as follows:
> Python 3.11.1
> pip 22.3.1 from /usr/local/lib/python3.11/site-packages/pip (python 3.11)

After that, I still get similar errors when executing:
> sudo pip3 install picframe

Stack trace:

> Collecting paho-mqtt (from picframe)
>   Downloading https://www.piwheels.org/simple/paho-mqtt/paho_mqtt-1.6.1-py3-none-any.whl (75kB)
>     100% |████████████████████████████████| 81kB 261kB/s 
> Collecting pi-heif>=0.8.0 (from picframe)
>   Using cached https://files.pythonhosted.org/packages/ec/75/a55256740a9564ab742fc9d3332c6140229b40e25374c05a052a42ae93ff/pi_heif-0.8.0.tar.gz
> Exception:
> Traceback (most recent call last):
>   File "/usr/lib/python3/dist-packages/pip/_internal/cli/base_command.py", line 143, in main
>     status = self.run(options, args)
>   File "/usr/lib/python3/dist-packages/pip/_internal/commands/install.py", line 338, in run
>     resolver.resolve(requirement_set)
>   File "/usr/lib/python3/dist-packages/pip/_internal/resolve.py", line 102, in resolve
>     self._resolve_one(requirement_set, req)
>   File "/usr/lib/python3/dist-packages/pip/_internal/resolve.py", line 256, in _resolve_one
>     abstract_dist = self._get_abstract_dist_for(req_to_install)
>   File "/usr/lib/python3/dist-packages/pip/_internal/resolve.py", line 209, in _get_abstract_dist_for
>     self.require_hashes
>   File "/usr/lib/python3/dist-packages/pip/_internal/operations/prepare.py", line 298, in prepare_linked_requirement
>     abstract_dist.prep_for_dist(finder, self.build_isolation)
>   File "/usr/lib/python3/dist-packages/pip/_internal/operations/prepare.py", line 100, in prep_for_dist
>     self.req.load_pyproject_toml()
>   File "/usr/lib/python3/dist-packages/pip/_internal/req/req_install.py", line 428, in load_pyproject_toml
>     str(self)
>   File "/usr/lib/python3/dist-packages/pip/_internal/pyproject.py", line 43, in load_pyproject_toml
>     pp_toml = pytoml.load(f)
>   File "/usr/share/python-wheels/pytoml-0.1.2-py2.py3-none-any.whl/pytoml/parser.py", line 303, in load
>     filename=fin.name)
>   File "/usr/share/python-wheels/pytoml-0.1.2-py2.py3-none-any.whl/pytoml/parser.py", line 370, in loads
>     toks.expect('=', 'expected_equals')
>   File "/usr/share/python-wheels/pytoml-0.1.2-py2.py3-none-any.whl/pytoml/parser.py", line 250, in expect
>     self.error(error_text)
>   File "/usr/share/python-wheels/pytoml-0.1.2-py2.py3-none-any.whl/pytoml/parser.py", line 253, in error
>     raise TomlError(message, self.pos[0][0], self.pos[0][1], self._filename)
> pytoml.core.TomlError: /tmp/pip-install-4ny50etv/pi-heif/pyproject.toml(31, 7): expected_equals

helgeerbe, 2022-12-08:
Hm,

@nyarrow 
`pip` points to python 2.7. As for as I know python 3.6 is minimum

So you have to go with `pip3`, which points to python 3.11.  This looks good to me. 

Can you try to install the lib manually as described here https://pypi.org/project/pi-heif/

````sh
python3 -m pip install -U pip
python3 -m pip install pi-heif
````

helgeerbe, 2022-12-08:
@bigcat88 as you are the owner of pi_heif. Do you have any idea?

bigcat88, 2022-12-08:
> ```shell
> python3 -m pip install -U pip
> python3 -m pip install pi-heif
> ```

this is the right code. i have actions that tests install almost on all OS(mac, win, many varioues types of Linux) with different pythons.
Even with python that built from sources...

It does not look like a problem with installing pi_heif, but i will try to help, give me 15 minutes =)

bigcat88, 2022-12-08:
```
update-alternatives: using /usr/local/bin/python3.11 to provide /usr/bin/python (python) in auto mode
# python3 -V
/bin/sh: 42: python3: not found
# python -V
Python 3.11.1
# update-alternatives --install /usr/bin/pip pip /usr/local/bin/pip3.11 1
update-alternatives: using /usr/local/bin/pip3.11 to provide /usr/bin/pip (pip) in auto mode
# pip -V
pip 22.3.1 from /usr/local/lib/python3.11/site-packages/pip (python 3.11)
# pip install pi_heif
Collecting pi_heif
  Downloading pi_heif-0.8.0-cp36-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (929 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 929.5/929.5 kB 1.1 MB/s eta 0:00:00
Collecting cffi>=1.14.6
  Downloading cffi-1.15.1-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (462 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 462.6/462.6 kB 673.1 kB/s eta 0:00:00
Collecting pillow>=6.2.0
  Downloading Pillow-9.3.0-cp311-cp311-manylinux_2_28_x86_64.whl (3.3 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.3/3.3 MB 395.4 kB/s eta 0:00:00
Collecting pycparser
  Downloading pycparser-2.21-py2.py3-none-any.whl (118 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 118.7/118.7 kB 399.5 kB/s eta 0:00:00
Installing collected packages: pycparser, pillow, cffi, pi_heif
Successfully installed cffi-1.15.1 pi_heif-0.8.0 pillow-9.3.0 pycparser-2.21
```

Do not see any problem, built python3.11 from hat tarball on Debian 11 from source by that guide and install it...
Maybe problem is that in that guide:
`/usr/local/bin/pip3.11` maps to `pip` 
and
`/usr/local/bin/python3.11` maps to `python`.
Imho, awful guide due to this. Some soft on debian requires python2, it will call `python` command and get python with version 3. Any admin will get easy rage and very fast get drunk due to this. =)

So for this guide correct command is `pip install pi_heif` - but it is a mess.

bigcat88, 2022-12-08:
@helgeerbe 
paste output of
`sudo pip3 -V`

nyarrow, 2022-12-08:
Okay - this worked:
> sudo python3 -m pip install -U pip
> sudo python3 -m pip install pi-heif
> sudo pip3 install picframe

I'm not sure if this would have worked without the painful process of compiling a new version of Python - but I am up and running. As such, it may take a bit more troubleshooting to recommend a path from clean install forward.

Thank y'all for your help!

helgeerbe, 2022-12-08:
@bigcat88  I downgraded to pip version 18.0 and got this error. So I guess upgrade pip to the latest version before installation  is a good advice.

````sh
herbe@develop:~$ /home/herbe/.local/bin/pip install pi-heif --upgrade -v
Created temporary directory: /tmp/pip-ephem-wheel-cache-tsrl6t93
Created temporary directory: /tmp/pip-req-tracker-yrm1zqo6
Created requirements tracker '/tmp/pip-req-tracker-yrm1zqo6'
Created temporary directory: /tmp/pip-install-wdvdqcpn
Collecting pi-heif
  1 location(s) to search for versions of pi-heif:
  * https://pypi.org/simple/pi-heif/
  Getting page https://pypi.org/simple/pi-heif/
  Looking up "https://pypi.org/simple/pi-heif/" in the cache
  Current age based on date: 124
  Freshness lifetime from max-age: 600
  Freshness lifetime from request max-age: 600
  The response is "fresh", returning cached response
  600 > 124
  Analyzing links from page https://pypi.org/simple/pi-heif/
    Skipping link https://files.pythonhosted.org/packages/82/21/675824a9378cc34e50f7c9d8af21baa4fa54d5b34f9f9a8836a86f0bde89/pi_heif-0.7.0-cp36-abi3-manylinux_2_17_aarch64.manylinux2014_aarch64.whl#sha256=4dc2093a5db1a80c4a25890389aa3a76dd55e78027520c58f7110451ac67b163 (from https://pypi.org/simple/pi-heif/) (requires-python:>=3.6); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/22/02/f80410ab853e654a469c2de8abc78f1dd22b44a5787da904d639629e9add/pi_heif-0.7.0-cp36-abi3-manylinux_2_17_i686.manylinux2014_i686.whl#sha256=34400f933f6d29e9fb539e0ebdfeae5ce5ae6a187014877a5579db961f65a19c (from https://pypi.org/simple/pi-heif/) (requires-python:>=3.6); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/af/0e/d0143e16a5747077cd45abeae166ac4fb01add72f6a5e0493b4f8c86ac55/pi_heif-0.7.0-cp36-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.whl#sha256=e93876a713e7c3331f420bb5cf141765845c39bb4ce7aae2305c865ae426417b (from https://pypi.org/simple/pi-heif/) (requires-python:>=3.6); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/9c/77/a23c3b8dc2f644e4d8d17ab63d7f02704aa5e8e3be81a17aa4d89869da8d/pi_heif-0.7.0-cp36-abi3-musllinux_1_1_aarch64.whl#sha256=a5f78c028c778d5984b00524539756e849f5538d7954857e43134b5a8cebf6e8 (from https://pypi.org/simple/pi-heif/) (requires-python:>=3.6); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/9b/60/bf986faaea23a62737dd642609fa12e47af157a6113aa681992660915d0e/pi_heif-0.7.0-cp36-abi3-musllinux_1_1_i686.whl#sha256=195048eee5b47b5c61f2a64b210ac3d57970ea3995f64abda043a2965233afef (from https://pypi.org/simple/pi-heif/) (requires-python:>=3.6); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/aa/1d/93397161a43af79c2ad050c077dd9ab858144726ff34ec56a5912d03622a/pi_heif-0.7.0-cp36-abi3-musllinux_1_1_x86_64.whl#sha256=cc095ff85aa0c15c3a7b5ffe28fe3c7079a8050bf63e59e12f75fadd8d6d631a (from https://pypi.org/simple/pi-heif/) (requires-python:>=3.6); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/ee/94/5b8e492806d2a84faaafe5d742cccbbca44632e61a45dc24b8635d55dbd4/pi_heif-0.7.0-cp37-abi3-macosx_10_9_x86_64.whl#sha256=2ec712a9f771ac2b527b353b25474484e6c24d2d0317724bf57ae879591ec16b (from https://pypi.org/simple/pi-heif/) (requires-python:>=3.6); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/da/a9/6933ef4d36a7b8dc8a08c4b55427de9b732c18e0560fda315dae0ca8697b/pi_heif-0.7.0-cp37-abi3-win_amd64.whl#sha256=4e9aa7d907d0ec9af36efdb95188d75f3abad1a088d8aa9b9b3b0e00be52f65e (from https://pypi.org/simple/pi-heif/) (requires-python:>=3.6); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/f5/e2/8a9a993f4d902fdcbad532c75039b22ef58ee3e3efe5b6db91c21d2fb131/pi_heif-0.7.0-cp38-abi3-macosx_12_0_arm64.whl#sha256=e7f385b4cd94d1c24255ae72804acef30b4134bf396955efcfb7e41c8fa01a45 (from https://pypi.org/simple/pi-heif/) (requires-python:>=3.6); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/ce/62/c52e9346e798da01d526daa732279cc164ceb9bf61393fa2202bfe857fc6/pi_heif-0.7.0-cp38-abi3-manylinux_2_31_armv7l.whl#sha256=0cb755bea35cdf132eab6b4c1de6234e518ddb475e1bdd309f7b8b849e8f30d3 (from https://pypi.org/simple/pi-heif/) (requires-python:>=3.6); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/81/2e/b37a7b9a6f80d336f9bc65fb1ae80d9558860a2692c101575a240e841473/pi_heif-0.7.0-pp37-pypy37_pp73-macosx_10_9_x86_64.whl#sha256=9119cf12d9a38b434b848233a69c74363765eb13d894afbe1189c22ff2d0c28e (from https://pypi.org/simple/pi-heif/) (requires-python:>=3.6); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/c5/67/687a1d278ca3c9f1a59af4546ebda7b9cb92820b55a14f325e6b7fead004/pi_heif-0.7.0-pp37-pypy37_pp73-manylinux_2_17_aarch64.manylinux2014_aarch64.whl#sha256=292a168cdbc0a9fdfac031045b154b1d67f1694ab5b4ddf0cc94ed4ecfc8fd3d (from https://pypi.org/simple/pi-heif/) (requires-python:>=3.6); it is not compatible with this Python
    Skipping link https://files.pythonhosted.org/packages/6e/84/bdefd3ae0f1df5ee5fe6a162487f2b07376b8518029fd7b6a53406169470/pi_heif-0.7.0-pp37-pypy37_pp73-manylinux_2_17_i686.manylinux2014_i686.whl#sha256=8348b04381b8f38de86357215899156003282142e88f4c64cfeaf4c2a8b1be7f (from https://pypi.org/simp [... cut ...]
  File "/home/herbe/.local/lib/python3.8/site-packages/pip/_vendor/pytoml/parser.py", line 124, in expect_eof
    return self._expect(self.consume_eof())
  File "/home/herbe/.local/lib/python3.8/site-packages/pip/_vendor/pytoml/parser.py", line 164, in _expect
    raise TomlError('msg', self._pos[0], self._pos[1], self._filename)
pip._vendor.pytoml.core.TomlError: /tmp/pip-install-wdvdqcpn/pi-heif/pyproject.toml(31, 1): msg
You are using pip version 18.0, however version 22.3.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
````

nyarrow, 2022-12-08:
I spoke too soon. The install was successful, but now I get this when running:

> pi@MasterTV:~/Pictures $ picframe ~/picframe_data/config/configuration.yaml
> INFO:start.py:starting ['/usr/local/bin/picframe', '/home/pi/picframe_data/config/configuration.yaml']
> INFO:model.Model:Open config file: /home/pi/picframe_data/config/configuration.yaml:
> Traceback (most recent call last):
>   File "/usr/local/bin/picframe", line 8, in <module>
>     sys.exit(main())
>   File "/usr/local/lib/python3.7/dist-packages/picframe/start.py", line 135, in main
>     c.start()
>   File "/usr/local/lib/python3.7/dist-packages/picframe/controller.py", line 330, in start
>     self.__viewer.slideshow_start()
>   File "/usr/local/lib/python3.7/dist-packages/picframe/viewer_display.py", line 452, in slideshow_start
>     display_config=pi3d.DISPLAY_CONFIG_HIDE_CURSOR, background=self.__background, use_glx=self.__use_glx)
>   File "/home/pi/.local/lib/python3.7/site-packages/pi3d/Display.py", line 570, in create
>     display_config=display_config, window_title=window_title, use_glx=use_glx)
>   File "/home/pi/.local/lib/python3.7/site-packages/pi3d/util/DisplayOpenGL.py", line 120, in create_display
>     assert self.context != EGL_NO_CONTEXT and self.context is not None
> AttributeError: 'DisplayOpenGL' object has no attribute 'context'

helgeerbe, 2022-12-08:
@nyarrow That's a different issue. On a notebook you might have to install openGL mesa  `mesa-utils-extra`

nyarrow, 2022-12-09:
Solved!

I started over with a clean install, using the instructions from here:

> https://www.thedigitalpictureframe.com/how-to-set-up-your-raspberry-pi-for-your-digital-picture-frame/

I then performed these steps:
> sudo python3 -m pip install -U pip
> sudo python3 -m pip install pi-heif

Then I resumed with the instructions from here:
> https://www.thedigitalpictureframe.com/how-to-add-crossfading-slide-transitions-to-your-digital-picture-frame-using-pi3d/

All is working now!

closed 2022-12-09 by nyarrow