helgeerbe/picframe #302: PicFrame stops rotating pictures. No log entries

nyarrow, 2022-12-11:
On my Raspberry Pi 4, I have PicFrame set to continually rotate pictures. After a variable interval, PicFrame stops rotating pictures, but leaves the final picture displayed.

I can't seem to find a root cause, nor have I figured out how to work around the issue.

Restarting PicFrame seems to temporarily resolve the issue.

Tried so far:

- I have set PicFrame Debug mode all the way down to INFO, and not seeing anything unusual.
- Validated "vcgencmd get_throttled" shows no issues
- Validated "tvservice -s" and "tvservice -n" look correct
- Reviewed all logs in /var/log
- I've tried setting PicFrame logging to mode "DEBUG", but the output is so verbose it seems to be a race between filling my SD card and encountering the issue...

The only things that seem odd:

In /var/log/kern.log, occasional messages like this:
```
./kern.log.1:Dec 10 23:06:12 MasterTV kernel: [   28.305253] broken atomic modeset userspace detected, disabling atomic
./kern.log.1:Dec 10 23:07:34 MasterTV kernel: [   98.050260] broken atomic modeset userspace detected, disabling atomic
./kern.log.1:Dec 10 23:29:33 MasterTV kernel: [   25.765626] broken atomic modeset userspace detected, disabling atomic
./kern.log.1:Dec 10 23:30:14 MasterTV kernel: [   54.466768] broken atomic modeset userspace detected, disabling atomic
./kern.log.1:Dec 10 23:32:52 MasterTV kernel: [  212.056889] broken atomic modeset userspace detected, disabling atomic

```
Also, in /var/log/daemon.log, occasional series of messages like this:

```
Dec 10 17:55:05 MasterTV xinit[1124]: /usr/bin/xinit: connection to X server lost
Dec 10 17:55:05 MasterTV xinit[1124]: #015
Dec 10 17:55:05 MasterTV systemd[1]: Stopping PictureFrame on Pi4...
Dec 10 17:55:05 MasterTV xinit[1124]: waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.
Dec 10 17:55:06 MasterTV xinit[1124]: /usr/bin/xinit: unexpected signal 15
Dec 10 17:55:06 MasterTV systemd[1]: picframe.service: Main process exited, code=exited, status=1/FAILURE
Dec 10 17:55:06 MasterTV systemd[1]: picframe.service: Failed with result 'exit-code'.
Dec 10 17:55:06 MasterTV systemd[1]: Stopped PictureFrame on Pi4.
Dec 10 17:55:06 MasterTV systemd[1]: Started PictureFrame on Pi4.
```

While the pictures are frozen, it looks like picframe is still running:
```
pi@MasterTV:/var/log $ ps -Af | grep pic
root       713     1  0 Dec10 ?        00:00:00 /usr/bin/xinit /usr/bin/python3 /home/pi/picframe_data/run_start.py /home/pi/picframe_data/config/configuration.yaml
root       718   713  2 Dec10 ?        00:10:50 /usr/bin/python3 /home/pi/picframe_data/run_start.py /home/pi/picframe_data/config/configuration.yaml
pi        1880   658  0 07:43 pts/0    00:00:00 grep --color=auto pic
```

I'm having difficulties troubleshooting further (or building a workaround). One workaround I have considered is detecting the last time the screen was re-written, and issuing a picframe restart if it is longer than expected. However, I have yet to find a timer I can work off of (and I don't want to kill my SD card by writing it all to disk).

Any suggestions on further troubleshooting steps or on building a work-around?

helgeerbe, 2022-12-12:
@nyarrow 
What do you mean with continually rotate pictures? Cycling through your pictures?

What OS are you using. Bullseye or Buster? Can you dump picframe -v (libs version picframe is using).
Did you upgrade to the latest release? Please upgrade pip before upgrading picframe.

Does the freeze occur on the same image?

Can you start picframe from command line in DEBUG mode. So you don't have to save the log file. If you start picframe via systemctl, disable restart on failure and check log file with journalctl (jurnalctl is using log rotation, so you shouldn't fill your SD card with the log file).

Updating cache runs in a second thread. So those messages seems to be fine. But I'm wandering what might block the main thread.

nyarrow, 2022-12-12:
Ok - looks like this is related to https://github.com/helgeerbe/picframe/issues/296. I have another picFrame instance running on a different Pi over the same photos. it is running on a different version of picFrame that doesn't seem to have this issue. As such, it looks like this issue was introduced in the 2022.12.04 release...

This one does not fail on the same bad EXIF data:
```
INFO:start.py:starting ['/home/pi/.local/bin/picframe', '-v']
picframe version:  2022.06.03

Checking required packages......
PIL :  9.3.0
exifread :  3.0.0
pi3d :  2.49
yaml :  6.0
paho.mqtt :  1.6.1
iptcinfo3 :  2.1.4
numpy :  1.16.2
ninepatch : installed, but no version info

Checking optional packages......
pyheif : Not found!
```

helgeerbe, 2022-12-12:
Hi @nyarrow Thanks for finding that. Answer is yes and no. No, because it is an exif error and not a xmp error.

Yes, because I removed exifread lib and use Pillow to extract the exif tags. Pillow seems to be more sensitive and I might have to make the tag reading more fault tolerant. 

Is there a chance to get the image, or is it to private?

nyarrow, 2022-12-12:
I've attached one of the files I found.

[image]

I'm working on a shell script to find files with bad EXIF data - I seem to have a non-trivial number of them, and they are a pain to find in the mix of 4k other photos.

nyarrow, 2022-12-12:
I've attached one of the files I found (zipped to assure it gets across intact)
[BadEXIF.zip](https://github.com/helgeerbe/picframe/files/10212975/BadEXIF.zip)

I'm playing around with a shell script to identify files with bad EXIF data - I seem to have an annoyingly-large quantity of them. I'll share if I get it working.

jgodfrey, 2022-12-13:
> I'm playing around with a shell script to identify files with bad EXIF data

In case you're unaware, EXIFTOOL is a swiss-army knife of EXIF goodness, and has the ability to batch-identify (and potentially repair) bad EXIF data... I don't have the command line syntax handy, but it should be easy to find online.

Here's [one hit](https://exiftool.org/forum/index.php?topic=9497.0) from the EXIFTOOL forums, though many others exist...

nyarrow, 2022-12-13:
I missed those options in EXIFTOOL... Much easier than stringing a bunch of shell commands together...

Thank you!

helgeerbe, 2022-12-13:
Thanks for sharing the image. Fix was quite easy. If an exif tag number is not in the dictionary to look up for the tag name, the number itself is used as tag name.

Should be fixed with release 2022.12.13

closed 2022-12-13 by helgeerbe

nyarrow, 2022-12-13:
Testing...

`sudo pip install git+https://github.com/helgeerbe/picframe.git`

nyarrow, 2022-12-14:
I'm going to call this a success - picframe has not stopped once since I sent the above message.

Thank you helgeerbe!