helgeerbe/picframe #303: Cannot install picframe

Dangerous2A, 2022-12-12:
**Describe the bug**
A clear and concise description of what the bug is.

Hi,

I have downloaded picframe several times recently but since last week I get a string of errors

I contacted piwheels Ben Nuttall who tells me there is an issue as follows:

<<The problem you're having is with package [pi-heif](https://www.piwheels.org/project/pi-heif/)>>

then when I asked if I needed to install it:

<<You don't need to install it first, as pip will resolve the dependencies. But if it fails, the whole installation will fail.
It doesn't look like that package has ever built successfully before. It may be possible for you to build it yourself if you find out what the requirements are.>>

No idea about Python (and still learning about R-pi, but it did install until recently). Do you have any ideas what the issue may be?

I have tried installing on a Rpi4 and 3 both give a similar error. The attached file is for Rpi 3 after running sudo pip3 picframe install.

**To Reproduce**
Steps to reproduce the behavior:

Try to install picframe on new installation of Buster on pi3 or pi4

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
Installation of program

**Screenshots**
If applicable, add screenshots to help explain your problem.

**What are your OS, Python and picframe versions(please complete the following information):?**

- Hardware: Rpi 3 / 4
- OS: Buster latest update, tried  install from Rpi, but same via SSH
- Python: 3
- picframe (run `picframe -v` if possible): not known

```yaml
configuration.yaml goes here
```
Not yet configured as not installed.

````txt
log output goes here

````
Cannot find log file. Where should it be? There is no Picframe folder as  it has not installed.

**Additional context**
Add any other context about the problem here.
Everything after install is run is in file:
[rpi3_error.txt](https://github.com/helgeerbe/picframe/files/10209084/rpi3_error.txt)

helgeerbe, 2022-12-12:
Hi,
I have seen this error a couple of times before. Can you check your pip version `pip -V` It should be something > 21.

Try this:
Upgrade pip, install pi-heif separately, install picframe.

````
python3 -m pip install -U pip
python3 -m pip install pi-heif
python3 -m install picframe
````

Dangerous2A, 2022-12-13:
Hello,

 

I got some odd results from pip -V

 

***@***.***:~ $ pip -V

pip 18.1 from /usr/lib/python2.7/dist-packages/pip (python 2.7)

 

So tried pip3 -V:

***@***.***:~ $ pip3 -V

WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.

Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.

To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.

pip 22.3.1 from /home/pi/.local/lib/python3.7/site-packages/pip (python 3.7)

 

Installed updates as instructed then  python3 -m pip install pi-heif  which ran OK
 
 
Then installed picframe as instructed and it worked fine.
 
Thanks for your help.

Andrew

helgeerbe, 2022-12-13:
Hi Andrew,
Thanks for sharing the link. Pip seems to be a real pain.

closed 2022-12-13 by helgeerbe