helgeerbe/picframe #305: Newest updates stutter on pi zero w 2

clintiepoo, 2022-12-15:
I can only describe this in words - the newest 2022.12.04 and likely later updates cause my raspberry pi 0W2 to have choppy transitions. It will hold one picture open too long and then on the next skip past it. I assume it’s due to the pyheif changes. 

I rolled back to 2022.04.19 and didn’t experience this. 

Another note - I also am running a different instance on a pi 3b+ and it doesn’t have this issue. Maybe the 0W2 is just underpowered enough it can’t handle pyheif?

jgodfrey, 2022-12-15:
I guess I can't speak to any details here, but AFAIK `pyheif` is only used during the image caching process. So, if it is indeed a bottleneck, I'd expect it to be temporary - only while new images are being cached. Otherwise, the cached image data is just extracted from the DB for inclusion in the slide show - so I wouldn't expect any additional `pyheif` overhead at that point.

clintiepoo, 2022-12-15:
> I guess I can't speak to the recent changes specifically, but AFAIK `pyheif` is only used during the image caching process. So, if it is indeed a bottleneck, I'd expect it to be temporary - only while new images are being cached. Otherwise, the cached image data is just extracted from the DB for inclusion in the slide show - so I wouldn't expect any additional `pyheif` overhead at that point.

Maybe that’s not the bottleneck then - is there any other reason the performance would be worse than before?

helgeerbe, 2022-12-15:
The lib exifread was used to extract the meta data in the caching process for all types of images. This is done now with Pillow directly, because with the pi-heif Pillow plugin also heic images are supported bei Pillow.

The old optional pyheif lib was used to extract the raw image data from heic images. This is no longer needed because  the pi-heif Pillow plugin allows you to access all images via Pillow. No special treatment is needed anymore.

But yes, image processing for heic images is now done on a different code basis.

@clintiepoo do you have just heic/heif images? Do you have the same problems with jpg images? For jpg images nothing has changed so far I'm aware of. If there are performance issue, they should only occur with heic/heif images.

One other thing, that was change was the database access. The main thread (viewer thread) request a database lock, to write reverse geodata (if not already in the database) and update the statistics. But on my tests on my pi3 this only takes a couple of milliseconds. 

Would be interesting so see, in which part of the code, the time is consumed. Are you capable to do some code changes to dump timestamps? Ich could show you where in the code you have to do it.

clintiepoo, 2022-12-15:
> The lib exifread was used to extract the meta data in the caching process for all types of images. This is done now with Pillow directly, because with the pi-heif Pillow plugin also heic images are supported by Pillow.
> 
> The old optional pyheif lib was used to extract the raw image data from heic images. This is no longer needed because the pi-heif Pillow plugin allows you to access all images via Pillow. No special treatment is needed anymore.
> 
> But yes, image processing for heic images is now done on a different code basis.
> 
> @clintiepoo do you have just heic/heif images? Do you have the same problems with jpg images? For jpg images nothing has changed so far I'm aware of. If there are performance issues, they should only occur with heic/heif images.
> 
> One other thing, was changed. The main thread (viewer thread) request a database lock, to write reverse geodata (if not already in the database) and update the statistics. But tests on my pi3 shows that this takes only ms.
> 
> Would be interesting so see, in which part of the code, the time is consumed. Are you capable to do some code changes to dump timestamps? Ich could show you where in the code you have to do it.

I have a mix of jpg and heic images. I’m not sure which is causing the issue. 

If you can show me how to dump the logs or whatever, I can do that.

clintiepoo, 2022-12-16:
Here's a debug log. I redacted some info. 

[picframe.log](https://github.com/helgeerbe/picframe/files/10241994/picframe.log)

helgeerbe, 2022-12-18:
Thanks for the log.

The db access seems not to be the problem. Normally it takes around 1 ms to update the stats.

This is the longest period I found in the log.
````sh
2022-12-15 19:31:10,035 - image_cache.ImageCache - DEBUG - Update file stats: Wait for 1 ms and need 39 ms for update 
````

Can you run `top`. I'm wondering what the system load is.

On my pi 3B+ I see

```` sh
pi@picframe:~ $ top
top - 09:12:03 up 20 min,  2 users,  load average: 0,03, 0,12, 0,12
Tasks: 141 total,   1 running, 140 sleeping,   0 stopped,   0 zombie
%CPU(s):  1,8 us,  0,9 sy,  0,0 ni, 97,2 id,  0,0 wa,  0,0 hi,  0,1 si,  0,0 st
````

clintiepoo, 2022-12-18:
thanks - here’s what I get from top

```
top - 16:40:05 up 17 min,  3 users,  load average: 1.09, 1.02, 0.87
Tasks: 156 total,   3 running, 152 sleeping,   0 stopped,   1 zombie
%Cpu(s): 25.0 us,  1.8 sy,  0.0 ni, 72.9 id,  0.0 wa,  0.0 hi,  0.2 si,
MiB Mem :    364.5 total,     58.9 free,    228.7 used,     76.9 buff/c
MiB Swap:    200.0 total,    107.5 free,     92.5 used.     84.7 avail 
```

helgeerbe, 2022-12-19:
I don't have a zero w 2. What I read so far is, that it's from performance close to a 3B+.

But Cpu is utilized with 25%. Since it has 4 cores, this means 1 core it totally busy. Can you run `top`again and press `1`. This should show each core.

Can you press `P` to sort processes by CPU load. Which are your top processes?

How dow you start picframe? Boot to console and start picframe with an `xinit` call? Or do you boot into the graphical desktop? Without picframe running. What is your load?

sapnho, 2022-12-19:
I have a zero W 2. It's much weaker than a Pi 3. I experienced many reliability problems with it running Pi3D and the same with a few other auxiliary apps, so I removed it from [my list of recommended hardware](https://www.thedigitalpictureframe.com/how-to-add-crossfading-slide-transitions-to-your-digital-picture-frame-using-pi3d/).

helgeerbe, 2022-12-19:
Hi @sapnho. What would be the recommended settings in raspi_config for the zero?

I agree that the zero w2 seems not to be the best choice. But I'm wondering if we can tweak the settings a little bit, to bring the load below 1.

Default setting for frame rate is 20. @clintiepoo can you reduce the the frame rate in `configuration.yml`?

sapnho, 2022-12-19:
The settings are the same as with the Pi 3. The thing is that besides PictureFrame, there are other apps at work if you want to add a bit of functionality around it, and this is where the Pi Zero becomes unreliable.

clintiepoo, 2022-12-19:
Here's a `top + 1  + P`. I will try reducing the frame rate and see if that helps.

I'm starting picframe as a service - followed instructions [here](https://www.thedigitalpictureframe.com/how-to-add-crossfading-slide-transitions-to-your-digital-picture-frame-using-pi3d/).

I don't have any other services running on this Pi Zero W 2 - it's just for the picture frame. 

`top - 06:59:39 up 14:37,  3 users,  load average: 1.19, 1.03, 1.01
Tasks: 154 total,   1 running, 152 sleeping,   0 stopped,   1 zombie
%Cpu0  :  1.3 us,  0.7 sy,  0.0 ni, 97.3 id,  0.0 wa,  0.0 hi,  0.7 si,  0.0 st
%Cpu1  :  2.6 us,  1.6 sy,  0.0 ni, 95.7 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
%Cpu2  :  0.3 us,  1.0 sy,  0.0 ni, 98.7 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
%Cpu3  :  2.0 us,  1.0 sy,  0.0 ni, 97.1 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :    364.5 total,    170.9 free,     91.2 used,    102.4 buff/cache
MiB Swap:    200.0 total,     86.0 free,    114.0 used.    221.3 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                        
  813 pi        20   0  138940  35020   6528 S   5.6   9.4 432:16.23 picframe                                                                                                       
 6689 root      20   0       0      0      0 I   2.0   0.0   0:04.83 kworker/u8:2-brcmf_wq/mmc1:0001:1                                                                              
 6952 root      20   0       0      0      0 I   0.7   0.0   0:01.84 kworker/0:2-events                                                                                             
 7230 pi        20   0   10428   2908   2468 R   0.7   0.8   0:00.69 top                                                                                                            
   12 root      20   0       0      0      0 I   0.3   0.0   0:32.06 rcu_sched            `

clintiepoo, 2022-12-19:
> Hi @sapnho. What would be the recommended settings in raspi_config for the zero?
>
> I agree that the zero w2 seems not to be the best choice. But I'm wondering if we can tweak the settings a little [...]

Reducing the frame rate to 10fps doesn't fix the issue. I can watch one CPU go to 100% every time it switches images. It still stutters, particularly when It transitions to/from the portrait pairs, most of which are heic images for me. 

The thing is - it didn't do this before the update. If I go back to the previous version, the transitions and portrait pairs work fine. I'm confused what's causing it.

helgeerbe, 2022-12-19:
My best guess is, that the performance of the old lib is a little bit better, than of the new pi-heif lib.
Using the new lib drives the CPU utilization now over 100% and this results in the stuttering.

As a test. You could install a convert tool like `imagemagick` and convert all heic files with a simple command to jpg files. Does this work? Do you have still stutters?

clintiepoo, 2022-12-19:
> My best guess is, that the performance of the old lib is a little bit better, than of the new pi-heif lib. Using the new lib drives the CPU utilization now over 100% and this results in the [...]

I did this with a couple hundred files. The transitions seem better with JPG only files. One CPU still jumps to 100% with each image transition, but there's no stuttering/pausing. 

Is there any way to disable pyheic? I had heic files working previously by doing [this](https://www.thedigitalpictureframe.com/the-heic-photo-format-now-works-with-the-pi3d-digital-picture-frame-image-viewer/).

helgeerbe, 2022-12-19:
I'm afraid this is not possible at the moment. I removed the the old lib and all extra handling for heic images from the code.

The pi zero w2 seems not powerfull enough. And if you see with jpg images a CPU utilization over 100% I guess, any other further changes to the code the may result in stutter. At the moment I would say you have to stay with an older version on a pi zero.

If I have time, I can check, if I can find any performance improvements. Or the owner of the lib will improve performance in one of his next releases.

clintiepoo, 2022-12-19:
> I'm afraid this is not possible at the moment. I removed the the old lib and all extra handling for heic images from the code.
>
> The pi zero w2 seems not powerfull enough. And if you see with jpg [...]

Thanks - I’ll just roll back for now and watch the releases for something new. I’ll go ahead and close the issue.

closed 2022-12-19 by clintiepoo

clintiepoo, 2022-12-22:
@helgeerbe - just wanted to mention another fix. I overclocked the Pi Zero 2W to 1.4 GHz and this gives it enough resources to power through. I overclocked the CPU only - the GPU was giving me weird issues. 

These may not be permanent settings for me, but I wanted to share this approach in case it helps somebody. 

```
#Overclock 1400
arm_freq=1400
core_freq=550
over_voltage=6

# RAM settings
sdram_freq=550
over_voltage_sdram=1
```

bigcat88, 2022-12-28:
@clintiepoo just interesting, with what version of pi-heif did you test? was it 0.9.0 version?

clintiepoo, 2023-01-07:
> @clintiepoo just interesting, with what version of pi-heif did you test? was it `0.9.0` version?

I installed pyheif with the large set of commands [on this page](https://www.thedigitalpictureframe.com/the-heic-photo-format-now-works-with-the-pi3d-digital-picture-frame-image-viewer/). 

When I do a ```pip show pyheif``` I get: 

Version: 0.6.1

sapnho, 2023-01-07:
Yes, it seems like a lot has changed with pyheif. Happy to update my blog page if anyone can point me in the right direction.