helgeerbe/picframe #308: Questions on show_text_tm

dnuber, 2023-01-14:
Just want to say thanks for this great project software. I had been looking for a way to establish a digital picture frame and this really fit the bill!  Now have two installations in place.

Couple of questions...

- With the more recent versions and new configuration file ... I'm trying to figure out the "show_text_tm" option. Is this meant to be in seconds? If set to 60, the text displays for waaaay longer than 1 minute (was displayed for full hour that image was displayed). If I set to 1.0, the text displays for about 20 seconds.
- When the text first comes up, it is really bright (for like a second or 2) but then becomes semi-transparent. Sometimes the text disappears completely.
- Also, Control-C does not seem to end the slideshow any more? Or maybe the Pi was in some kind of weird state where it was not accepting keyboard input. Looks like the "exit" option (under peripherals) is now meant to handle this function?

Perhaps I've got something screwed up on my end, with respect to the above. Appreciate any suggestions/pointers. 

Dave

helgeerbe, 2023-01-16:
Hi @dnuber,

that sounds strange. `show_text_tm`should be indeed seconds. It should be less than `time_delay`.
After the image is fully faded in, the overlay text should be displayed for `show_text_tm` seconds. Text itself is faded in and out. 

Can you provide the following infos:
- What kind of raspberry? 
- raspian buster or bullesyes?
- Python version `python3 -V`
- picframe version and version of its used packages `picframe -v`
- Your `configuration.yaml`(please remove any passwords)

How do you start picframe? From console or as systems service? From console Control-C should work. What is reported on console, if you press Control-C?

Control-C is just for debugging, if you run picframe from console. Normally you should run picframe as systemd service and start stop it with `systemctl start...` or `systemctl stop`

The peripherals interface could be used, if you have a touchscreen and or mouse/keyboard attached. This code is not well tested by me, because I don't use any peripherals on my own frame. And I never got the pygame engine working on my Mac, which I use for developement. If there are any issues please let me know.

By the way. Can you run top? Maybe you have, for whatever reason a high load on your raspberry.

dnuber, 2023-01-16:
Hi @helgeerbe 
Thanks for the reply. Here is requested info:

Raspberry Pi 3 Model B Rev 1.2
Buster "legacy" (just freshly installed)
Python 3.7.3
picframe version:  2022.12.15

[configuration.yaml.txt](https://github.com/helgeerbe/picframe/files/10427587/configuration.yaml.txt)

I don't think the load is particularly high for this unit. It is used exclusively for picframe and it runs as systemd automatically on startup, per instructions here:
https://www.thedigitalpictureframe.com/how-to-add-crossfading-slide-transitions-to-your-digital-picture-frame-using-pi3d/

Here is top output:
Tasks: 143 total,   3 running, 140 sleeping,   0 stopped,   0 zombie
%Cpu(s): 26.8 us,  8.9 sy,  0.0 ni, 62.3 id,  0.0 wa,  0.0 hi,  2.0 si,  0.0 st
MiB Mem :    744.8 total,    261.0 free,    197.6 used,    286.2 buff/cache
MiB Swap:    100.0 total,    100.0 free,      0.0 used.    479.8 avail Mem 

Hope this helps?

Cheers,
Dave

helgeerbe, 2023-01-16:
The config file looks fine for my. text_tm is set to 2.0. If I remember right textfading takes in minimum 5 sec. Any number below 5 is set internally to 5.

I run python 3.9.2. But I'm not aware, that I use something, which requires a newer version.

Can you run `picframe -v`

I get:
````sh
pi@picframe:~ $ picframe -v
INFO:start.py:starting ['/home/pi/.local/bin/picframe', '-v']
picframe version:  0+untagged.445.gf9d1065.dirty

Checking required packages......
PIL :  9.3.0
pi3d :  2.49
yaml :  6.0
paho.mqtt :  1.6.1
iptcinfo3 :  2.1.4
numpy :  1.19.5
ninepatch : installed, but no version info
pi_heif :  0.9.0
defusedxml :  0.7.1
````

You missed the load from top. Can you share it. Your CPU load seems to high for me. Wondering what your average load is.

````sh
top - 20:13:29 up  6:13,  2 users,  load average: 0,14, 0,19, 0,18
Tasks: 143 total,   1 running, 142 sleeping,   0 stopped,   0 zombie
%CPU(s):  1,6 us,  0,9 sy,  0,0 ni, 97,5 id,  0,0 wa,  0,0 hi,  0,0 si,  0,0 st
MiB Spch:    734,0 total,    102,2 free,    340,4 used,    291,4 buff/cache
MiB Swap:    100,0 total,     72,2 free,     27,8 used.    332,9 avail Spch

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     ZEIT+ BEFEHL
    885 pi        20   0  544728 124296  17604 S   6,6  16,5  29:21.43 python3
````

Just to be sure: on pi3 you have set in raspi-config do_memory_split to 256  and A2 GL Driver to Choose G1 (Legacy)?

dnuber, 2023-01-17:
OK - thanks for the info on 5 sec minimum. Looks like trying to use text_tm <5s was causing funky stuff with text fading/opacity. I have reverted to text_tm of 20.0 and delay between images of 5 minutes and it seems to be behaving as expected. Still not sure why I was originally getting text that did not go away even though text_tm was default of 20.0. Will let you know if I run into any other problems.

For completeness ... confirm that gpu mem is 256 and A2 GL Driver is disabled (Legacy).

And:
pi@picframe-wembley:~ $ picframe -v
INFO:start.py:starting ['/home/pi/.local/bin/picframe', '-v']
picframe version:  2022.12.15

Checking required packages......
PIL :  9.4.0
pi3d :  2.49
yaml :  6.0
paho.mqtt :  1.6.1
iptcinfo3 :  2.1.4
numpy :  1.16.2
ninepatch : installed, but no version info
pi_heif :  0.9.1
defusedxml :  0.7.1
pi@picframe-wembley:~ $ 

Perhaps load is higher because I am running VNC (and remotely connected while running top).  But still not maxed out so could that really cause text timing issues?

top - 14:37:43 up 59 min,  2 users,  load average: 0.90, 0.71, 0.66
Tasks: 142 total,   1 running, 141 sleeping,   0 stopped,   0 zombie
%Cpu(s): 10.7 us,  5.1 sy,  0.0 ni, 83.5 id,  0.0 wa,  0.0 hi,  0.7 si,  0.0 st
MiB Mem :    744.8 total,    299.4 free,    190.8 used,    254.6 buff/cache
MiB Swap:    100.0 total,    100.0 free,      0.0 used.    488.4 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND    
  469 root      20   0   48848  33056  17616 S  17.1   4.3   0:08.59 vncserver+ 
  504 root      20   0  217020  59532  32168 S  11.8   7.8   0:24.08 Xorg       
  912 pi        20   0  236328  49784  38300 S  11.8   6.5   0:12.52 lxterminal 
  543 pi        20   0  110024  47800  15632 S   6.6   6.3   3:17.75 picframe   
  967 root      20   0       0      0      0 I   6.6   0.0   0:01.13 kworker/u+ 
  977 root      20   0       0      0      0 D   5.3   0.0   0:00.78 kworker/0+ 
  511 root      20   0   19576  12404  11924 S   3.3   1.6   0:12.69 vncagent   
  933 pi        20   0   10440   2848   2468 R   1.0   0.4   0:20.30 top        
  962 root      20   0       0      0      0 I   1.0   0.0   0:10.31 kworker/0+ 
   12 root      20   0       0      0      0 I   0.3   0.0   0:00.94 rcu_sched  
   73 root       1 -19       0      0      0 S   0.3   0.0   0:02.23 vchiq-rec+ 
  961 root      20   0       0      0      0 I   0.3   0.0   0:00.36 kworker/u+ 
  976 root      20   0       0      0      0 I   0.3   0.0   0:00.01 kworker/1+ 
    1 root      20   0   33720   8024   6420 S   0.0   1.1   0:05.90 systemd    
    2 root      20   0       0      0      0 S   0.0   0.0   0:00.02 kthreadd 

Thanks again for your responses and insight.

Dave

paddywwoof, 2023-01-17:
Dave, this does seem odd. The [code](https://github.com/helgeerbe/picframe/blob/dab5ecf787c34212c6e06af293cfbf9faa7f2668/src/picframe/viewer_display.py#L540) uses the greater of 4.0s or show_text_tm/4.0, but it should cope if you put in a value lower than 5s. There is probably some kind of bug in the code as it's fairly convoluted (alpha going from 0 to 1, staying there for a bit then dropping to 0. But having to cope with images fading, loading etc), I will will play around later if I get a chance.

Paddy

helgeerbe, 2023-02-20:
@paddywwoof Any update on this?

closed 2026-05-06 by github-actions[bot]