helgeerbe/picframe #310: PictureFrame.py kompiliert mit pyinstaller. Executable file liefert Error

Imre-maker, 2023-01-23:
Leider beim starten des ausführbaren PictureFrame objektes bekomme ich Fehler:
WARNING:pi3d.Display:create Display with (...use_glx=True) for transparent background on x11 window. libGLX needs to be available
Traceback (most recent call last):
  File "pi3d/Shader.py", line 192, in _load_shader
  File "pkg_resources/__init__.py", line 1167, in resource_string
  File "pkg_resources/__init__.py", line 1412, in get_resource_string
  File "pkg_resources/__init__.py", line 1579, in _get
  File "PyInstaller/loader/pyimod02_importers.py", line 344, in get_data
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/_MEI4GYMew/pi3d/shaders/std_head_vs.inc'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "PictureFrame_2MI_My.py", line 1194, in <module>
  File "pi3d/Shader.py", line 149, in create
  File "pi3d/Shader.py", line 110, in __init__
  File "pi3d/Shader.py", line 97, in make_shader
  File "pi3d/Shader.py", line 199, in _load_shader
  File "pi3d/Shader.py", line 195, in _load_shader
FileNotFoundError: [Errno 2] No such file or directory: 'std_head_vs.inc'
[6566] Failed to execute script 'PictureFrame_2MI_My' due to unhandled exception!

Ich wäre unendlich dankbar wenn jemand mir helfen könnte

helgeerbe, 2023-01-23:
@Imre-maker Kann es sein, dass du im falschen Repository bist? Du versuchst PictureFrame aus dem Demo Repository zu starten (https://github.com/pi3d/pi3d_demos).

Bei picframe hier handelt es sich um eine komplett neu geschriebenen Anwendung, welche die pi3d Bibliothek benutzt.

closed 2023-01-23 by Imre-maker

Imre-maker, 2023-01-24:
Heureka!
Die Lösung war einfach. Beim kompilieren mit pyinstaller sollte ich extra folgendes an die Optionen anhängen: --collect -all pi3d damit alle submodule und attribute... eingebunden werden.