helgeerbe/picframe #320: buster - Fresh install error on launch TypeError: 'type' object is not subscriptable

jamespreedy, 2023-05-16:
Hi folks, hope you're doing well.

I've clean installed my Raspberry Pi 4 2gb this evening, and can't get PicFrame to launch.
My config is default, no changes since running `picframe -i .`

I've included the error, plus my config file and my system versions below - please let me know if I can get you any additional information. FYI, my username isn't pi - I did a find and replace into the output below.

Thanks in advance for your help!

James.

```

$ sudo xinit /usr/bin/python3 /home/pi/picframe_data/run_start.py /home/pi/picframe_data/config/configuration.yaml

X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 5.4.0-109-generic aarch64 Debian
Current Operating System: Linux picframe 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60D smsc95xx.macaddr=DC:A6:32:3A:59:9F vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=96f6dc32-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Build Date: 04 April 2023  07:50:56AM
xorg-server 2:1.20.4-1+rpt4+deb10u9 (https://www.debian.org/support)
Current version of pixman: 0.36.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Tue May 16 22:10:58 2023
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(II) modeset(0): Initializing kms color map for depth 24, 8 bpc.
INFO:start.py:starting ['/home/pi/picframe_data/run_start.py', '/home/pi/picframe_data/config/configuration.yaml']
INFO:model.Model:Open config file: /home/pi/picframe_data/config/configuration.yaml:
Traceback (most recent call last):
  File "/home/pi/picframe_data/run_start.py", line 2, in <module>
    start.main()
  File "/usr/local/lib/python3.7/dist-packages/picframe/start.py", line 136, in main
    c.start()
  File "/usr/local/lib/python3.7/dist-packages/picframe/controller.py", line 330, in start
    from picframe.interface_peripherals import InterfacePeripherals
  File "/usr/local/lib/python3.7/dist-packages/picframe/interface_peripherals.py", line 17, in <module>
    class InterfacePeripherals:
  File "/usr/local/lib/python3.7/dist-packages/picframe/interface_peripherals.py", line 29, in InterfacePeripherals
    model: type[Model],
TypeError: 'type' object is not subscriptable```

- Hardware:
- OS: Debian buster - Linux picframe 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64
- Python: Python 3.7.3
- picframe (run `picframe -v` if possible): 

```$ picframe -v
INFO:start.py:starting ['/usr/local/bin/picframe', '-v']
picframe version:  2022.12.15post6

Checking required packages......
PIL :  9.5.0
pi3d :  2.49
yaml :  6.0
paho.mqtt :  1.6.1
iptcinfo3 :  2.1.4
numpy :  1.16.2
ninepatch : installed, but no version info
pi_heif :  0.11.1
defusedxml :  0.7.1```

```yaml
$ cat configuration.yaml
viewer:
[... 108 lines cut ...]
  certfile: "path/to/cert.pem"            # server certificate

peripherals:
  input_type: null                        # default=null, valid options: {null, "keyboard", "touch", "mouse"}
  buttons:
    pause:                                # pause/unpause the show
      enable: True                        # default=True
      label: "Pause"                      # default="Pause"
      shortcut: " "                       # default=" "
    display_off:                          # turn off the display (when off, any input from selected peripheral will turn it back on)
      enable: True                        # default=True
      label: "Display off"                # default="Display off"
      shortcut: "o"                       # default="o"
    location:                             # shows or hides location information
      enable: False                       # default=False
      label: "Location"                   # default="Location"
      shortcut: "l"                       # default="l"
    exit:                                 # exit PictureFrame
      enable: False                       # default=False
      label: "Exit"                       # default="Exit"
      shortcut: "e"                       # default="e"
    power_down:                           # power down the device, uses sudo
      enable: False                       # default=False
      label: "Power down"                 # default="Power down"
      shortcut: "p"                       # default="p"```

paddywwoof, 2023-05-16:
Hi, if it's any consolation I can't run picframe for the same reason. It should be something we can figure out quite quickly.On 16 May 2023 22:23, James Preedy ***@***.***> wrote:
Hi folks, hope you're doing well.
I've clean installed my Raspberry Pi 4 2gb this evening, and can't get PicFrame to launch.
My config is default, no changes since running picframe -i .
I've included the error, plus my config file and my system versions below - please let me know if I can get you any additional information. FYI, my username isn't pi - I did a find and replace into the output below.
Thanks in advance for your help!
James.

$ sudo xinit /usr/bin/python3 /home/pi/picframe_data/run_start.py /home/pi/picframe_data/config/configuration.yaml

X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 5.4.0-109-generic aarch64 Debian
Current Operating System: Linux picframe 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 ***@***.*** smsc95xx.macaddr=DC:A6:32:3A:59:9F vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=96f6dc32-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Build Date: 04 April 2023  07:50:56AM
xorg-server 2:1.20.4-1+rpt4+deb10u9 (https://www.debian.org/support)
Current version of pixman: 0.36.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Tue May 16 22:10:58 2023
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(II) modeset(0): Initializing kms color map for depth 24, 8 bpc.
INFO:start.py:starting ['/home/pi/picframe_data/run_start.py', '/home/pi/picframe_data/config/configuration.yaml']
INFO:model.Model:Open config file: /home/pi/picframe_data/config/configuration.yaml:
Traceback (most recent call last):
  File "/home/pi/picframe_data/run_start.py", line 2, in <module>
    start.main()
  File "/usr/local/lib/python3.7/dist-packages/picframe/start.py", line 136, in main
    c.start()
  File "/usr/local/lib/python3.7/dist-packages/picframe/controller.py", line 330, in start
    from picframe.interface_peripherals import InterfacePeripherals
  File "/usr/local/lib/python3.7/dist-packages/picframe/interface_peripherals.py", line 17, in <module>
    class InterfacePeripherals:
  File "/usr/local/lib/python3.7/dist-packages/picframe/interface_peripherals.py", line 29, in InterfacePeripherals
    model: type[Model],
TypeError: 'type' object is not subscriptable```

- Hardware:
- OS: Debian buster - Linux picframe 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64
- Python: Python 3.7.3
- picframe (run `picframe -v` if possible): 

```$ picframe -v
INFO:start.py:starting ['/usr/local/bin/picframe', '-v']
picframe version:  2022.12.15post6

Checking required packages......
PIL :  9.5.0
pi3d :  2.49
yaml :  6.0
paho.mqtt :  1.6.1
iptcinfo3 :  2.1.4
numpy :  1.16.2
ninepatch : installed, but no version info
pi_heif :  0.11.1
defusedxml :  0.7.1```

```yaml
$ cat configuration.yaml
viewer:
  blur_amount: 12                         # default=12,  larger values than 12 will increase processing load quite a bit
  blur_zoom: 1.0                          # default=1.0, must be >= 1.0 which expands the backgorund to just fill the space around the image
  blur_edges: False                       # default=False, use blurred version of image to fill edges - will override FIT = False
  edge_alpha: 0.5                         # default=0.5, background colour at edge. 1.0 would show reflection of image
  fps: 20.0                               # default=20.0
[... 105 lines cut ...]
peripherals:
  input_type: null                        # default=null, valid options: {null, "keyboard", "touch", "mouse"}
  buttons:
    pause:                                # pause/unpause the show
      enable: True                        # default=True
      label: "Pause"                      # default="Pause"
      shortcut: " "                       # default=" "
    display_off:                          # turn off the display (when off, any input from selected peripheral will turn it back on)
      enable: True                        # default=True
      label: "Display off"                # default="Display off"
      shortcut: "o"                       # default="o"
    location:                             # shows or hides location information
      enable: False                       # default=False
      label: "Location"                   # default="Location"
      shortcut: "l"                       # default="l"
    exit:                                 # exit PictureFrame
      enable: False                       # default=False
      label: "Exit"                       # default="Exit"
      shortcut: "e"                       # default="e"
    power_down:                           # power down the device, uses sudo
      enable: False                       # default=False
      label: "Power down"                 # default="Power down"
      shortcut: "p"                       # default="p"```

—Reply to this email directly, view it on GitHub, or unsubscribe.You are receiving this because you are subscribed to this thread.Message ID: ***@***.***>

jamespreedy, 2023-05-16:
@paddywwoof - Least its not just me! Thanks 😀

helgeerbe, 2023-05-17:
OK, I installed python 3.7 through miniconda and got the same error. Had to learn that the ability to use the [] operator on types for type hinting was added in 3.9. Wondering, why this wasn't observed before.

So I see two possibilities. We declare python 3.9 as minimum python release, or we remove the type hinting. 

Btw. I played around to use sanic as new webserver, which also requires at least 3.9. sanic runs in a separate process, so one can use the other cores of the pi.

dexn, 2023-05-17:
Personally I like the idea of python 3.9 but others might not like the extra steps involved.

jamespreedy, 2023-05-17:
I broke my instance trying to install Python 3.9 last night 🤣
I'm reinstalling buster from scratch and will try with 2023.05.17 and let you know.

Thank you so much for your help!

jamespreedy, 2023-05-17:
@helgeerbe - works great, thank you so much for your help!
Really appreciate all your efforts. 🍺

helgeerbe, 2023-05-17:
Glad, that it works.

closed 2023-05-17 by helgeerbe