helgeerbe/picframe #326: AttributeError: module 'PIL.Image' has no attribute 'ANTIALIAS'

junglebells, 2023-07-19:
**Describe the bug**
Fresh install of picframe results into this error:
`AttributeError: module 'PIL.Image' has no attribute 'ANTIALIAS'`

**Additional context**
Pillow 10.0.0 is getting installed. It seems ANTIALIAS has been deprecated in this version. See [here](https://pillow.readthedocs.io/en/stable/releasenotes/10.0.0.html#constants).

Please help.

helgeerbe, 2023-07-26:
Sorry for the late response and thanks for reporting.

Indeed in ninpatch I found  the standard filter defined as`render(self, width, height, img_filter=Image.ANTIALIAS, cache=False)`.

As a workaround I replaced all render calls with `render(image.width, image.height,  Image.Resampling.LANCZOS)` and changed the minimum Pillow version to 10.0.0.

I will prepare a new release.

helgeerbe, 2023-07-26:
Should be fixed with 2023.07.26

helgeerbe, 2023-07-26:
On my test system (Mac) it works, but on the raspberry it seem we have to pin Pillow to release 9.5.0 until a fix for ninepatch is available.

I released 2023.07.26.post1

junglebells, 2023-07-26:
Thanks @helgeerbe, appreciate your help with this.

closed 2023-07-26 by junglebells