helgeerbe/picframe #327: Run as Service does not show pictures

peegee1412, 2023-09-14:
Hi picframe-friends

i run a Buster installation and when i start picframe as a service it seems to start (manual start works and shows pictures, while service is running i cant start manual).
The service seems to run successfully but does not show anything - the desktop just remains as it is. Any idea what i am doing wrong?

helgeerbe, 2023-09-14:
Hi, 

what Pi you are using. 3 or 4?

There are multiple ways to start picframe as a service. When I understand you right, you boot to desktop and try to start a service?

Can you show us your script?

peegee1412, 2023-09-14:
Hi helgeerbe,

i use a 3 and copied the described content of the script:
[Unit]
Description=PictureFrame on Pi3

[Service]
ExecStart=/home/pi/.local/bin/picframe ~/picframe_data/config/configuration.yaml
Restart=always

[Install]
WantedBy=default.target

It also looks like it starts successfull and is indicated as running (status request). Also after a reboot is indicated as active.

But unfortenately the monitor only starts presenting when stoping the service and manually starting the picframe.

I fear something in the base settings or in the user setup might be it? (user kept as 'pi'; no huge changes in system done...)

and yes - buster with desktop in use. When starting the service and ask for the status the start sequence shows the typical 2 lines which i think indicate that the script itself runs.

One more test which might help: I activated http and can reach the page when i start the script manual. If started with the service it is just indicating to be started - incl. the two lines "starting...2 and open config... but monitor and http not apearing...

peegee1412, 2023-09-14:
by the way - thank you for responding that fast! Really appreciate this - preparing the project as apresent (or at least trying :-))

helgeerbe, 2023-09-14:
There is good description how to run picframe with the same setup on pi3 and pi4: https://gist.github.com/thomasedoff/3973846101da1e7496524be075f1c02f

But you already setup your pi. I would propose you should change 

````bash
sudo raspi-config nonint do_boot_behaviour B4 && sudo raspi-config nonint do_memory_split 256 && sudo raspi-config 
````

In the raspi-config module, go to 6 Advanced Options > A2 GL Driver > Choose G1 (Legacy)

Try from the console to run picframe:

````bash
picframe ~/picframe_data/config/configuration.yaml
````

Does is start?

After reboot, does your service work?

peegee1412, 2023-09-14:
I followed all your changes - GL driver resonds with not active

When starting from console with "picframe ..." ist starts and performs as expected. It also indicates in the service that it is started but - no picture is seen and no http server behaviour seen.

puhh, seems like so close - only service does not bring it up

peegee1412, 2023-09-14:
Here we go: Fatal server error - server is already active for display 0 via the service?!

helgeerbe, 2023-09-14:
If you run from the console, than you have to stop the service first. But you say from the console it worked. 

Then it would be good to get the logfile. When the service is running. Run
````cmd
journalctl --user-unit picframe.service
````

To see the logfile. Anything of interest. You can change the loglevel in configuration.yml

peegee1412, 2023-09-14:
Thanks helgeerbe, you made my day!

I re-started with your shared alternative description step by step and got it running perfectly!! (re-adjusted a few configs already in the picframe + different setting for samba, but working. thinking if i want once the home-assistant seen or going to bed :-))

There is good description how to run picframe with the same setup on pi3 and pi4: https://gist.github.com/thomasedoff/3973846101da1e7496524be075f1c02f

Thanks that you supported me - that helped jumping further on and ... finally got it running! Very very happy! well done!

closed 2023-10-05 by helgeerbe