helgeerbe/picframe #329: 2023.07.26.post1 not working with Pillow 10.0.0

helgeerbe, 2023-10-05:
Pillow versions before v10.0.1 bundled libwebp binaries in wheels that are vulnerable to https://github.com/advisories/GHSA-hhrh-69hc-fgg7 (previously https://github.com/advisories/GHSA-j7hp-h8jx-5ppr). Pillow v10.0.1 upgrades the bundled libwebp binary to v1.3.2.

Unfortunately there are some functions removed in Pillow 10.0.0
https://pillow.readthedocs.io/en/stable/deprecations.html

1. ninepatch from pypi is not updated with the actual repo code as a workaround change `pyproject.toml`to
  `"ninepatch@git+https://github.com/vindolin/ninepatch",`to install code from github not from pypi
  https://github.com/vindolin/ninepatch/issues/6
2. update pi3d to get rid of  ImageFont.getsize() (see issue https://github.com/tipam/pi3d/issues/262)

paddywwoof, 2023-10-05:
Yes, you will see that I have pushed pi3d to v2.50 However I am locked out of PYPI so can't update the pip source!! I must have replaced my phone since I last updated anything on PYPI and don't have a recovery option... so I have to push a randomly named branch to picframe and wait a few weeks.

paddywwoof, 2023-10-05:
Helge, that would be great short term. I'm away from home for a week yet so I can't update PYPI till I'm back at a computer.

As mentioned earlier, I expect there might be a few mods over the coming month with changes following RPi5. There may be more examples for getting OpenGL without X that I can work from...

helgeerbe, 2023-10-05:
I created a pr.

If I run locally `pip install -e . --upgrade` it seems to work.

helgeerbe, 2024-02-19:
should be fixed with 2024.02.19

PR #330 (2023-10-05): Make use of Pillow >= 10.0.1
Should fix #329
closed 2024-02-19 by helgeerbe