helgeerbe/picframe #332: Has vcgencmd display_power stopped working?

sapnho, 2023-10-08:
I did a regular "sudo apt update && sudo apt upgrade -y" yesterday, and since then the "vcgencmd display_power 0" command doesn't turn off the HDMI output anymore.

Has anyone else come across this and has found a fix to turn the frame off and on via the command line?

Running Buster, latest update.

helgeerbe, 2023-10-09:
Hi Wolfgang,

I don't know, if this was done on purpose. From bulleseye on, "vcgencmd display_power" is no longer supported.
You have to use `xset` instead.

You can give it a try on buster also.

configuration.yaml
````yml
display_power: 1                        # default=0. choices={0, 1}, 0 will use legacy `vcgencmd` and 1 will use `xset` to blank the display
````

picframe.service
```ìni
[Unit]
Description=PictureFrame

[Service]
ExecStart=xinit /home/pi/start_picframe.sh
Restart=always

[Install]
WantedBy=default.target
````

start_picframe.sh
````sh
#!/bin/sh
#
# ~/.xinitrc
#
xset -display :0 dpms 0 0 0 &
xset -display :0 s off &
exec /usr/bin/python3 /home/pi/.local/bin/picframe /home/pi/picframe_data/config/configuration.yaml
````

sapnho, 2023-10-12:
I can't find 

`display_power: 1                        # default=0. choices={0, 1}, 0 will use legacy `vcgencmd` and 1 will use `xset` to blank the display` 

in the configuration.yaml.

Or am I missing something?

sapnho, 2023-10-12:
I tried 

`xset -display :0 s off`

in the Terminal but it doesn't shut down the system.

The solution that I found was to use 

`DISPLAY=:0.0 xrandr --output HDMI-1 --off` to turn it off and

`DISPLAY=:0.0 xrandr --output HDMI-1 --mode 3840x2160 --rate 30` to turn my 4K screen back on again.

sapnho, 2023-10-14:
I summarized it in an article: https://www.thedigitalpictureframe.com/xrandr-the-new-command-to-turn-your-connected-monitor-on-and-off/

sapnho, 2023-10-14:
And I found a cool way to check the display status: https://forums.raspberrypi.com/viewtopic.php?p=2144655#p2144655

sapnho, 2024-11-03:
I guess this issue can be closed.

closed 2024-11-03 by helgeerbe