helgeerbe/picframe #333: Raspberry Pi Buster no longer available for download

sapnho, 2023-10-12:
I have just been contacted by a blog reader who told me that Raspberry Pi Buster is no longer available for download.

I will start experimenting with Bookworm on a Pi4 this weekend to see what still works and what might be broken in Pi3D PictureFrame.

sapnho, 2023-10-12:
Ok, I have started the check. Downloaded and installed Bookworm with Desktop on a Pi 4. So far so good. Screen Blanking is D4 now (was D2 before) but that doesn't matter.

Heading over to my [blog article](https://www.thedigitalpictureframe.com/how-to-add-crossfading-slide-transitions-to-your-digital-picture-frame-using-pi3d/) on the installation of Pi3D. 

`sudo raspi-config nonint do_boot_behaviour B2 && sudo raspi-config nonint do_memory_split 512 && sudo raspi-config
` needs to have the `do_memory_split 512` removed. Can't find that anymore in the raspi config menu. Maybe this is automatic now?

When I then enter the first installation command

`pi@pi4test:~ $ sudo python3 -m pip install -U pip`

I get

`error: externally-managed-environment`

`× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.`
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

`note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.`

sapnho, 2023-10-12:
So, I guess, there's some work to do to get Pi3D running on Bookworm...

sapnho, 2023-10-15:
So, it seems like Python packages in Bookworm must now be installed in a virtual environment.

I did
`python3 -m venv myenv` and`source myenv/bin/activate` which then allows me to `pip install picframe`.

This resulted in `Successfully installed IPTCInfo3-2.1.4 Pillow-9.5.0 PyYAML-6.0.1 defusedxml-0.7.1 ninepatch-0.1.22 numpy-1.26.1 paho-mqtt-1.6.1 pi-heif-0.13.0 pi3d-2.49 picframe-2023.7.26.post1
`
So far, so good.

sapnho, 2023-10-15:
Entering `picframe -i . ` however, then results in a mouthful.

Traceback (most recent call last):
  File "/home/pi/myenv/lib/python3.11/site-packages/numpy/core/__init__.py", line 24, in <module>
    from . import multiarray
  File "/home/pi/myenv/lib/python3.11/site-packages/numpy/core/multiarray.py", line 10, in <module>
    from . import overrides
  File "/home/pi/myenv/lib/python3.11/site-packages/numpy/core/overrides.py", line 8, in <module>
    from numpy.core._multiarray_umath import (
ImportError: libopenblas.so.0: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/myenv/lib/python3.11/site-packages/numpy/__init__.py", line 130, in <module>
    from numpy.__config__ import show as show_config
  File "/home/pi/myenv/lib/python3.11/site-packages/numpy/__config__.py", line 4, in <module>
    from numpy.core._multiarray_umath import (
  File "/home/pi/myenv/lib/python3.11/site-packages/numpy/core/__init__.py", line 50, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.11 from "/home/pi/myenv/bin/python3"
  * The NumPy version is: "1.26.1"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libopenblas.so.0: cannot open shared object file: No such file or directory

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/myenv/bin/picframe", line 5, in <module>
    from picframe.start import main
  File "/home/pi/myenv/lib/python3.11/site-packages/picframe/start.py", line 8, in <module>
    from picframe import model, viewer_display, controller, __version__
  File "/home/pi/myenv/lib/python3.11/site-packages/picframe/viewer_display.py", line 1, in <module>
    import pi3d
  File "/home/pi/myenv/lib/python3.11/site-packages/pi3d/__init__.py", line 19, in <module>
    from pi3d.Buffer import Buffer
  File "/home/pi/myenv/lib/python3.11/site-packages/pi3d/Buffer.py", line 4, in <module>
    import numpy as np
  File "/home/pi/myenv/lib/python3.11/site-packages/numpy/__init__.py", line 135, in <module>
    raise ImportError(msg) from e
ImportError: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.
(myenv) pi@pi4test:~ $ picframe -i .
Traceback (most recent call last):
  File "/home/pi/myenv/lib/python3.11/site-packages/numpy/core/__init__.py", line 24, in <module>
    from . import multiarray
  File "/home/pi/myenv/lib/python3.11/site-packages/numpy/core/multiarray.py", line 10, in <module>
    from . import overrides
  File "/home/pi/myenv/lib/python3.11/site-packages/numpy/core/overrides.py", line 8, in <module>
    from numpy.core._multiarray_umath import (
ImportError: libopenblas.so.0: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/myenv/lib/python3.11/site-packages/numpy/__init__.py", line 130, in <module>
    from numpy.__config__ import show as show_config
  File "/home/pi/myenv/lib/python3.11/site-packages/numpy/__config__.py", line 4, in <module>
    from numpy.core._multiarray_umath import (
  File "/home/pi/myenv/lib/python3.11/site-packages/numpy/core/__init__.py", line 50, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.11 from "/home/pi/myenv/bin/python3"
  * The NumPy version is: "1.26.1"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libopenblas.so.0: cannot open shared object file: No such file or directory

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/myenv/bin/picframe", line 5, in <module>
    from picframe.start import main
  File "/home/pi/myenv/lib/python3.11/site-packages/picframe/start.py", line 8, in <module>
    from picframe import model, viewer_display, controller, __version__
  File "/home/pi/myenv/lib/python3.11/site-packages/picframe/viewer_display.py", line 1, in <module>
    import pi3d
  File "/home/pi/myenv/lib/python3.11/site-packages/pi3d/__init__.py", line 19, in <module>
    from pi3d.Buffer import Buffer
  File "/home/pi/myenv/lib/python3.11/site-packages/pi3d/Buffer.py", line 4, in <module>
    import numpy as np
  File "/home/pi/myenv/lib/python3.11/site-packages/numpy/__init__.py", line 135, in <module>
    raise ImportError(msg) from e
ImportError: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.

paddywwoof, 2023-10-15:
Forcing virtual env seems like a big departure from previously. There will need to be a massive rewrite of documentation all over the place, not just for picframe and pi3d!

sapnho, 2023-10-15:
Could tools like Replin help in the conversion? I was generating and testing several Python scripts yesterday to see how well they work, and I was blown away by the result.

Also, just checked out Replin, which is like Copilot but apparently a lot better. It even has a gree option.

sapnho, 2023-10-15:
> Forcing virtual env seems like a big departure from previously. There will need to be a massive rewrite of documentation all over the place, not just for picframe and pi3d!

What is the biggest impact of this virtual environment?

paddywwoof, 2023-10-15:
Virtual envs good for doing different things with a computer (especially different dependencies or python versions). However for a dedicated application like picframe they're an extra layer of complexity.

The main issue will be all the documentation and examples on line where ```pip install x``` stops working

sapnho, 2023-10-15:
Hmm, that's a bummer. I guess then at the moment, there is no way to use Pi3D with the current RPI OS anymore.

It's astonishing how many breaking changes OS changes introduce.

paddywwoof, 2023-10-16:
It doesn't seem too bad. There might be one or two things needing to be fine tuned, but I have just done this and it all seems OK:
```
1. Put bullseye on an SD card then stick it in an RPi4
2. Log in, open a terminal and git clone pi3d and pi3d_demos
3. cd to pi3d_demos and python3 RunTests.py
```
Not sure about FPS but didn't look too bad, obviously should have set GPU driver, memory and all the other fine details, but it seems a positive start.

Paddy

paddywwoof, 2023-10-17:
Hmm. As you say, these OS upgrades are full of breaking changes.

I've tried pip installing pi3d and, as you discovered, this has now become non-trivial. I had to use `venv` and had to specify source on github: `python -m pip install git+https://github.com/tipam/pi3d` having gradually discovered that `python` has changed from being an alias for `python2` to `python3`. However the dependency in `pi3d.setup.py` for numpy downloads and installs version `1.26.1` although `1.24` is already installed with  _bookworm_. And the pip installed version of numpy is missing the compiled libraries for blas... Surely this problem will have happened with lots of other people! Anyway, I think we need to specify the debian `python3-numpy` rather than whatever pip dredges up!

PS Some progress: At least with testing pi3d as above, if I delete the environment I created earlier and recreate it using the --system-site-packages option it seems to work (I didn't change my user name as instructed during setup!)
```
(venv) pi@raspberrypi:~ $ deactivate
pi@raspberrypi:~ $ rm -r venv
pi@raspberrypi:~ $ python -m venv --system-site-packages venv
pi@raspberrypi:~ $ source /home/pi/venv/bin/activate
(venv) pi@raspberrypi:~ python -m pip install git+https://github.com/tipam/pi3d
```
NB To stop the demos using the version of pi3d in /home/pi/pi3d, if you cloned it there, you need to `$ mv ~/pi3d_demos/demo.py ~/pi3d_demos/demo.tmp`

This does mean there will probably need work-arounds for picframe installation because we ask for specific version number of some packages, possibly not using the debian site package. More testing, then, when picframe seems to work I will start to see if we can manage without the desktop!!

Paddy

sapnho, 2023-10-17:
> One quirk I've found, that might be related to @helgeerbe's work getting pi3d to run on the mac, is that I have to set `use_glx=True` if I try to start a demo program over SSH. Most demos will run if started locally even without `use_glx` apart from TriceratopsModel which just hangs. One or two, such as TigerTank start but then freeze the process inescapably.

I guess the key is to get it going on the Pi, so that's probably a sacrifice to make. :-)

sapnho, 2023-10-17:
Ok, so I just did a clean install with bullseye on a Pi 4.

I encountered the first difference

`In the raspi-config module, go to 6 Advanced Options > A2 GL Driver > Choose G2 GL Fake KMS.`

There is no fake driver anymore. Just an experimental Wayland backend.

Pi3D installation seems to have worked but gave a few warnings:

> pi@pi4bullseye:~ $ sudo python3 -m pip install -U pip
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pip in /usr/lib/python3/dist-packages (20.3.4)
Collecting pip
  Downloading https://www.piwheels.org/simple/pip/pip-23.3-py3-none-any.whl (2.1 MB)
     |████████████████████████████████| 2.1 MB 1.1 MB/s 
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 20.3.4
    Not uninstalling pip at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'pip'. No files were found to uninstall.
Successfully installed pip-23.3

> pi@pi4bullseye:~ $ sudo python3 -m pip install pi-heif
> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> Collecting pi-heif
>   Downloading pi_heif-0.13.1-cp39-cp39-manylinux_2_28_armv7l.whl.metadata (6.1 kB)
> Collecting pillow>=9.1.1 (from pi-heif)
>   Downloading https://www.piwheels.org/simple/pillow/Pillow-10.1.0-cp39-cp39-linux_armv7l.whl (1.2 MB)
>      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.2/1.2 MB 3.9 MB/s eta 0:00:00
> Downloading pi_heif-0.13.1-cp39-cp39-manylinux_2_28_armv7l.whl (625 kB)
>    ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 625.3/625.3 kB 5.4 MB/s eta 0:00:00
> DEPRECATION: gpg 1.14.0-unknown has a non-standard version number. pip 24.0 will enforce this behaviour change. A possible replacement is to upgrade to a newer version of gpg or contact the author to suggest that they release a version with a conforming version number. Discussion can be found at https://github.com/pypa/pip/issues/12063
> Installing collected packages: pillow, pi-heif
>   Attempting uninstall: pillow
>     Found existing installation: Pillow 8.1.2
>     Uninstalling Pillow-8.1.2:
>       Successfully uninstalled Pillow-8.1.2
> Successfully installed pi-heif-0.13.1 pillow-10.1.0
> WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv

> pi@pi4bullseye:~ $ sudo pip3 install picframe
> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> Collecting picframe
>   Downloading https://www.piwheels.org/simple/picframe/picframe-2023.7.26.post1-py3-none-any.whl (3.2 MB)
>      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.2/3.2 MB 3.8 MB/s eta 0:00:00
> Collecting Pillow==9.5.0 (from picframe)
>   Downloading https://www.piwheels.org/simple/pillow/Pillow-9.5.0-cp39-cp39-linux_armv7l.whl (1.1 MB)
>      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.1/1.1 MB 3.9 MB/s eta 0:00:00
> Collecting defusedxml (from picframe)
>   Downloading https://www.piwheels.org/simple/defusedxml/defusedxml-0.7.1-py2.py3-none-any.whl (25 kB)
> Collecting pi3d>=2.49 (from picframe)
>   Downloading https://www.piwheels.org/simple/pi3d/pi3d-2.49-py3-none-any.whl (270 kB)
>      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 270.8/270.8 kB 4.0 MB/s eta 0:00:00
> Requirement already satisfied: PyYAML in /usr/lib/python3/dist-packages (from picframe) (5.3.1)
> Collecting paho-mqtt (from picframe)
>   Downloading https://www.piwheels.org/simple/paho-mqtt/paho_mqtt-1.6.1-py3-none-any.whl (75 kB)
>      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 75.6/75.6 kB 3.0 MB/s eta 0:00:00
> Collecting IPTCInfo3 (from picframe)
>   Downloading https://www.piwheels.org/simple/iptcinfo3/IPTCInfo3-2.1.4-py3-none-any.whl (12 kB)
> Requirement already satisfied: numpy in /usr/lib/python3/dist-packages (from picframe) (1.19.5)
> Collecting ninepatch (from picframe)
>   Downloading ninepatch-0.1.22-py3-none-any.whl (16 kB)
> Requirement already satisfied: pi-heif>=0.8.0 in /usr/local/lib/python3.9/dist-packages (from picframe) (0.13.1)
> DEPRECATION: gpg 1.14.0-unknown has a non-standard version number. pip 24.0 will enforce this behaviour change. A possible replacement is to upgrade to a newer version of gpg or contact the author to suggest that they release a version with a conforming version number. Discussion can be found at https://github.com/pypa/pip/issues/12063
> Installing collected packages: paho-mqtt, IPTCInfo3, Pillow, defusedxml, pi3d, ninepatch, picframe
>   Attempting uninstall: Pillow
>     Found existing installation: Pillow 10.1.0
>     Uninstalling Pillow-10.1.0:
>       Successfully uninstalled Pillow-10.1.0
[... 28 lines cut ...]
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Tue Oct 17 21:53:50 2023
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
error: XDG_RUNTIME_DIR not set in the environment.
DRM_IOCTL_MODE_CREATE_DUMB failed: Cannot allocate memory
Failed to create scanout resource
(EE) 
Fatal server error:
(EE) AddScreen/ScreenInit failed for driver 0
(EE) 
(EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
(EE) 
(EE) Server terminated with error (1). Closing log file.
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error

I guess that's a result of the graphic driver.

paddywwoof, 2023-10-23:
So this is what I did next, and pi3d seems to run OK (though no where near the FPS of rust_pi3d)
```
$ sudo apt update
$ sudo apt upgrade
$ sudo apt install libsdl2-dev python3-numpy python3-pil python3-pip git
$ python -m pip install pysdl2 --break-system-packages ## really should use venv or some other option!
$ cd ~
$ git clone https://github.com/tipam/pi3d
$ cd pi3d
$ git checkout sdl2 ## use sdl2 branch
$ cd ~
$ git clone https://github.com/pi3d/pi3d_demos
$ cd pi3d_demos
## edit ForestWalk.py, say, and change
  DISPLAY = pi3d.Display.create(x=100, y=100, frames_per_second=30, use_sdl2=True)
##
$ python ForestWalk.py
```

paddywwoof, 2023-10-26:
I've got a workable system: For the Pi up to and including 3 I can run from the command line with SDL2, for the Pi 4 (and 5, who knows?) it runs using `xinit`.
Once we've tested it, the sdl2 branches will be merged into main so the local install steps will be simplified. However the new system of `venv` seems fraught and I couldn't get it to work without scrambling the modules, so used `--break-system-packages`, which doesn't sound very promising but seems to work.
1. Flash SD with bookworm lite and set county, language, user, password, wifi, boot. (I didn't change anything else in raspi-config as I had an ethernet connection and connected keyboard.)
2. apt install the libraries for either <3 or 4+
```
sudo apt update
sudo apt upgrade
sudo apt install python3-numpy python3-pil python3-pip git
```
2.a. install libraries for SDL2 with Pi <3 - obviously this could be added to above command
```
sudo apt install libsdl2-dev python3-sdl2
```
2.b install X11 libraries and xinit application if Pi is 4 or 5
```
sudo apt install xserver-xorg xserver-xorg-legacy xinit
```
3. pi3d, sdl2 branch
```
git clone https://github.com/tipam/pi3d
cd pi3d
git checkout sdl2
cd ..
```
4. picframe, sdl2 branch
```
git clone https://github.com/helgeerbe/picframe
cd picframe
git checkout sdl2
nano pyproject.toml
# change dependency to "Pillow>=9.4.0"
cd ..
```
5. install pi3d and picframe from local repos
```
python -m pip install file:///home/pi/pi3d --prefer-binary --break-system-packages
python -m pip install file:///home/pi/picframe --prefer-binary --break-system-packages
/home/pi/.local/bin/picframe -i /home/pi/ # accept defaults
mkdir {Pictures, DeletedPictures}
```
6.a Set config to use SDL2 and run in Pi <3
```
nano ~/picframe_data/config/coonfiguration.yaml
# use_sdl2: True
python /home/pi/.local/bin/picframe /home/pi/picframe_data/config/configuration.yaml
```
6.b Can still set configuration use_sdl2: True but not needed for Pi 4+
```
sudo nano /etc/X11/Xwrapper.config
# change
allowed_users=anybody
needs_root_rights=yes
# save
xinit /usr/bin/python /home/pi/.local/bin/picframe /home/pi/picframe_data/config/configuration.yaml
```

sapnho, 2023-10-27:
Thanks, Paddy. Will try this on the weekend.

rock-meister, 2023-10-27:
Newbie alert!
I am running it on Model 3B
cat /proc/cpuinfo | grep ^Model
Model           : Raspberry Pi 3 Model B Rev 1.2

Here are a few minor issues
Issue 1: Unable to set locale to en_US.UTF-8
ERROR:model.Model:error trying to set locale to en_US.UTF-8

Issue 2: mkdir {Pictures, DeletedPictures} doesn't work.  It's creating 2 directories "{Pictures" and "DeletedPictures}"

Issue 3: Typo on 6.a  filename coonfiguration.yaml should be configuration.yaml

After fixing those minor issues, it is running fine.  Thanks everyone for all the hard work!

paddywwoof, 2023-10-28:
@rock-meister thanks for testing and sorry for the typos which I've fixed. It's rather a 'work in progress' as git cloning, changing branch then pip installing locally is very clunky. (I'm not sure about the en_US but it should be possible to set this up at an earlier stage in `raspi-config`). I would really like to find a way for the SDL2 route to work on RPi 4 and 5 which would simplify the whole thing further. Short term it's really helpful of you to test the system on different setups.

sapnho, 2023-10-28:
When I get to point 5 `python -m pip install file:///home/pi/pi3d --prefer-binary --break-system-packages` I get the error

> Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing ./pi3d
ERROR: Could not install packages due to an OSError: [Errno 2] No such file or directory: '/home/pi/pi3d'

(Pi4 Bookworm)

sapnho, 2023-10-28:
Just installed it on a Pi4 with Bookworm (64-bit with Desktop (the recommended version)) and it works. Will let it run for a while to see if it's stable. Thanks, @paddywwoof !

sapnho, 2023-10-28:
The only unusual thing I can notice at this point, is when I add pictures on the fly while Pi3D is running, the I get this error

> WARNING:get_image_meta.GetImageMeta:Cause: cannot identify image file '/home/pi/Pictures/2021. 09.15 - 12.46. 24 Fnatenrahmen.jpg'

However, the image is shown and the metadata are correct.

sapnho, 2023-10-28:
@paddywwoof Can you test if the performance is the same? Not sure how would you measure it.

paddywwoof, 2023-10-28:
Hi, I will check out that error, might be more checking for various things.. especially getting video working after a fashion.

I can test the frame rate on the RPi3 in the kitchen then update it to new stem and try that.

sapnho, 2023-10-29:
BTW, this `sudo apt install python3-numpy python3-pil python3-pip git
` does not seem to be necessary with the recommended Bookworm version. I get

> python3-numpy is already the newest version (1:1.24.2-1).
python3-pil is already the newest version (9.4.0-1.1+b1).
python3-pil set to manually installed.
python3-pip is already the newest version (23.0.1+dfsg-1+rpt1).
git is already the newest version (1:2.39.2-1.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

sapnho, 2023-10-29:
And `sudo apt install xserver-xorg xserver-xorg-legacy xinit ` only `xserver-xorg-legacy` needs to be installed.

> xserver-xorg is already the newest version (1:7.7+23).
xinit is already the newest version (1.4.0-1).
The following NEW packages will be installed:
  xserver-xorg-legacy
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.

sapnho, 2023-10-29:
> The only unusual thing I can notice at this point, is when I add pictures on the fly while Pi3D is running, the I get this error
> 
> > WARNING:get_image_meta.GetImageMeta:Cause: cannot identify image file '/home/pi/Pictures/2021. 09.15 - 12.46. 24 Fnatenrahmen.jpg'
> 
> However, the image is shown and the metadata are correct. And I think the error message came with the old version as well.

I think this comes up because the file analysis is triggered before the full transfer of the file has actually taken place.

sapnho, 2023-10-29:
I have started an post with your new instructions, @paddywwoof here: https://www.thedigitalpictureframe.com/how-to-build-a-gorgeous-digital-picture-frame-with-a-raspberry-pi-based-on-2023-bookworm/

Haven't tested it on the Pi3 yet.

Does this `2.a. install libraries for SDL2 with Pi <3` include the Pi 3 is it just for the Pi2?

sapnho, 2023-10-29:
I am using Raspberry Pi OS (64-bit) with the Raspberry Pi Desktop, which has "Recommended" in brackets. I always try to stay as close to the default settings, which avoids issues when readers try to follow the instructions.

And definitely not the end of the world. :-))))

sapnho, 2023-10-29:
Just made a fresh install on a Pi 3. Also works as it should! :-)

helgeerbe, 2023-10-30:
Hi, sorry I was away on a longer trip and hadn't time to write.

This is my shortest setup guide to run picframe on bookworm:

1. Install bookworm desktop. I configured it as host=raspberrypi , user=pi and using wifi
2. After first boot open terminal and run:
````bash
sudo apt-get update 
sudo apt-get full-upgrade 
sudo reboot 
````
3.
````bash
sudo raspi config
````
- 1 System Options -> S5 Boot -> B2 Sonsole Autologin as 'pi' user
- 2 Display Option -> D2 Screen Blanking -> Disable Screen Blanking
- 3 Interface Options -> I1 SSH -> Enable SSH
- 5 Localisation Options -> L1 Locale -> de_DE.UTF-8 UTF-8 (or whatever what your country is)
- 5 Localisation Options -> L2 Timezone -> Europe Berlin (or whatever your timezone is)
- 5 Localisation Options -> L3 Keyboard -> my keboard layout
- 6 Advanced Options -> A1 Expand Filesystem and reboot
4. Install picframe using a virtual environment for python
 ````bash
mkdir pvenv 
python –v venv /home/pi/pvenv 
source pvenv/bin/activate
pip install picframe
picframe –i .  
````
5 create startscript that activates virtual python env
`nano start_picframe.sh`
````bash
#!/bin/bash
source /home/pi/pvenv/bin/activate # activate phyton virtual env
picframe &  #start picframe
````
make script executable
`chmod +x start_picframe.sh`

6. create user systemd service to start picframe on boot
````bash
mkdir ~/.config/systemd/user/ -p
nano ~/.config/systemd/user/picframe.service
````
```ÌNI
[Unit]
Description=PictureFrame on Pi3

[Service]
ExecStart=/usr/bin/wayfire
Restart=always

[Install]
WantedBy=default.target
````

7. Controlling the service

You have the following options:

- Enable service `systemctl --user enable picframe.service`
- Disable service `systemctl --user disable picframe.servcie`
- Start service `systemctl --user start picframe.service`
- Stop service `systemctl --user stop picframe.service`
- Stop restart `systemctl --user restart picframe.service`
- Check logfiles `journalctl --user-unit picframe.service`

Good: Easy setup
Bad: Before the first image is displayed, you will see the desktop for a short time.

sapnho, 2023-10-30:
THanks much, @helgeerbe ! Is this for both Pi 3 and 4?

paddywwoof, 2023-10-30:
Helge, that's great fixing the issues with venv.

In latest news: I've found the cause of SDL2 not working on the RPi 4+. Basically trying to set the depth buffer to 24 bits causes the problem, I'll check why, but it works without specifying anything. So that would mean one set of instructions and no need for any desktop, X11 or otherwise... However I've not yet found a way to blank the screen without X11. Hmm😕

helgeerbe, 2023-10-31:
> THanks much, @helgeerbe ! Is this for both Pi 3 and 4?

I tested it on Pi3. But there is nothing special, so I would assume this works with a Pi 4 as well.

But this setup uses xWayland. So its still an X application. 

If @paddywwoof get the SDL support ready and we set SDL_VIDEODRIVER=wayland. It should run completely natively on wayland.

Anyhow, I found some hickups in my setup:
- starting the service is not working over ssh. You have to run `sudo reboot` as a workaround
- Picframe log does not appear in in `journalctl`. You have to set the log path in configuration.yml
- I have to find out, which command can be used to blank the screen.

Performance seems to be (on Pi 3) the same, as under bulleseye.

paddywwoof, 2023-10-31:
I've updated the SDL2 branch so it works with Piv4+ and, once I've done a little more testing I will merge into main and bump to v2.51.

I got a bit of a lead here https://forums.raspberrypi.com/viewtopic.php?t=358689 but doesn't seem to work so far. There is an option in SDL to enable and disable screen blanking so maybe we can set it to a short period in the setup then toggle with SDL... Who knows!

helgeerbe, 2023-10-31:
@paddywwoof I tested the sdl branches and it works, without using X11 anymore.

What have I done:
- sudo apt install libsdl2-dev python3-sdl2
- in the venv environment I removed picframe and pi3d and installed the sdl versions
- in the venv enrironment `pip install  pysdl2-dll` so python can import sdl. (Maybe you don't need to install the libsdl2-dev python3-sdl2)
-  add to configuration.yaml `use_sdl2: True`
- change `start_picframe.sh` so sdl videodriver is wayland and not xWalyand anymore
````sh
#!/bin/bash
source /home/pi/pvenv/bin/activate # activate phyton virtual env
export SDL_VIDEODRIVER=wayland # sdl has to use wayland
picframe &  #start picframe
````

I would assume that the same setup would work for Pi3, Pi4 and Pi5. Would be good, if someone can confirm this.

If this is working we should:
- make sdl the default
- add pysdl2-dll to the dependencies
- put a sample picframe_service, and start_picframe.sh file in the config folder

Maybe we should find out, what additional packages are needed, if you install the lite version

helgeerbe, 2023-10-31:
commands to blank screen under wayland on/off are
````sh
wlr-randr --output HDMI-A-1 --off
wlr-randr --output HDMI-A-1 --on
````

paddywwoof, 2023-10-31:
@helgeerbe I've started trying to follow your instructions with a new SD but I can't really get past first base. I had to insert a step ```sudo apt install python3-venv``` but after ```pip install picframe``` I don't have a any usable modules. `numpy` is missing the openblas library, PIL is missing various components, pi3d can't find the OpenGL libraries... What am I missing to make it work?

Also, do you get screen blanking with `wlr-randr`? I just get `wlr-randr says failed to connect to display` and if I run as root `XDG_RUNTIME_DIR is invalid or not set in the environment` If I run `setterm` from a keyboard on the RPi then it will blank and wake up but it doesn't work either on the remote picframe.

helgeerbe, 2023-10-31:
@paddywwoof which image are you using?. I used

Raspberry Pi OS with desktop

Release date: October 10th 2023
System: 64-bit
Kernel version: 6.1
Debian version: 12 (bookworm)
Size: 1,093MB

`wlr-randr` does not work remotely (ssh) and wayfinder (compositor) has to be running. To test ist run it from a terminal on the desktop:

````sh
wlr-randr --output HDMI-A-1 --off; sleep 10; wlr-randr --output HDMI-A-1 --on
````

paddywwoof, 2023-10-31:
I am using `2023-10-10-raspios-bookworm-armhf-lite` but the first SD I burned was the desktop version and I think I had the same error messages when I tried to `pip install numpy` in a venv.

I was running wlr-randr in a python `subprocess.run()` on the picframe RPi but I hadn't got the wayfinder compositor running (is that right or should it be wayfire, I can't find much info on either on line)

sapnho, 2023-10-31:
I also can't get Helge's instructions to work. using a Pi 4 with Bookworm Desktop (Recommended). Does it work through Terminal? I don't have a keyboard connected to my Pi and it's not really the way people would configure their picture frame.

helgeerbe, 2023-10-31:
I assume, that lite is missing too many packages. I would go with desktop, to show that it's working. Using lite we have to identify the missing packages.

Wayfinder must be running. Using wayland, the compositor provides the api. If we are happy, we could replace wafinder with cage. Cage is a compositor for a single kiosk application. But for now, I would use the raspberry standard tools.

@sapnho I don't have a spare pi4. But yes, you should be able to do the configuration through a terminal. Only starting the service through a terminal seems not to work. 

Tomorrow I will do a clean set up and will document each step again.

paddywwoof, 2023-10-31:
picframe itself runs fine on lite with only `libsdl2-dev`, `python3-sdl2`, `python3-numpy`,  `python3-pil` and the other modules needed by picframe the problem is getting python3-venv working. I did a very basic venv just install numpy and it was impossible to load. Surely that shouldn't happen! Maybe it's a bug in pywheels.org?

helgeerbe, 2023-10-31:
Yes it's running. But you are using xWayland (even with sdl). xWayland is a xServer layer in wayland for legacy applications. The future is to get rid of X completely. 

I hadn't any issue creating the python virtual environment. So I could just run `pip install picframe` and all dependencies were installed from pypi.org

paddywwoof, 2023-11-01:
Helge, yes you are right. Even on RPi 4 it's using xWayland. It would be nice if SLD could 'automatically' start wayland like it does for X11 but it doesn't seem to, if the SDL_VIDEODRIVER is set to wayland it fails. I think wayfire needs to be started in another process. Somehow!

I can't decide whether using using the pip route for things like numpy, pill and pysdl2 is the best route. I know we've had problems in the past and needed to specify versions but current 32 bit armhf lite pip packages seem to be in disarray. Just looking at the piwheels.org website for numpy it says you need to
```
sudo apt install libgfortran5 libopenblas0-pthread
pip3 install numpy
```
so why not just `sudo apt install python3-numpy` instead? There is an option when creating a venv to use --system-site-packages

helgeerbe, 2023-11-01:
Hi, I reinstalled picframe on bookworm. Using native Wayland (no X). I created a wiki page, that can be edited more easily:

https://github.com/helgeerbe/picframe/wiki/Setup-guide-Bookworm-and-Wayland

helgeerbe, 2023-11-01:
Hi @paddywwoof I could install all dependent packets with a simple `pip install picframe`. Using a virtual environment we can control even the versions of the packets which are required by picframe.

paddywwoof, 2023-11-01:
@helgeerbe that works great. The SD burning and OS loading is significantly more time consuming than the lite versions but that only really matters during testing where I'm doing it lots of times. The average user will just do it once. I'm in the process of merging pi3d sdl2 branch to main but in the mean time I found I could replace https://github.com/helgeerbe/picframe/wiki/Setup-guide-Bookworm-and-Wayland#install-sdl-branches-of-picframe-and-pi3d with
```
pip uninstall picframe
pip uninstall pi3d
pip install git+https://github.com/tipam/pi3d@sdl2
pip install git+https://github.com/helgeerbe/picframe@sdl2
pip install pysdl2
```
i.e. it's possible to pip install directly from a git branch, which I didn't realise before!

Now we just need to figure out how to launch wayfire without all the desktop stuff. Thanks again for sorting this out.

Paddy

sapnho, 2023-11-01:
You guys are just so great! I was afraid it would take a long time to run PictureFrame on Bookworm! Thank you two so much!

helgeerbe, 2023-11-02:
This is my minimum `wayfinder.ini`

````ini
[core]
plugins = alpha autostart
#xwayland = false

[autostart]
picframe = /home/pi/start_picframe.sh
````

With this setting you will get a black screen with a mouse pointer in the middle of the screen, before picframe starts.
We coud start wayfinder with a separate ini file, to keep the original wayfinder.ini untouch.

@paddywwoof I tried also to disable xwayland, but picframe is complaining, that it still need X11 to be running. I did some tests with SDL_VIDEODRIVER=wayland and checked that picframe is using wayland. So why pi3d needs still X11?

paddywwoof, 2023-11-02:
@helgeerbe I've made a bit of progress:
1. I've done a major hack on pi3d using `project.toml` and getting rid of `setup.py` as well as binning the support for python2 and taking out a couple of other things that said they were deprecated. *Hopefully* it will all still work, I tried installing from wheels I've built here and it runs OK but I will put it on the RPi 3 and 4 at least and run the tests. At the moment it's in sdl2 branch, I've not yet merged to main.
2. I've managed to run pi3d with SDL_VIDEODRIVER=wayland on the lite version of bookworm. Essentially with your wayfire.ini setup - but I left the `xwayland = false` line in. I had the `export SDL_VIDEODRIVER=wayland ` in the bash script but also added it to /etc/environment

paddywwoof, 2023-11-03:
@helgeerbe OK I've made a new SD card using 64 bit bookworm lite and basically followed your instructions with the following changes:
```
# apt update
sudo apt install python3-venv git libsdl2-dev wayfire

mkdir /home/pi/.config
nano /home/pi/.config/wayfire.ini
...
    [core]
    plugins = autostart
    xwayland = false

    [autostart]
    picframe = /home/pi/start_picframe.sh
...
mkdir {Pictures,DeletedPictures}
```
Also I have merged the sdl2 branch to main in pi3d and made pysdl2 a dependency so you can
```
pip uninstall picframe
pip uninstall pi3d
pip install git+https://github.com/tipam/pi3d
pip install git+https://github.com/helgeerbe/picframe@sdl2
```
All in all it seems simpler than the previous X11 descriptions

helgeerbe, 2023-11-03:
I'll give it a try. I'm wondering what the load is. Using top I get on my bullseye (lite) pi3 a load of 0.3, on bookworm (desktop) it's 1.2. That is OK, but not as good as the old OS. Maybe it becomes better using lite)

paddywwoof, 2023-11-03:
Seems to be around 1.1 to 1.2 so only a very marginal improvement. As time goes on it might be improved I suppose.
```
top - 09:05:49 up  7:22,  2 users,  load average: 1.09, 1.14, 1.09
Tasks: 129 total,   2 running, 127 sleeping,   0 stopped,   0 zombie
%Cpu(s):  2.0 us,  1.5 sy,  0.0 ni, 96.5 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st 
MiB Mem :    908.5 total,    185.6 free,    441.9 used,    342.8 buff/cache     
MiB Swap:    100.0 total,    100.0 free,      0.0 used.    466.6 avail Mem 

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND  
    750 pi        20   0  769004 200292  76296 S   6.2  21.5  32:40.08 picframe 
    730 pi        20   0  234748  78540  55236 S   5.9   8.4  24:43.31 wayfire  
   1505 root      20   0       0      0      0 D   1.3   0.0   0:04.05 kworker+ 
   1531 pi        20   0   11712   4676   2776 R   1.0   0.5   0:04.10 top      
   1529 root      20   0       0      0      0 I   0.7   0.0   0:02.52 kworker+ 
     15 root      20   0       0      0      0 I   0.3   0.0   1:37.35 rcu_pre+ 
     25 root      20   0       0      0      0 S   0.3   0.0   0:01.51 ksoftir+ 
    636 root      20   0  262600  19120  16160 S   0.3   2.1   0:17.66 Network+ 
   1504 root      20   0       0      0      0 I   0.3   0.0   0:00.28 kworker+ 
   1507 root      20   0       0      0      0 I   0.3   0.0   0:00.43 kworker+ 
      1 root      20   0  168644  11816   8584 S   0.0   1.3   0:07.62 systemd  
      2 root      20   0       0      0      0 S   0.0   0.0   0:00.16 kthreadd 
      3 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 rcu_gp   
      4 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 rcu_par+ 
      5 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 slub_fl+ 
      6 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 netns    
     10 root       0 -20       0      0      0 I   0.0   0.0   0:00.00 mm_perc+
```

helgeerbe, 2023-11-03:
Looks good.

I have to test, if screenblanking is working with this minimal setup of wayfire.ini or if we have to load an additonal plugin.

I will change the wiki page.

helgeerbe, 2023-11-03:
I updated the wiki.

- add missing `wlr-randr` for screen blanking
- update `start_picframe.sh` Blanking screen to have a nicer startup
- Add an additional tweak chapter
  - disable bluetooth
  - silent boot

helgeerbe, 2023-11-03:
@paddywwoof how do you select the hdmi-port for picframe? Do we need a config to select the hdmi port? Using HMI-A-1 means we would assume that picframe is always running on the first hdmi port.

paddywwoof, 2023-11-03:
I think I found a command to list the displays. I will dig around and see if I can find it. Does the wlr-randr --on and --off work on the desktop SD image for you?

helgeerbe, 2023-11-03:
hm, could ist be more a sdl2 problem? 
I installed lxterminal. Run terminal and picframe (with reduced size). Then I run 'wlr-randr --output HDMI-A-1 --off; sleep 10; wlr-randr --output HDMI-A-1 --on`in the terminal.

picframe terminates with memory exception

rock-meister, 2023-11-03:
@paddywwoof I can confirm that the latest instructions works on using 64 bit bookworm lite on Model 3B+.
@helgeerbe both `systemctl --user start picframe.service` and `systemctl --user restart picframe.service` works fine over ssh.  Not sure what errors you see.

paddywwoof, 2023-11-03:
@rock-meister thanks for checking that. I'm not sure what causes the memory exception but the general problem is turning the screen on and off i.e. at night or if you have an infrared sensor. None of the old methods seem to work anymore on wayland but the nearest is using `wlr-randr`. However it seems to stop the slideshow as well as blanking the monitor, which could be solved by shutting everything down then restarting it again (i.e. with systemd) but that seems like overkill, and it's nice for the slideshow to pickup where you left it rather than re-shuffling everything.

[35 more replies]

closed 2024-11-03 by helgeerbe