helgeerbe/picframe #335: Current image not shown in Home Assistant on Chrome and Opera

sapnho, 2023-10-22:
An old problem has come back, and after many months of trying here and there, I still haven't found a solution.

@helgeerbe's Home Assistant integration has this great feature where you can see the image currently displayed on your picture frame. My problem is: It works great on Safari and Firefox on my desktop, on all browsers (incl. Chrome) on my iPhone and iPad, BUT it doesn't work on my desktop Chrome or Opera browser.

My old solution was to change the website settings and allow the downloading of images (and I basically greenlit everything else), but it only shows the placeholder icons for images but not the current image.

Any thoughts on how I could allow Chrome to display the image? I believe it has something to do with security settings on Chrome. I worked in the past, but at one point, after a number of Chrome updates, it doesn't anymore.

paddywwoof, 2023-10-22:
I think it will be a cross-origin issue, meaning that the web page displaying the image in the browser needs to explicitly allow it, something like [here.](https://developer.mozilla.org/en-US/docs/Web/HTML/CORS_enabled_image) Browsers probably keep adding hurdles whenever new cunning exploits are discovered. I haven't looked at what the HA functionality actually does, so it could well turn out to be caused by something different!

sapnho, 2023-10-22:
Yes, I guess it's something in that direction. Is there anybody on this forum who uses Chrome with Home Assistant using Nabu Casa and can see the current frame image?

closed 2026-05-06 by github-actions[bot]