helgeerbe/picframe #341: Would Pi3D PictureFrame work on an Orange Pi?

sapnho, 2023-11-22:
I was wondering if there is anything specific about Pi3D that would require a Raspberry Pi instead of one of the clones. I am toying with the idea of playing around with an [Orange Pi](http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-5.html). 

I guess since Pi3D works on Ubuntu, it should run on an Orange Pi. Or am I missing something?

paddywwoof, 2023-11-22:
I would have thought that it should work. So long as libsdl2-dev can be installed and will use the mali GPU. At one point, ages ago, someone managed to get pi3d working with an earlier RPi look-alike (banana?) which had the mali GPU too. Since then pi3d has got more functionality to use standard (i.e. non Broadcom) graphics drivers so it should be easier. As much as I can I will help you to get it working as it would be nice to have an alternative to the RPi.

sapnho, 2023-11-23:
Ok, I think that's worth a try! Thanks for your support!

sapnho, 2023-12-03:
I should get my Orange Pi this week and should be able to do a few tests when I get back from a short break. I am really curious as to its compatibility.

sapnho, 2023-12-16:
With customs clearance, it took a bit longer but I finally got my delivery yesterday. It's an Orange Pi Model 5, unfortunately without wifi, but with an Ethernet connection it's fine. My first question is, what operating system would be best as there are plenty of options? Debian, Ubuntu, Orange Pi? I believe someone on this forum used Debian successfully. Any thought, @paddywwoof ?

paddywwoof, 2023-12-16:
Thought I'd answered this from my phone but it seems all that effort was in vain and it's disappeared into the aether. As I said before...
I would go for the OS that has lots of other people using it. Is the Orange Pi OS a form of Android? Arch linux? I've run pi3d on Android previously so that might be possible, but everything a bit different (and hopefully easier) using the SDL2 render surface. I've also set up Arch to run pi3d as well, ages ago! Debian is the basis of many other linux versions so it will almost certainly work fine but you might have to search around for missing dependencies. I might start with orange Pi and if it turned out to be too difficult try a different one.
Let me know how you get on

sapnho, 2023-12-17:
I guess the equivalent to Raspbian would be the [Orange Pi OS](http://www.orangepi.org/html/softWare/orangePiOS/index.html) which they have for Droid, Arch or OH. Any particular flavor I should start out with?

paddywwoof, 2023-12-17:
I would go for Arch. I don't know what will be installed by default but the equivalent of `apt install` is `pacman -S` I think. You might have to install things like `python-numpy`, `python-pillow` as well as `sdl2` components. You might end up having to compile more things as there is probably a smaller user base than for the Raspberry Pi.

sapnho, 2023-12-17:
Ok, I will try that right after Christmas. It wouldn't be possible without your help, so thanks in advance for your support!

sapnho, 2023-12-26:
Hello @paddywwoof I started [my post about the Orange Pi 5](https://www.thedigitalpictureframe.com/how-to-build-an-8k-digital-picture-frame-with-an-orange-pi-5-and-debian-bookworm/). I ended up using Debian, I hope that works for you as well.

The Pi3D installation worked without any errors. But as I am new to Debian, I still need to figure out how to boot to console. Would you have idea?

sapnho, 2023-12-26:
So I installed the server version of Debian, which boots right into the console. So one problem gone.

However picframe.service doesn't run. Upon investigation I get this error:
`orangepi@orangepi5:~/.config/systemd/user$ systemctl --user status picframe.service
`
```orangepi@orangepi5:~/.config/systemd/user$ systemctl --user status picframe.service
× picframe.service - PictureFrame
     Loaded: loaded (/home/orangepi/.config/systemd/user/picframe.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Tue 2023-12-26 19:54:42 CET; 4s ago
   Duration: 1.185s
    Process: 2813 ExecStart=/usr/bin/wayfire (code=exited, status=255/EXCEPTION)
   Main PID: 2813 (code=exited, status=255/EXCEPTION)
        CPU: 505ms

Dec 26 19:54:42 orangepi5 systemd[2105]: picframe.service: Scheduled restart job, restart counter is at 5.
Dec 26 19:54:42 orangepi5 systemd[2105]: Stopped picframe.service - PictureFrame.
Dec 26 19:54:42 orangepi5 systemd[2105]: picframe.service: Start request repeated too quickly.
Dec 26 19:54:42 orangepi5 systemd[2105]: picframe.service: Failed with result 'exit-code'.
Dec 26 19:54:42 orangepi5 systemd[2105]: Failed to start picframe.service - PictureFrame.
```

paddywwoof, 2023-12-26:
Hi, well done getting things going. I'm not sure how much of the Debian desktop will be available in the server version. You will need to have something to render onto with OpenGL. I suspect that it will be easier to use the X11 xserver system as we did prior to the rollout of bookworm with it's switch to Wayland.

There is a set of instructions to install picframe with raspbian lite from a couple of years ago on the picframe forum. That might give you an idea of the packages needed on Debian.

sapnho, 2023-12-27:
I guess you mean the good old 
```
orangepi@orangepi5:~$ export DISPLAY=:0
orangepi@orangepi5:~$ xinit /home/orangepi/.local/bin/picframe /home/orangepi/picframe_data/config/configuration.yaml

```
command. 

Just did that and installed Xinit. The first error message that I got was

```
xinit: unable to run server "X": No such file or directory
Use the -- option, or make sure that /usr/bin is in your path and
that "X" is a program or a link to the right type of server
for your display.  Possible server names include:

    Xorg        Common X server for most displays
    Xvfb        Virtual frame buffer
    Xfake       kdrive-based virtual frame buffer
    Xnest       X server nested in a window on another X server
    Xephyr      kdrive-based nested X server
    Xvnc        X server accessed over VNC's RFB protocol
    Xdmx        Distributed Multi-head X server

xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error
orangepi@orangepi5:~$
```

Then I installed with `sudo apt install xserver-xorg` and did the Xinit command again. This time the error message looked somewhat familiar:

```
orangepi@orangepi5:~$ export DISPLAY=:0
orangepi@orangepi5:~$ xinit /home/orangepi/.local/bin/picframe /home/orangepi/picframe_data/config/configuration.yaml

X.Org X Server 1.21.1.7
X Protocol Version 11, Revision 0
Current Operating System: Linux orangepi5 5.10.160-rockchip-rk3588 #1.1.8 SMP Mon Nov 13 11:11:04 CST 2023 aarch64
Kernel command line: root=UUID=349fd087-2dc6-442a-8175-ecf252f20f52 rootwait rootfstype=ext4 splash=verbose console=ttyS2,1500000 console=tty1 consoleblank=0 loglevel=1 ubootpart=8b3b1554-7514-3442-a011-406f51d77b4e usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u cma=128M  cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1
xorg-server 2:21.1.7-3+deb12u2 (https://www.debian.org/support) 
Current version of pixman: 0.42.2
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/home/orangepi/.local/share/xorg/Xorg.0.log", Time: Wed Dec 27 22:19:05 2023
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE) 
Fatal server error:
(EE) xf86OpenConsole: Cannot open virtual console 2 (Permission denied)
(EE) 
(EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
(EE) Please also check the log file at "/home/orangepi/.local/share/xorg/Xorg.0.log" for additional information.
(EE) 
(EE) Server terminated with error (1). Closing log file.
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error
```
Maybe I am exporting to the wrong display? Any other Xserver stuff I need to install?

paddywwoof, 2023-12-28:
Hi there were various X related things that @thomsedoff suggested https://gist.github.com/thomasedoff/3973846101da1e7496524be075f1c02f#6-install-necessary-software-and-dependencies but there might be others. I'm not really sure what's the best approach to doing this scientifically! There will probably be some of the OpenGL libraries needed as well once you have X running.

I'm also wondering if there's a way to autostart a full screen app like picframe when the desktop boots... I'm sure that's been used with pi3d or picframe at some time. But, obvious, getting a lite image working would be tidier.

Paddy

paddywwoof, 2023-12-28:
PS the permission error might suggest trying `sudo xinit...` as a first test

sapnho, 2023-12-28:
> PS the permission error might suggest trying `sudo xinit...` as a first test

A little bit better but not much yet...

```
sudo xinit /home/orangepi/.local/bin/picframe /home/orangepi/picframe_data/config/configuration.yaml
[sudo] password for orangepi: 

X.Org X Server 1.21.1.7
X Protocol Version 11, Revision 0
Current Operating System: Linux orangepi5 5.10.160-rockchip-rk3588 #1.1.8 SMP Mon Nov 13 11:11:04 CST 2023 aarch64
Kernel command line: root=UUID=349fd087-2dc6-442a-8175-ecf252f20f52 rootwait rootfstype=ext4 splash=verbose console=ttyS2,1500000 console=tty1 consoleblank=0 loglevel=1 ubootpart=8b3b1554-7514-3442-a011-406f51d77b4e usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u cma=128M  cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1
xorg-server 2:21.1.7-3+deb12u2 (https://www.debian.org/support) 
Current version of pixman: 0.42.2
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Thu Dec 28 20:55:09 2023
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
xinit: Unable to run program "/home/orangepi/.local/bin/picframe": No such file or directory
Specify a program on the command line or make sure that /usr/bin
is in your path.

xinit: connection to X server lost

waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.
```

sapnho, 2023-12-28:
Where is this now?

`xinit: Unable to run program "/home/orangepi/.local/bin/picframe": No such file or directory
`

paddywwoof, 2023-12-29:
Hmm, I'd imagined that Debian based systems would do things the same way, but maybe not. If you used the venv system then picframe is installed in that. You can use the linux `find` command to search for programs.Paddy