helgeerbe/picframe #342: Issue with Bookworm when monitor is turned off

sapnho, 2023-11-25:
I have just been alerted to the following issue: 

He installed Pi3D Picture Frame with Bookworm according to the latest instructions. 

Everything works fine, however, when he turns off the attached monitor and then back on again, only a grey screen with the mouse pointer appears. Pi3D does not continue.

Could that have anything to do with our general HDMI on/off issue?

paddywwoof, 2023-11-25:
I suspect it is all related. When I tried to get the screensaver working, I could get it to turn off (I think, *lots* of different permutations with every conceivable variable changed!) but when the screensaver was disabled again I had to move the mouse or press a key to get the slide show to be visible.

I noticed @CraigeryTheKid suggested tvservice which needs systemctl restart, I woonder if that might do it?

sapnho, 2023-11-26:
Upon trying I get `-bash: tvservice: command not found`.

paddywwoof, 2023-11-26:
Yes I thought as much (about tvservice not working) but it might be worth trying the systemctl restart to get the monitor to show something after it's been turned off. Of course that would restart the whole show and loose the position it was in when it was closed down so not really a good solution. I suppose we could store the 'state' which could be as little as a random number generator seed and a sequence number but still messy.

sapnho, 2023-11-26:
Maybe we just need to wait a while until a fix for Bookworm arrives. What is really annoying is that even [this solution](https://www.thedigitalpictureframe.com/with-vcgencmd-no-longer-working-determine-the-power-status-of-a-connected-display-with-xrandr/) doesn't work.

CraigeryTheKid, 2023-12-09:
Not to go completely barbaric on it - but if screen auto on/off is the "last" thing not working right - I might just put the monitor power on a christmas tree timer! Thanks to all for your work!