helgeerbe/picframe #343: Samba Question

GitRayc, 2023-11-26:
Got picframe up and working beautifully on pi4.   Gave up on new code on pi3,  just two many hasstles. !  

Samba is working to connect with my Mac nicely.  However my question is:

I am storing all my pics on a usb drive   It is called USBPICS and it automounts as it should.

When I use Samba on my mac I see that folder and I can open it.   However in Samba on my mac the folder appears to be empty   I can see it and open it but I cannot see any of the contents (pics) in it. 

How do I fix that?

Ray

GitRayc, 2023-11-26:
Sorry should have  mentioned the path is /home/pi/USBPICS   The same directory as the Pictures folder etc. of course.

paddywwoof, 2023-11-26:
Have you tried refreshing the file index on the Mac (not sure how to do that but you might find something on Google)

GitRayc, 2023-11-26:
It sorted itself out.  Working well today and can use USBPICS drive through mac.

closed 2023-11-29 by GitRayc