helgeerbe/picframe #347: Potential problem with pictureframe.db3 updating

GitRayc, 2023-12-01:
Paddy:

What triggers the pictureframe.db3 file to refresh/update?

I think there my be a problem here.  I have set my "yaml" file line "reshuffle_num" parameter to 1.   So as I understand it, a new shuffle will happen once all images have been displayed once.   I did this because I have been a having an ongoing problem with new images added to my usb thumbdrive not showing up. I thought perhaps that once the previous images had run through once, that perhaps this database would be rebuilt to recognize new  images, a new shuffle would take place and I would see them then.   Maybe it doesn't work that way.   

However I have added new images that never show up after hours of running through the images. Several restarts do not seem to fix the problem but eventually a reboot will solve the problem.

I have found that if I delete the pictureframe.db3 file and reboot; that new images added will be recognized and they show up immediately as one would expect.

So perhaps there is a problem with this file refreshing on a consistent basis?  Just wondered what triggers it to rebuild.

Hope I am not being a bother.  But I love this package!

Ray

GitRayc, 2023-12-02:
Thanks for the update mate!  I run my slideshow from a images that I store on a USB Thumbdrive which of course mounts automatically when the pi boots up.

But the only way I can get new images to be recognized as I add them is to delete the pictureframe_db3 file and reboot.  Not  a huge deal but just noting such.

Ray

paddywwoof, 2023-12-02:
Just to clarify. If you stop the picframe app and restart it (or reboot the whole Raspberry Pi) the added pictures don't show? That *definitely*shouldn't be the case. See point c above.

GitRayc, 2023-12-02:
Could it be that db3 is looking at the pictures directory and not the USB Thumbdrive where my pics are stored? Maybe I need to edit a line to point to my thumbdrive rather than looking at the pictures directory,

Of course my yaml file has been edited to look at the thumbdrive rather than the pictures directory  
Generally things are finally running beautifully.

paddywwoof, 2023-12-05:
It sounds more like the part of `image_cache` that checks if anything has changed in a directory isn't registering a change when files are added, and not putting the new images onto the db. Maybe this is some quirk of USB Thumbdrives (though it does seem very unlikely to me). You could test it by running something like this in your terminal before and after adding a new image to the drive
```python
import os
for (dir, _, _) in os.walk('/media/pi/my_thumb_drive/'): # use wherever the USB is mounted, whatever you use to look at files from the RPi terminal
    print(dir, os.stat(dir).st_mtime)
```

jgodfrey, 2023-12-05:
> It sounds more like the part of `image_cache` that checks if anything has changed in a directory isn't registering a change when files are added

Agreed. We first ask the OS for any *directories* that have been modified. Then, in those directories only, we look for update or new files. If that initial "modified directories" check is failing for some reason, we won't find any changes. IIRC, *updating* an existing file in a folder won't mark the folder as modified, so that change is missed by default. I wonder if your updates get picked up if you `touch` the containing folder?

closed 2023-12-26 by GitRayc