helgeerbe/picframe #348: Problem editing yaml to enable MQTT

GitRayc, 2023-12-09:
Sorry to bother you all again.

I am trying to take the next step to be able to change to subdirectories of my picture data base.  But I am doing something wrong when I edit the yaml file which then causes booting up.  It hangs when trying to load/start picframe.

I have mosquitto installed okay and the test you suggest confirms this, 

Next is editing the yaml file where I am having problems.

At default it looks like this:

mqtt:
  use_mqtt: False                                 # default=False. Set True true, to enable mqtt
  server: "your_mqtt_broker"              # No defaults for server
  port: 8883                                        # default=8883 for tls, 1883 else (tls must be "" then !!!!!)
  login: "name"                                   # your mqtt user
  password: "your_password"             # password for mqtt user
  tls: "/path/to/your/ca.crt"                # filename including path to your ca.crt. If not used, must be set to "" !!!!
  device_id: "picframe"                       # default="picframe" unique id of device. change if there is more than one PictureFrame
  device_url: ""                                    # if use_http==True, set url to picframe config page. Must be a valid url, or

You say to edit the following lines.     Can you tell me if this is correct in my example below:

use_mqtt: True 
server: 192.168.1.209           I do not include the quotation marks correct?  192.168.1.209 is my ip address
port: 1883                             it is 8883 by default.   Should I leave it as 8883  or change it to 1883?

jgodfrey, 2023-12-09:
If you're having a hang problem, it's likely indicative of a syntax problem in the config file. It's going to be hard to diagnose without the complete file - can you attach it?

Regarding the values, they generally look OK to me.

- You can simply use `"localhost"` for the server (with the quotes)
- You probably want port 1883. Only use 8883 if your using TLS

GitRayc, 2023-12-09:
Which config file would you like to see?  The complete configuration.yaml?

GitRayc, 2023-12-09:
If you meant the configurtion.yaml then here it is.    Set to False right now so picframe will load/start
viewer:
  blur_amount: 12                         # default=12,  larger values than 12 will increase processing load quite a bit
  blur_zoom: 1.0                           # default=1.0, must be >= 1.0 which expands the backgorund to just fill the space around the image
  blur_edges: False                       # default=False, use blurred version of image to fill edges - will override FIT = False
  edge_alpha: 0.5                         # default=0.5, background colour at edge. 1.0 would show reflection of image
  fps: 20.0                                    # default=20.0
  background: [0.2, 0.2, 0.3, 1.0]        # default=[0.2, 0.2, 0.3, 1.0], RGBA to fill edges when fitting
  blend_type: "blend"                        # default="blend", choices={"blend", "burn", "bump"}, type of blend the shader can do
  font_file: "/home/pi/picframe_data/data/fonts/NotoSans-Regular.ttf"
  shader: "/home/pi/picframe_data/data/shaders/blend_new"
  show_text_fm: "%b %d, %Y"               # default "%b %d, %Y", format to show date over the image
  show_text_tm: .1                          # default=20.0, time to show text over image with file name
  show_text_sz: 1                           # default=40, text character size
  show_text: "title caption name date folder location"  # default="title caption name date folder location", show text, include combination of words: title, caption name, date, location, folder
  text_justify: "L"                       # text justification L, C or R
  text_bkg_hgt: 0.25                      # default=0.25 (0.0-1.0), percentage of screen height for text background texture
  text_opacity: 0                       # default=1.0 (0.0-1.0), alpha value of text overlay
  fit: False                              # default=False, True => scale image so all visible and leave 'gaps'
                                          #                False => crop image so no 'gaps'
  kenburns: False                         # default=False, will set fit->False and blur_edges->False
  display_x: 0                            # offset from left of screen (can be negative)
  display_y: 0                            # offset from top of screen (can be negative)
  display_w: null                         # width of display surface (null->None will use max returned by hardware)
  display_h: null                         # height of display surface
  display_power: 0                        # default=0. choices={0, 1}, 0 will use legacy `vcgencmd` and 1 will use `xset` to blank the display
  use_glx: False                          # default=False. Set to True on linux with xserver running
  use_sdl2: True                          # default=True. Needed for running pi3d with sdl2 on Pi2 and Pi3

  mat_images: 0.01                        # default=0.01, True, automatically mat all images. False, don't automatically mat any images. Real value, auto-mat all images with aspect ratio difference > than value
  mat_type: null                          # default=null, A string containing the mat types to choose from when matting images. It can consist of any or
                                          # all of 'float float_polaroid float_color_wrap single_bevel double_bevel double_flat' (null or '' will use all mat types)
  outer_mat_color: null                   # default=null, Color of the outer mat as an RGB list. null will auto-select a reasonable color based on the image.
  inner_mat_color: null                   # default=null, Color of the inner mat as an RGB list. null will auto-select a reasonable color based on the image.
  outer_mat_border: 75                    # default=75, Minimum outer mat border in pixels
  inner_mat_border: 40                    # default=40, Minimum inner mat border in pixels (for styles that use it)
  outer_mat_use_texture: True             # default=True, True uses a texture for the outer mat. False creates a solid-color outer mat.
  inner_mat_use_texture: False            # default=False, True uses a texture for the inner mat. False creates a solid-color inner mat.
  mat_resource_folder: "/home/pi/picframe_data/data/mat" # Folder containing mat image files

  show_clock: False                       # default=False, True shows clock overlay. False does not show clock overlay
  clock_justify: "R"                      # default="R", clock justification L, C, or R
  clock_text_sz: 120                      # default=120, clock character size
  clock_format: "%I:%M"                   # default="%I:%M", strftime format for clock string
  clock_opacity: 1.0                      # default=1.0 (0.0-1.0), alpha value of clock overlay

  menu_text_sz: 40                        # default=40, menu character size
  menu_autohide_tm: 10.0                  # default=10.0, time in seconds to show menu before auto hiding (0 disables auto hiding)
  geo_suppress_list: []                   # default=None, substrings to remove from the location text

model:
  pic_dir: "/home/pi/USBPICS"                   # default="/home/pi/Pictures", root folder for images
  deleted_pictures: "/home/pi/DeletedPictures"   # move deleted pictures here
  follow_links: False                     # default=False, By default, picframe will not walk down into symbolic links that resolve to directories. Set follow_links to True to visit directories pointed to by symlinks, on systems that support them.
[... cut ...]
      shortcut: "l"                       # default="l"
    exit:                                 # exit PictureFrame
      enable: False                       # default=False
      label: "Exit"                       # default="Exit"
      shortcut: "e"                       # default="e"
    power_down:                           # power down the device, uses sudo
      enable: False                       # default=False
      label: "Power down"                 # default="Power down"
      shortcut: "p"                       # default="p"

GitRayc, 2023-12-09:
Updated OS to the latest patches and all seems to work now

Thanks all

closed 2023-12-09 by GitRayc