helgeerbe/picframe #349: Question in regards to subdirectory setting in yaml file

GitRayc, 2023-12-09:
Again, than you all so much.   I truly am learning!!!

Question now is trying to understand how the pic directory and subdirectory entries work inthe configuration.yaml file.

As mentioned before, I store all my pics on a usb thumbdrive that automounts on pi 4  bookworm.  It is called USBPICS

In my .yaml file I have set the pic file directory to /home/pi/USBPICS    and all works fine.  This is where my general slideshow pics live.

Now I want to have a subdirectory of family pics.

So on that drive I created a folder called "Family"

In the yaml file I left the pic directory set to /home/pi/USBPICS
I set the subdirectory to  /home/pi/USBPICS/Family

When I run picframe however it access both the main directory and the Family sub directory as well, playing all pics from both directorties.

I thought by using a sub directory (Folder) that I would only see the pics in the main root directory and that I could ask the system to play form the Family directory only when asked to do so with for example:

mosquitto_pub -h localhost -t picframe/subdirectory -m "Family"

I am misunderstanding how this works?

Thanks
Ray

GitRayc, 2023-12-09:
Also must paho be installed to enable the mosquitto command to work?  I dont think it is

GitRayc, 2023-12-09:
If is use the command sudo pip3 install paho-mgtt

It responds with:
error: externally-managed-environment

This environment is externally managed
To install Python packages system-wide, try apt install
python3-xyz, where xyz is the package you are trying to install.

If you wish to install a non-Debian-packaged Python package,
create a virtual environment using python3 -m venv path/to/venv.
Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
sure you have python3-full installed.
For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

I have no idea what all that means or how I should procede if necessary

paddywwoof, 2023-12-09:
@GitRayc it is harder with the virtual environment but you do need to do that as that's what picframe uses and it's where all of it's dependencies are installed. I'm on the train at the moment and will put more complete answer when I'm home, but basically, before pip installing you need to
```
source /home/pi/venv_picframe/bin/activate
then
python -m pip install ...
```

GitRayc, 2023-12-10:
I'll give it try.

python -m pip install paho-mqtt  or it is pip3 install?

GitRayc, 2023-12-10:
Okay I think I go it.

So if i want to change to my "Family" subdirectory:
Do I simply issue:

mosquitto_pub -h localhost -t picframe/subdirectory -m "Family"
or it this a two part command to make it work?

Ray

GitRayc, 2023-12-10:
So:
To degub; because I thought perhaps the problem I have is that I am running my pics from a usb Thumbdrive.

So I switched to the Pictures directory.  As a test I put three images in the Pictures directory and two images in the "Family" folder (as subdirectory in the Pictures directory)

And I set my yaml to:
"model:
  pic_dir: "/home/pi/Pictures"                   
  deleted_pictures: "/home/pi/DeletedPictures"   
  follow_links: False                             
  subdirectory: ""                            
  recent_n: 10                        
  reshuffle_num: 1                        
  time_delay: 90.0                       
  fade_time: 10.0                         
  shuffle: True                           
  sort_cols: 'fname ASC'  

I rebooted the system.
It is indeed accessing the Pictures directory now rather than my thumbdrive 
However:
It is now playing all images in the Pictures root directory and the Family directory.  Even thought subdirectory is set to ""

Then I tried
mosquitto_pub -h localhost -t picframe/subdirectory -m ""  

But it is still accessing pics in the Pictures root directory and those in the Family sub folder.

Any ideas what is happening?

GitRayc, 2023-12-11:
I cant help but thinking there is something happening with db3 that it is not updating correctly.  I Installed Table Plus open and watch it.  It has some of the pics in it that reside in the "Family" sub folder.    Also I changed the name of one of the pic files in the root directory and hours later the old name is still there and the new name has never shown up.    Just wandering.

GitRayc, 2023-12-11:
Just qit slideshow and shutdown pi.  Restarted and opened db3 again.
the old filename is still  showing, and not the new one on the file I renamed.
Aslo the files from the Family subdirectory are still there as well.    This mean anything.

GitRayc, 2023-12-11:
I am watching the picframe-data  data directory though and each time a transition to a new image takes place a new db3 file is being creating.  It appears and then immediately disappears.   I guess it is replacing the previous version?

closed 2023-12-22 by GitRayc