helgeerbe/picframe #351: How to purge deleted photos from the database?

clintiepoo, 2023-12-11:
I’ve been getting a lot of picframe restarts and I think it’s because pictures were deleted, but the database still looks for them. 

How do I purge the database?
- I clicked the button on the web interface, but it didn’t seem to work
- I saw MQTT mentioned, but I didn’t see this topic in there. 
- is there a python command I can run regularly to keep the database purged?
- I saw that this used to happen automatically but it must be manually done now. Is there an option to turn auto-purge back on?

thanks!

GitRayc, 2023-12-11:
Dont know if this will help you or not.  I have been having trouble with the database as well.   It seems to grab images from my subdirectory as well even though my setting in my yaml file is set to ""
To clear the database up I downloaded "TablePlus" to be able to read the database file and then selected those entries and deleted them.

jgodfrey, 2023-12-11:
A few random things...

- I don't use the HTML interface often, but I just did a quick PURGE test with it and it seems to work as expected
- My MQTT config publishes topic `picframe/purge_files` to fire the same PURGE command (though, I configured it several years ago now).
- There is no ready-built mechanism you can call directly (that I'm aware of) to kick off the purge workflow
- The code that does the purge is [here ](https://github.com/helgeerbe/picframe/blob/67a3e3f9231b461a6f364657a69984e1ca2c22ed/src/picframe/image_cache.py#L432)if you want to attempt to call it yourself
- We removed the auto-purge mechanism some time ago as it was causing (potentially) large collections of data to removed from the database when a folder or or mount was temporarily unavailable or missing.

clintiepoo, 2023-12-11:
> A few random things...
>
>     * I don't use the HTML interface often, but I just did a quick PURGE test with it and it seems to work as expected
>
>     * My MQTT config publishes topic [...]

Thanks - it makes sense why it was removed. A couple things.

I was previously getting an error about missing files. Here's an example: 
`2023-12-09 13:45:40,891 - image_cache.ImageCache - WARNING - Image '/home/pi/Pictures/2023-12/IMG_7425.JPG' does not exists or is inaccessible`

I think the purge may have worked. I'm still seeing the frame freeze and restart itself, but here are the errors I'm seeing now: 

`2023-12-11 07:10:49,395 - viewer_display.ViewerDisplay - WARNING - Can't create tex from file: "/home/pi/Pictures/2023-11/IMG_7361.JPG" or "<picframe.model.Pic object at 0x6bc47898>"
2023-12-11 07:10:49,395 - viewer_display.ViewerDisplay - WARNING - Cause: height and width must be > 0`

This file is real, just a jpeg picture. 3024x4032.

[image]

And on the MQTT topic (pun intended), I don't see purge_files anywhere. Here's what I have. 

[image]

jgodfrey, 2023-12-11:
Hmmm... So, it looks like we're getting either a 0 width, 0 height, or both for that image (for some reason). The image in question seems to be a portrait orientation based on the dimensions you posted. The code where this is failing is where most of the image manipulation happens for presentation (resizing, orientation corrections, combining of 2 portraits, image matting, ...). I'd guess that something in all of that manipulation is getting tripped up by this image and ultimately resulting in an invalid temporary image - that then triggers the error you see.

Is it possible to make the offending image available for further investigation?

jgodfrey, 2023-12-11:
I'm no MQTT expert, but there seem to be a number of topics that aren't exposed (maybe because they don't have any useful status info). That mentioned topic (`purge_files`) is correct however. See [here](https://github.com/helgeerbe/picframe/blob/67a3e3f9231b461a6f364657a69984e1ca2c22ed/src/picframe/interface_mqtt.py#L404).

clintiepoo, 2023-12-12:
I honestly don't know if this is the issue because these logs aren't changing with the crashes. I'm trying to get to the bottom of what's happening and get to root cause. Here's my setup:

Raspberry Pi 3A+
I set up the frame according to this: https://gist.github.com/thomasedoff/3973846101da1e7496524be075f1c02f, using the X-server and no desktop environment. 
32 Bit OS, Bullseye lite.

The program keeps crashing and going back to the text-only boot screen.

clintiepoo, 2023-12-12:
> Yeah, it doesn't seem to be a problem with the image - it works fine here on my Pi 4 setup...

I think I just need to buy a better raspberry pi. I will close this issue, thanks.

clintiepoo, 2023-12-12:
Closed - hardware issue

closed 2023-12-12 by clintiepoo