helgeerbe/picframe #353: Paho install problem

GitRayc, 2023-12-11:
Pi4 Bookworm

I followed through the article on installing MQTT that was updated Dec 6, 2023 at:
https://www.thedigitalpictureframe.com/how-to-install-an-mqtt-broker-locally-on-your-digital-picture-frame-raspberry-pi/

All went well and the test for "Hello Digital Picture Frame" as described went well.

I  then went on to try the article on Pi Helper (SimplePi) to be able to run commands for date filtering etc.

---------------------------------------------------------
When trying to execute these commands however I get this message on my ipad:  Similar of course from all commands
Traceback (most recent call last):
   File "home/pi/Pi3D_Smart_Filter_Date_Today.py, line 5, in <module>
       import paho.mqtt.client as mqtt
ModuleNotFoundError: No Module named 'paho'

---------------------------------------------------------
This is the associated script:

#!/usr/bin/env python3

# this script sends a MQTT to Pi3D which sets a smart filter to only show images from today

import paho.mqtt.client as mqtt       #This if course is line 5   The one it doesn't like
from datetime import date

client = mqtt.Client()
#client.username_pw_set( "username" , "password" ) - uncomment if you are not sending anonymously
client.connect("pictureframe.local", 1883, 60) # server address here

today = date.today()
d0 = today.strftime("%Y:%m:%d")

client.publish("picframe/date_from", d0)

---------------------------------------------------------
If I look in:
/Volumes/pi/venv_picframe/lib64/python3.11/site-packages
I do have paho folders with files       So I guess paho was installed in that evironment

GitRayc, 2023-12-11:
If this helps:
If I attempt to install again I get:
(venv_picframe) pi@raspberrypi:/home $ pip3 install paho-mqtt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: paho-mqtt in ./pi/venv_picframe/lib/python3.11/site-packages (1.6.1)

GitRayc, 2023-12-11:
Sorry all,  my latest update
I looked at the python script suggested and read it more carefully. 
I overlooked the comment on this line:

client.connect("pictureframe.local", 1883, 60) # server address here

So I changed the "60" to be my ip address 192.168.1.209
I not reads:
client.connect("pictureframe.local", 1883, 192.168.1.209) # server address here

Once I did that I no longer got the error message "No Module named 'paho'"

But now I get the error message:
Syntax error: invalid syntax. Perhaps you forgot a comma?

What is the server address entry that is needed?

paddywwoof, 2023-12-11:
Hi, it's a long time since I've messed around with mqtt but maybe try `client.connect("192.168.1.209", 1883, 60)` or `client.connect("localhost", 1883, 60)` if the mqtt server is on the same machine as the client. Try the docs [here](https://pypi.org/project/paho-mqtt/#connect-reconnect-disconnect), you may be able to figure something out. I tend to use the http connection as I'm not running with home assistant.

GitRayc, 2023-12-11:
Thank you so much for your help.  You are very kind.
Unfortunately I am just in over my head, I am afraid.
The defualt when you install and follow the web page is:

client.connect("pictureframe.local", 1883, 60)
But that will cause the paho module error.

Just wanted a way to set the picture subdirecty which you seem do through Mqtt  so I can choose from a few. 
But Mqtt seems to be beyond my level of expertice.   (sad face)  lol

paddywwoof, 2023-12-11:
Hi, maybe the web page route would be easier then. I the configuration.yaml file set the mqtt option to False and the http option to True. Once the picframe has restarted it should be running a very simple web server at the address `192.168.1.209:9000` (or whatever IP address your RPi has and whatever port number is set in the configuration.yaml file). The web page that appears should give you the option to enter a subdirectory, as well as various other things. Let me know if you try this and get stuck.

Dropbear67, 2023-12-19:
Hi. I have an external MQTT broker set up for Home Assistant and I can connect to the broker using the test script you get to from digital frame web site 

client = mqtt.Client()
client.username_pw_set( "user" , "passwd" )
client.connect( "192.168.XX.XX", 1883, 60)

where obviously XX.XX is the address of the MQTT server

so with the default install it seems pahoo etc is installed and working. Of course had to activate the venv on RPI 5 and Bookworm