helgeerbe/picframe #354: Can't get portrait mode working on Pi5

Notrial, 2023-12-12:
I did as per instructions here: https://www.thedigitalpictureframe.com/raspberry-pi-digital-picture-frame-portrait-orientation/

Appended run_start.py with 

`from picframe import start
import os
os.system("DISPLAY=:0 xrandr --output HDMI-0 --rotate left")
start.main()`

But after I execute:

`sudo xinit /usr/bin/python3 /home/pi/picframe_data/run_start.py /home/pi/picframe_data/config/configuration.yaml`

I get the following error:

[image]

Notrial, 2023-12-13:
OK so after fiddling with this for hours...in case someone runs into same problem.
I did two things...

I edited (appended at the end) cmdline.txt in /boot/firmware with video=HDMI-A-1:1920x1200@60,rotate=270
That rotated the CLI.

Then I changed the start_picframe.sh 
This line:
wlr-randr --output HDMI-A-1 --transform 270 --on

That did the trick!

adammhaile, 2024-11-17:
see #401

closed 2026-05-06 by github-actions[bot]