helgeerbe/picframe #355: Problem with Picture directory subfolders

GitRayc, 2023-12-12:
Wandering if anyone else has run into this problem, or has any suggestions for me.

Brand new install of Bookworm on Pi4 and complete new picture frame install.    Started clean to try to ensure there were no old or corrupted files hanging around causing problems.

The issue is that I have a Subdirectory that includes only Family photos in the picture directory.   My root contains just general landscapes etc.

My .yaml file points to the root directory of course and my subdirectory entry is set to "".

When I run picframe all works beautifully but it includes and shuffles all the images in the subfolder even though the subdirectory parameter is set to "".    I can not seem to get it to ignore that directory.  If I look in the .db3 file it list all the files in that subdirectory as well as the root.

The other problem I have, which is think it somehow related; is that if I add new photos to the root directory they will never be recognized.  Not even after a reboot.   The only way I can have them recognized and picked up is to delete the .db3 database file all together.    Then when I stop and restart picfame a brand new .db3 file is created of course and all the new images are the recognized, but only then.

Anyone have any ideas.

Thanks all.

jgodfrey, 2023-12-12:
I think you have a misunderstanding of how the `subdirectory` feature works. A few points of interest:

- By default, the software is designed to present *everything* (including any subdirectory contents) found beneath the defined picture directory. That allows you to "organize" your pictures into folder structures (if you want to) and still have them all be displayed without requiring any complicated configurations
- When you defined the `subdirectory` property, you're telling the software to only display the contents of the specified subdirectory (and any subdirectories that it may include).
- The database should always contain references to EVERYTHING in your main pictures folder (and any subdirectories). What's in the database and what is actually displayed on the frame in any given slideshow is not necessarily the same thing depending on how you've filtered the images via other configuration options (including the subdirectory setting)
- New photos not being recognized is abnormal and should not happen. If that's the case then something is wrong and it'd be interesting to see what's getting logged. IIRC, the discovery of new photos should cause some log info to be generated.

GitRayc, 2023-12-12:
Thank you so much for clearly explaining this to me.  Yes I misunderstood.   Thanks!

As far as new photos added not being recognized as they should, is there anything I might check to see what could be going wrong?  Or is there anything I could send you to help troubleshoot?

I might mentioned that I save and store all my pics on a USB thumbdrive that automounts on boot.   Could this be the problem?  I didnt think that would matter.    I just thought that might be better for my system ssd health.     Do you think it would be better to have them installed directly on the ssd card and live in the default pictures directory.

Thanks for being patient with a newby with just enough knowledge to get into trouble.

Your support is so appreciated.

jgodfrey, 2023-12-12:
While I'm not aware of any specific issue with an auto-mounted USB thumb drive, I guess I'd consider that a atypical part of your setup. I'm curious what happens if you put your photos on the internal card - even if only temporarily? Does the problem go away? If so, that'd give us a direction for further troubleshooting.

GitRayc, 2023-12-12:
I will certainly try that!

GitRayc, 2023-12-13:
So as a test (as suggested) I copied everything from my Thumbdrive over to the default Picture directory and then updated my .yaml accordingly.

I created a new image and dropped it into the Picture directory. The data base updated immediately to list the new image.

So it does have something to do with storing images on the Thumbdrive.   

Good to know.

Ray

paddywwoof, 2023-12-13:
Possibly related to how the software mapping the drive to the filesystem passes info back and forth.

The first option might be to try to 'trick' the drive software as Jeff suggested on an earlier occasion by
```bash
touch /mountedUSBdirectory/
```
after copying files there.

If that doesn't work it might be worth having an option to recursively find the latest modification date for files in each directory rather than just the directory.

closed 2023-12-14 by GitRayc