helgeerbe/picframe #359: Screen cropped to bottom left corner after service restart

Dropbear67, 2023-12-19:
Issue: Restart of service on RPi 5 with a 4K external monitor hooked up to HDMI port 1 causes images to be displayed tiny in bottom left of screen

Installed PicFrame as per instructions 
https://www.thedigitalpictureframe.com/how-to-build-a-gorgeous-digital-photo-frame-with-a-raspberry-pi-based-on-2023-bookworm/

On Full Reboot of Pi - Images are displayed normally and correctly.
On Service Restart images are tiny as described - screenshot attached

Version:
picframe version:  2023.07.26.post1+2.ge437440

Checking required packages......
PIL :  9.5.0
pi3d :  2.51
yaml :  6.0.1
paho.mqtt :  1.6.1
iptcinfo3 :  2.1.4
numpy :  1.26.2
ninepatch : installed, but no version info
pi_heif :  0.14.0
defusedxml :  0.7.1

Python 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0] on linux

[image]
[image]

Dropbear67, 2023-12-20:
I forgot to add here that I am SSHing into the Pi when I go to run these commands.
Following the instructions on thedigitalpictureframe.com link above I did this

"Select your Raspberry Pi model, and choose Raspberry Pi OS Lite (64-bit) under “Operating System” and “Raspberry Pi (other)”."

Given this I do not seem to have X11 installed. I had a look at some other issues and one of the hints was to run wayfire-pi and when I try this I get command not found.. weird.

paddywwoof, 2023-12-20:
@Dropbear67 this does look strange. There are a few things that need to be sorted out, but generally speaking wayland isn't really expecting to run graphical user interface headless so there are loop-holes and different behaviour depending on how applications are started (keyboard, systemd, bash script etc). Hopefully the people that use raspberry pis commercially for displays etc will start to weigh in and the problems will get sorted out.

There must be other things that need to be installed to get the desktop working again on the lite SD image.

Dropbear67, 2023-12-20:
"There must be other things that need to be installed to get the desktop working again on the lite SD image."

Would it be better to start with an image other than lite to make sure everything is properly installed? 
I would just like to take the opportunity to thank you for your efforts to support the community, and answering our questions. I can understand how thankless the job must be sometimes - especially when new releases of hardware and the OS come along and break all the hard work.

It's not the end of the world to "sudo reboot" to get the frame running again. It would just be nice to be able to stop and start the service thats all. I don't know whether doing it from a ssh session and not from a keyboard directly plugged into the pi is complicating matters .. I must admit to not knowing much about desktops on linux.

Dropbear67, 2023-12-21:
Update: I changed the configuration.yaml file to the following;
  display_x: 0                            # offset from left of screen (can be negative)
  display_y: 0                            # offset from top of screen (can be negative)
  display_w: 3840                         # width of display surface (null->None will use max returned by hardware)
  display_h: 2160                         # height of display surface

I then did a systemctl --user restart picframe.service

It almost worked.. the display was kind of full screen but off set maybe 5% in the horizontal and vertical.. almost like some kind of overscan type issue (so a 5% black border on the top and left hand side of the screen). How annoying.

paddywwoof, 2023-12-21:
With the earlier RPis there was an OpenGL `GL_MAX_TEXTURE_SIZE` of 2048, which is smaller than a 4k screen and (most) image files that people might be trying to display. To avoid all kinds of undefined (C) behaviour picframe resizes images before it tries to feed them into the GPU. I probably need to review that process in view of the new GPU on the RPi 5

Dropbear67, 2023-12-21:
> @Dropbear67 this does look strange. There are a few things that need to be sorted out, but generally speaking wayland isn't really expecting to run graphical user interface headless so there are [...]

hi Paddy thanks for the reply. just to clarify the images are appearing fine after a reboot and initial load of the game. it's just after a service restart or stop/start that the small image issue takes place. i think it might be something to do with the HDMI resolution getting confused after the service stops. it's still in 4K but the software thinks it's 1080p or something weird like that

Dropbear67, 2023-12-22:
I believe I have gone someway into solving this ..

I can simulate the issue by doing the following;

in an ssh-window I type 
$ export WAYLAND_DISPLAY="wayland-1"; wlr-randr --output HDMI-A-1 --off

This will power down my HDMI Monitor (put it into sleep mode) but unfortunately it also kills picframe.. (if you could solve this issue - that would be brilliant)

in the same window I type
$ export WAYLAND_DISPLAY="wayland-1"
$ wlr-randr --output HDMI-A-1 --on --mode "3840x2160,59.981998"
$ source /home/ian/venv_picframe/bin/activate 
$ picframe &

This successfully starts picframe however the image is boxed in the bottom left corner of the display like in the screenpics in my initial post..

now if I wait 2-3 seconds between source command and the picframe command it starts normally.

I modified by start_picframe.sh command to look like this now 

#!/bin/bash
export SDL_VIDEODRIVER=wayland 
export WAYLAND_DISPLAY="wayland-1"
wlr-randr --output HDMI-A-1 --on --mode "3840x2160,59.981998"
sleep 2
source /home/ian/venv_picframe/bin/activate 
picframe &  

and it works fine .. starts up and displays fine..

I believe if you can solve the wlr-randr --output HDMI-A-1 --off command crashing picframe we will just about have our Bookworm issues sorted.

paddywwoof, 2023-12-22:
Hi, this might be really useful info. I've been trying to get the screensaver to work with simulated mouse or keyboard input and found it was fine if I typed the contents of start_picframe.sh in a terminal but not if I ran it. I will see what a few sleeps do.

paddywwoof, 2023-12-22:
Oh well, quick test and `sleep` doesn't fix the ydotool idea. It's probably a user/process problem.

Dropbear67, 2023-12-26:
> Oh well, quick test and `sleep` doesn't fix the ydotool idea. It's probably a user/process problem.

Merry Christmas to you and your family Paddy.. Any ideas if the picframe service not surviving the HDMI display being turned off is something that might be fixable? Unfortunately i am not seeing anything written to the DEBUG logs before the process dies.