helgeerbe/picframe #360: Menu Buttons With Touch Not Working

Ben10lightning, 2023-12-21:
I have setup my picture frame with a usb touchscreen and a pi zero 2 w and the touch input mostly works except for the menu. I can touch the top of the screen to make the menu show up, however the buttons do nothing when I tap them. All other touch functionality works as expected. 
Here is my configuration.yaml file
`viewer:
  blur_amount: 10                         # default=12,  larger values than 12 will increase processing load quite a bit
  blur_zoom: 1.0                          # default=1.0, must be >= 1.0 which expands the backgorund to just fill the space around the image
  blur_edges: False                       # default=False, use blurred version of image to fill edges - will override FIT = False
  edge_alpha: 0.5                         # default=0.5, background colour at edge. 1.0 would show reflection of image
  fps: 20.0                               # default=20.0
  background: [0.2, 0.2, 0.3, 1.0]        # default=[0.2, 0.2, 0.3, 1.0], RGBA to fill edges when fitting
  blend_type: "blend"                     # default="blend", choices={"blend", "burn", "bump"}, type of blend the shader can do
  font_file: "/home/pi/picframe_data/data/fonts/NotoSans-Regular.ttf"
  shader: "/home/pi/picframe_data/data/shaders/blend_new"
  show_text_fm: "%b %d, %Y"               # default "%b %d, %Y", format to show date over the image
  show_text_tm: 20.0                      # default=20.0, time to show text over image with file name
  show_text_sz: 40                        # default=40, text character size
  show_text: "date location"  # default="title caption name date folder location", show text, include combination of words: title, caption name, date, location, folder
  text_justify: "L"                       # text justification L, C or R
  text_bkg_hgt: 0.25                      # default=0.25 (0.0-1.0), percentage of screen height for text background texture
  text_opacity: 1.0                       # default=1.0 (0.0-1.0), alpha value of text overlay
  fit: True                              # default=False, True => scale image so all visible and leave 'gaps'
                                          #                False => crop image so no 'gaps'
  kenburns: False                         # default=False, will set fit->False and blur_edges->False
  display_x: 0                            # offset from left of screen (can be negative)
  display_y: 0                            # offset from top of screen (can be negative)
  display_w: null                         # width of display surface (null->None will use max returned by hardware)
  display_h: null                         # height of display surface
  display_power: 0                        # default=0. choices={0, 1}, 0 will use legacy `vcgencmd` and 1 will use `xset` to blank the display
  use_glx: False                          # default=False. Set to True on linux with xserver running

  mat_images: 0.01                        # default=0.01, True, automatically mat all images. False, don't automatically mat any images. Real value, auto-mat all images with aspect ratio difference > than value
  mat_type: null                          # default=null, A string containing the mat types to choose from when matting images. It can consist of any or
                                          # all of 'float float_polaroid float_color_wrap single_bevel double_bevel double_flat' (null or '' will use all mat types)
  outer_mat_color: null                   # default=null, Color of the outer mat as an RGB list. null will auto-select a reasonable color based on the image.
  inner_mat_color: null                   # default=null, Color of the inner mat as an RGB list. null will auto-select a reasonable color based on the image.
  outer_mat_border: 0                    # default=75, Minimum outer mat border in pixels
  inner_mat_border: 40                    # default=40, Minimum inner mat border in pixels (for styles that use it)
  outer_mat_use_texture: True             # default=True, True uses a texture for the outer mat. False creates a solid-color outer mat.
  inner_mat_use_texture: False            # default=False, True uses a texture for the inner mat. False creates a solid-color inner mat.
  mat_resource_folder: "/home/pi/picframe_data/data/mat" # Folder containing mat image files

  show_clock: True                       # default=False, True shows clock overlay. False does not show clock overlay
  clock_justify: "R"                      # default="R", clock justification L, C, or R
  clock_text_sz: 120                      # default=120, clock character size
  clock_format: "%I:%M"                   # default="%I:%M", strftime format for clock string
  clock_opacity: 1.0                      # default=1.0 (0.0-1.0), alpha value of clock overlay

  menu_text_sz: 40                        # default=40, menu character size
  menu_autohide_tm: 10.0                  # default=10.0, time in seconds to show menu before auto hiding (0 disables auto hiding)
  geo_suppress_list: []                   # default=None, substrings to remove from the location text

model:
  pic_dir: "/home/pi/Pictures"                   # default="/home/pi/Pictures", root folder for images
  deleted_pictures: "/home/pi/DeletedPictures"   # move deleted pictures here
  follow_links: False                     # default=False, By default, picframe will not walk down into symbolic links that resolve to directories. Set follow_links to True to visit directories pointed to by symlinks, on systems that support them.
  no_files_img: "/home/pi/picframe_data/data/no_pictures.jpg" # default="PictureFrame2020img.jpg", image to show if none selected
  subdirectory: ""                        # default="", subdir of pic_dir - can be changed by MQTT"
  recent_n: 7                             # default=7 (days), when shuffling file change date more recent than this number of days play before the rest
  reshuffle_num: 1                        # default=1, times through before reshuffling
  time_delay: 50.0                       # default=200.0, time between consecutive slide starts - can be changed by MQTT
  fade_time: 5.0                         # default=10.0, change time during which slides overlap - can be changed by MQTT"
  shuffle: True                           # default=True, shuffle on reloading image files - can be changed by MQTT"
  sort_cols: 'fname ASC'                  # default='fname ASC' can be any columns in the table with optional ASC or DESC separated by commas
                                          # fname, last_modified, file_id, orientation, exif_datetime, f_number,
                                          # exposure_time, iso, focal_length, make, model, lens, rating,
                                          # latitude, longitude, width, height, title, caption, tags,
                                          # is_portrait, location
  image_attr: [                           # image attributes send by MQTT, Keys are taken from exifread library, "PICFRAME GPS" is special to retrieve GPS lon/lat, "PICFRAME LOCATION" is special to retrieve geo reverse (load_geoloc hast to be True)
    "PICFRAME GPS",
    "PICFRAME LOCATION",
    "EXIF FNumber",
[... 162 lines cut ...]
  use_http: False                         # default=False. Set True to enable http NB THIS SERVER IS FOR LOCAL NETWORK AND SHOULD NOT BE EXPOSED TO EXTERNAL ACCESS
  path: "/home/pi/picframe_data/html"            # path to where html files are located
  port: 9000                              # port used to serve pages by http server < 1024 requires root which is *bad* idea
  use_ssl: False
  keyfile: "path/to/key.pem"              # private-key
  certfile: "path/to/cert.pem"            # server certificate

peripherals:
  input_type: "touch"                        # default=null, valid options: {null, "keyboard", "touch", "mouse"}
  buttons:
    pause:                                # pause/unpause the show
      enable: True                        # default=True
      label: "Pause"                      # default="Pause"
      shortcut: " "                       # default=" "
    display_off:                          # turn off the display (when off, any input from selected peripheral will turn it back on)
      enable: True                        # default=True
      label: "Display off"                # exit PictureFrame
      enable: True                       # default=False
      label: "Exit"                       # default="Exit"
      shortcut: "e"                       # default="e"
    power_down:                           # power down the device, uses sudo
      enable: True                       # default=False
      label: "Power down"                 # default="Power down"
      shortcut: "p"                       # default="p"
`

helgeerbe, 2023-12-21:
Hi,

unfortunately I don't own a touch screen and it's hard to say why this is not working. Which OS are you using? Is the touchscreen calibrated and works well, if you use the desktop and push browser buttons or so?

Ben10lightning, 2023-12-21:
I am using raspberry pi OS legacy 32 bit. The touchscreen position is accurate and works fine in the OS and going to the next or previous pictures.

Ben10lightning, 2023-12-21:
Didn't mean to close, I am using raspberry Pi OS Legacy 32 bit. The touchscreen works fine in the other parts of the OS and allows me to go to the next or previous picture without issue.

Ben10lightning, 2023-12-22:
After more testing I believe I am having a similar issue to the one described in the known issues section of the peripherals page on the wiki. When I set the input method to mouse and try to use the touchscreen, the cursor position is not anywhere near where I am touching on the screen, even though in the desktop the cursor closely matches my finger position.

paddywwoof, 2023-12-22:
With the move to SDL2 the whole pi3d mouse side might need to be reviewed, but if you are using legacy OS you are probably using the X11 system. i.e. not setting use_sdl2 in configuration.yaml

helgeerbe, 2023-12-22:
Well, have you ever calibrated the touchscreen?
https://forums.raspberrypi.com/viewtopic.php?t=358581

Maybe this will solve your issue.

h0d3nt3uf3l, 2025-01-04:
Greetings!
Sadly I've the same error today.

I'm using a RPI4 with Raspberry Pi OS (64-bit, Released: 2024-11-19). It's using Wayland with labwc as compositor.
Attached is a Iiyama North America PL2453MT in Portrait-Mode (works ot of the box with the rotation made on the Desktop, the touch function was rotated as well)

When I'm using "mouse" as input the pointer will be displayed correctly under my finger.
It's a very confusing error because sometimes I could switch the pictures, sometimes not. Another time I was able to swtich the pictures with a wipe over the display.

Want to use picframe as "screensaver" while not using the Display. It's mainly used as Controlpanel for Homeassistant, so an ability to touch and exit the programm would be nice :)
The screensaver-part would be done with a little shell-script and logging for touch inputs.