helgeerbe/picframe #364: Pi 4 doesn't seem to be compatible with Buster on a fresh install

sapnho, 2023-12-31:
Last night, I updated [one of my earliest articles](https://www.thedigitalpictureframe.com/how-to-set-up-your-raspberry-pi-for-your-digital-picture-frame/) to reflect the fact that Buster isn't available in the Raspberry Pi Imager software anymore. Upon testing, I discovered that the Pi 4 apparently now requires Bullseye or Bookworm.

I also saw [this post](https://forums.raspberrypi.com/viewtopic.php?t=362498&sid=6542a2dd8b9dcd3ce45a41813a84f2d4) on the Pi forum.

Good thing, we almost have Pi3D PictureFrame running on Bookworm.

Can anybody confirm the Pi 4 incompatibility with Buster?

sapnho, 2023-12-31:
I found a more recent Buster image that works on the Pi4: https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2021-05-28/

That works!

Dropbear67, 2024-01-01:
> Can anybody confirm the Pi 4 incompatibility with Buster?

yes I can. There was a way around it by overwriting files in the boot directory but hopefully your new image fixes the need to do that

sapnho, 2024-01-01:
Yes, it does.

sapnho, 2024-06-05:
I guess we can close this issue.