helgeerbe/picframe #365: enabling the G2 GL Fake KMS sends Rpi4 to terminal on reboot, no GUI

talljon, 2024-01-11:
Hello, I am using this guide to run a Rpi4 with Pi3D for a digital picture frame:  

https://www.thedigitalpictureframe.com/how-to-set-up-your-raspberry-pi-for-your-digital-picture-frame/

I am using a fresh install of buster on a Rpi4.  I have reformatted/reinstalled 3x and I always get hung up on the same spot.

Following the above guide everything is spot on until I enable the G2 GL Fake KMS. After that, the Rpi will no longer boot to the GUI (just goes to terminal). And after I install picframe reboot and run, instead of the happy "no pictures selected" image I am supposed to get on the Rpi, I get this in my ssh terminal:

X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 5.4.0-109-generic armv8l Raspbian
Current Operating System: Linux mug 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=DC:A6:32:95:A2:E3 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=1182eb90-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
Build Date: 04 August 2022  12:56:42PM
xorg-server 2:1.20.4-1+rpt4+deb10u5 (https://www.debian.org/support) 
Current version of pixman: 0.36.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Tue Oct 25 11:03:34 2022
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(II) modeset(0): Initializing kms color map for depth 24, 8 bpc.
/usr/bin/python3: can't open file '/home/pi/picframe_data/run_start.py': [Errno 2] No such file or directory
xinit: connection to X server lost

waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.

I know there is probably something simple that I'm missing, so any help would be appreciated.  I am not a software engineer, I am an airline pilot trying to figure this out.  I am sorry if I did not post this correctly.  Thank you for your help.

**What are your OS, Python and picframe versions(please complete the following information):?**

- Hardware: Rpi4
- OS: Buster
- Python: 
- picframe (run `picframe -v` if possible): 
- 
pi@pictureframe:~ $ picframe -v
INFO:start.py:starting ['/usr/local/bin/picframe', '-v']
picframe version:  2023.07.26.post1

Checking required packages......
PIL :  9.5.0
pi3d :  2.51
yaml :  6.0.1
paho.mqtt :  1.6.1
iptcinfo3 :  2.1.4
numpy :  1.16.2
ninepatch : installed, but no version info
pi_heif :  0.12.0
defusedxml :  0.7.1
pi@pictureframe:~ $ 

```yaml

pi@pictureframe:~ $ sudo xinit /usr/bin/python3 /home/pi/picframe_data/run_start.py /home/pi/picframe_data/config/configuration.yaml

X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 5.4.0-109-generic aarch64 Debian
Current Operating System: Linux pictureframe 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708_fb.fbswap=1 smsc95xx.macaddr=D8:3A:DD:90:75:BA vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=f818b664-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=US
Build Date: 04 April 2023  07:50:56AM
xorg-server 2:1.20.4-1+rpt4+deb10u9 (https://www.debian.org/support) 
Current version of pixman: 0.36.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Wed Jan 10 18:26:05 2024
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
/usr/bin/python3: can't open file '/home/pi/picframe_data/run_start.py': [Errno 2] No such file or directory
xinit: connection to X server lost

waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.

paddywwoof, 2024-01-11:
@talljon sorry, this must be frustrating for you. Recently I've been working on getting things working with the latest Bookworm OS but it sounds like there might be some issues that have kept into Buster I will try following Wolfgang's instructions for that when I get home tonight.

The error sounds like `picframe_data/run_start.py` wasn't installed. Could you check that directory? When you ran the line to install picframe did everything work as expected?

Paddy

talljon, 2024-01-11:
Paddy,

Thank you for the reply.  Like I said, I am not a software person so hopefully I will answer this correctly.  I think you are correct that it is not installed.  When I look for picframe_data/run_start.py with the find command, here is what I see:

pi@pictureframe:~ $ find . picframe_data/run_start.py
.
./Documents
./.Xauthority
./.bashrc
./DeletedPictures
./.bash_history
./.config
./.config/pulse
./.config/pulse/1242a69d5bd84fa4a0d04184ca65dc92-default-sink
./.config/pulse/1242a69d5bd84fa4a0d04184ca65dc92-stream-volumes.tdb
./.config/pulse/1242a69d5bd84fa4a0d04184ca65dc92-default-source
./.config/pulse/1242a69d5bd84fa4a0d04184ca65dc92-device-volumes.tdb
./.config/pulse/1242a69d5bd84fa4a0d04184ca65dc92-card-database.tdb
./.config/pulse/cookie
./.config/user-dirs.locale
./.config/user-dirs.dirs
./.config/lxpanel
./.config/lxpanel/LXDE-pi
./.config/lxpanel/LXDE-pi/panels
./Music
./Videos
./Downloads
./.xsession-errors.old
./.local
./.local/share
./.local/share/applications
./.local/share/desktop-directories
./.profile
./.xsession-errors
./Templates
./.gnupg
./.gnupg/private-keys-v1.d
./Public
./Pictures
./.cache
./.cache/openbox
./.cache/openbox/openbox.log
./.cache/openbox/sessions
./.cache/lxsession
./.cache/lxsession/LXDE-pi
./.cache/lxsession/LXDE-pi/run.log
./.cache/menus
./.cache/menus/6cc2bb09817b99bdf0f62d9abf3c60be
./.cache/menus/3d5ec69d3d984f5df1f165835f6d03ea
./.bash_logout
./Bookshelf
./Bookshelf/000_RPi_BeginnersGuide_DIGITAL.pdf
./Desktop
find: ‘picframe_data/run_start.py’: No such file or directory
pi@pictureframe:~ $ 

Again, my apologies if this is not where I'm suppose to be looking.

Jon

paddywwoof, 2024-01-11:
As you say there doesn't look to be a directory `~/picframe_data/` One thing that is worth checking is the point in your installation history where you ran `picframe -i .` Note that there is a full stop on the end of that line, a short cut for the explicit `/home/pi/` which is the active directory after rebooting. You can see a list of everything you've typed at the terminal by running `cat /home/pi/.bash_history` but, obviously, that won't show you what the response was.

The other thing that would be worth checking is whether pi3d is correctly installed. You can do that by
```bash
python3
>>> import pi3d
```
type CtrlZ to close the interactive python prompt. Hopefully there won't be any error messages.

talljon, 2024-01-11:
Thank you for all your help.

Indeed there is no directory `~/picframe_data/ `

Here is my screen:  
`ratlab@Rats-MacBook-Pro` `~ % ~/picframe_data/`
`zsh: no such file or directory: /Users/ratlab/picframe_data/`

It also doesn't look like it was installed correctly if that is what I'm deciphering from the python3 command:

>>> import pi3d
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'pi3d'
>>> ratlab@Rats-MacBook-Pro ~ % ~/picframe_data/
  File "<stdin>", line 1
    ratlab@Rats-MacBook-Pro ~ % ~/picframe_data/
                            ^
SyntaxError: invalid syntax
>>> zsh: no such file or directory: /Users/ratlab/picframe_data/
  File "<stdin>", line 1
    zsh: no such file or directory: /Users/ratlab/picframe_data/
            ^
SyntaxError: invalid syntax
>>> 
zsh: suspended  python3
ratlab@Rats-MacBook-Pro ~ % 

At least I know where I stand now with why it is not functioning.  I'm not too sure where to go next, but willing to try anything.

paddywwoof, 2024-01-11:
Hi, that `ratlab@Rats-MacBook-Pro ~ %` is a little confusing. Are you `ssh`ing into your raspberry pi? You showed `pi@pictureframe: $` before.

Open a terminal in RPi and check back over `/home/pi/.bash_history` to see what the `picframe` command was, and the ones for install it. Let me know what you see.

talljon, 2024-01-11:
Sorry, that was my ineptness at being a software person.  I will check this now and be right back with you.

talljon, 2024-01-11:
Problem solved.  Here is where I went wrong, because I am good at following directions, but not a software programmer.

Please see the attached screenshot of the directions from the guide I was using:

[image]

My understanding was to copy and paste the entire block into terminal, which worked.  The the Rpi rebooted as the second to last line instructs it to.  I did not realize that I needed to reenter the last line `picframe -i .` after the reboot and then finish the installation.  

So it comes down to me not understanding the directions clearly.  I will email Wolfgang and have him correct this by separating the last command in his tutorial.

My apologies about taking up your free time and I thank you for all that I have learned throughout this process.

Jon

closed 2024-01-11 by talljon

paddywwoof, 2024-01-11:
Ah, that makes sense - the linux command line and bash syntax is rather cryptic to say the lease. If you run into any other problems don't hesitate to post a question. Github issue forums are a really excellent way for people to find solutions to problems that they have, so you are really contributing to the documentation.