helgeerbe/picframe #366: How to access /home/pi directory on a corrupted raspberry pi ssd card

GitRayc, 2024-01-19:
Yes I do back up but:

During a recent back up something went wrong and my ssd card failed or got corrupted or something..   Long story short it will no longer boot my pi.  So I reverted back to previous backup.   However this backup is missing a couple of dozen new image files i had in my pictures directory.

I hoped I could access those files in my home directory by plugging the card in as a usb drive on my mac.   However when I do that, as you know I only see the boot bootfs files and have not access to the files in my home directory on the card.  Nor can you even see them as I assume that they are a different partition on the ssd card.

Is there some way to view/access my home directory on that ssd card  (/home/pi/pictures?

Obviously they are there on that ssd card but how can I view that directory/partition?

Or, is there a way of reinstalling the OS onto the card without overwriting the data (home directory)

Help  Please!

paddywwoof, 2024-01-19:
I don't have a Mac but it does have a 'nix type of OS, and on Linux desktop I get two file systems mounted the rootfs can be read and edited. There are possibly security implications of being able to do this so maybe Mac OS requires an extra step? Or there might be a corrupted SD card, in which case you might be able to install an application on your Mac to recover pictures from a failed SD card such as you might use in a camera. That would do a low level scrape like forensic police use to rever deleted files.

helgeerbe, 2024-01-19:
This worked for me, if you have an intel mac: https://www.switchdoc.com/2016/01/tutorial-repairing-corrupted-sd-cards-for-the-raspberry-pi-on-mac/

GitRayc, 2024-01-19:
Thank you all.   The site suggested is currently down for upgrades.   But I found this tool which you can install on a windows machine.   Amazing tool..   Accesses' everything.   I hope this can help others!!!!    What a life saver!!!!

DiskInternals Linux Reader

Fast and reveals everything!

Ray

closed 2024-02-19 by helgeerbe