helgeerbe/picframe #370: Picframe crash on database access

Mettbrot, 2024-02-06:
**Describe the bug**
After a while, picframe stops and the console shows the log below. The Database does not contain data up to this point, but grows to 5MB. (I downloaded it to windows and checked with SQLiteBrowser)

/mnt/bilder is a read only mounted network share with ~ 40000 pictures.

I want to filter on the Tag "Favourites", was I correct to put the option in the yaml file?

**To Reproduce**
start picframe and wait for it to crash

**What are your OS, Python and picframe versions(please complete the following information):?**
RaspberryPi 3B with RaspberryOS: Debian version: 12 (bookworm)

```yaml
viewer:
  blur_amount: 12                         # default=12,  larger values than 12 will increase processing load quite a bit
  blur_zoom: 1.0                          # default=1.0, must be >= 1.0 which expands the backgorund to just fill the space around the image
  blur_edges: False                       # default=False, use blurred version of image to fill edges - will override FIT = False
  edge_alpha: 0.0                         # default=0.5, background colour at edge. 1.0 would show reflection of image
  fps: 20.0                               # default=20.0
  background: [0.0, 0.0, 0.0, 1.0]        # default=[0.2, 0.2, 0.3, 1.0], RGBA to fill edges when fitting
  blend_type: "blend"                     # default="blend", choices={"blend", "burn", "bump"}, type of blend the shader can do
  font_file: "/home/pi/picframe_data/data/fonts/NotoSans-Regular.ttf"
  shader: "/home/pi/picframe_data/data/shaders/blend_new"
  show_text_fm: "%b %d, %Y"               # default "%b %d, %Y", format to show date over the image
  show_text_tm: 20.0                      # default=20.0, time to show text over image with file name
  show_text_sz: 40                        # default=40, text character size
  show_text: "date location"  # default="title caption name date folder location", show text, include combination of words: title, caption name, date, location, folder
  text_justify: "C"                       # text justification L, C or R
  text_bkg_hgt: 0.25                      # default=0.25 (0.0-1.0), percentage of screen height for text background texture
  text_opacity: 1.0                       # default=1.0 (0.0-1.0), alpha value of text overlay
  fit: True                               # default=False, True => scale image so all visible and leave 'gaps'
                                          #                False => crop image so no 'gaps'
  kenburns: False                         # default=False, will set fit->False and blur_edges->False
  display_x: 0                            # offset from left of screen (can be negative)
  display_y: 0                            # offset from top of screen (can be negative)
  display_w: null                         # width of display surface (null->None will use max returned by hardware)
  display_h: null                         # height of display surface
  display_power: 0                        # default=0. choices={0, 1}, 0 will use legacy `vcgencmd` and 1 will use `xset` to blank the display
  use_glx: True                           # default=False. Set to True on linux with xserver running

  mat_images: False                        # default=0.01, True, automatically mat all images. False, don't automatically mat any images. Real value, auto-mat all images with aspect ratio difference > than value
  mat_type: null                          # default=null, A string containing the mat types to choose from when matting images. It can consist of any or
                                          # all of 'float float_polaroid float_color_wrap single_bevel double_bevel double_flat' (null or '' will use all mat types)
  outer_mat_color: null                   # default=null, Color of the outer mat as an RGB list. null will auto-select a reasonable color based on the image.
  inner_mat_color: null                   # default=null, Color of the inner mat as an RGB list. null will auto-select a reasonable color based on the image.
  outer_mat_border: 75                    # default=75, Minimum outer mat border in pixels
  inner_mat_border: 40                    # default=40, Minimum inner mat border in pixels (for styles that use it)
  outer_mat_use_texture: True             # default=True, True uses a texture for the outer mat. False creates a solid-color outer mat.
  inner_mat_use_texture: False            # default=False, True uses a texture for the inner mat. False creates a solid-color inner mat.
  mat_resource_folder: "/home/pi/picframe_data/data/mat" # Folder containing mat image files

  show_clock: False                       # default=False, True shows clock overlay. False does not show clock overlay
  clock_justify: "R"                      # default="R", clock justification L, C, or R
  clock_text_sz: 120                      # default=120, clock character size
  clock_format: "%I:%M"                   # default="%I:%M", strftime format for clock string
  clock_opacity: 1.0                      # default=1.0 (0.0-1.0), alpha value of clock overlay

  menu_text_sz: 40                        # default=40, menu character size
  menu_autohide_tm: 10.0                  # default=10.0, time in seconds to show menu before auto hiding (0 disables auto hiding)
  geo_suppress_list: []                   # default=None, substrings to remove from the location text

model:
  pic_dir: "/mnt/bilder"                   # default="/mnt/bilder", root folder for images
  deleted_pictures: "/home/pi/DeletedPictures"   # move deleted pictures here
  follow_links: False                     # default=False, By default, picframe will not walk down into symbolic links that resolve to directories. Set follow_links to True to visit directories pointed to by symlinks, on systems that support them.
  no_files_img: "/home/pi/picframe_data/data/no_pictures.jpg" # default="PictureFrame2020img.jpg", image to show if none selected
  subdirectory: ""                        # default="", subdir of pic_dir - can be changed by MQTT"
  recent_n: 7                             # default=7 (days), when shuffling file change date more recent than this number of days play before the rest
  reshuffle_num: 1                        # default=1, times through before reshuffling
[... 73 lines cut ...]
      label: "Exit"                       # default="Exit"
      shortcut: "e"                       # default="e"
    power_down:                           # power down the device, uses sudo
      enable: False                       # default=False
      label: "Power down"                 # default="Power down"
      shortcut: "p"                       # default="p"

```

````txt
Exception in thread Thread-1 (__loop):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/picframe/lib/python3.11/site-packages/picframe/image_cache.py", line 54, in __loop
    self.update_cache()
  File "/home/pi/picframe/lib/python3.11/site-packages/picframe/image_cache.py", line 91, in update_cache
    self.__insert_file(file)
  File "/home/pi/picframe/lib/python3.11/site-packages/picframe/image_cache.py", line 415, in __insert_file
    self.__db.execute(meta_insert, vals)
sqlite3.ProgrammingError: Error binding parameter 9: type 'tuple' is not supported

````

jgodfrey, 2024-02-06:
That stack trace indicates a problem with the constructed SQL statement when attempting to insert meta-data for a given image into the DB. It'd probably be fairly easy to find and fix with access to the offending image. Do you happen to know what image is causing the issue?

jgodfrey, 2024-02-06:
Assuming you don't know which file is causing the issue, you might try to activate the logging feature, wait for the problem to happen, and then see if the log contains some interesting info. To do that, you'll need to adjust the following in the `configuration.yaml` file:

- `log_file: /path/to/your/logfile.txt` - set it to the name of a file to contain the logging information
- `log_level: DEBUG` 

Once you're done with the logging, you'll want to turn it off again as the file will get large quickly...

Mettbrot, 2024-02-08:
Thank you.
So the file is 280MB large, the last lines before the crash are these

```
2024-02-07 23:45:07,291 - image_cache.ImageCache - DEBUG - Inserting: /mnt/bilder/Wallpapers/Dual/dmb - Dual Monitor Backgrounds - 30 Stück/A Sea Lion In Summer/ASeaLionInSummer.jpg.php.jpg
2024-02-07 23:45:07,341 - PIL.TiffImagePlugin - DEBUG - tag: Make (271) - type: string (2) Tag Location: 22 - Data Location: 122 - value: b'Canon\x00'
2024-02-07 23:45:07,341 - PIL.TiffImagePlugin - DEBUG - tag: Model (272) - type: string (2) Tag Location: 34 - Data Location: 128 - value: b'Canon EOS DIGITAL REBEL\x00'
2024-02-07 23:45:07,342 - PIL.TiffImagePlugin - DEBUG - tag: Orientation (274) - type: short (3) - value: b'\x01\x00'
2024-02-07 23:45:07,342 - PIL.TiffImagePlugin - DEBUG - tag: XResolution (282) - type: rational (5) Tag Location: 58 - Data Location: 152 - value: b'\xb4\x00\x00\x00\x01\x00\x00\x00'
2024-02-07 23:45:07,343 - PIL.TiffImagePlugin - DEBUG - tag: YResolution (283) - type: rational (5) Tag Location: 70 - Data Location: 160 - value: b'\xb4\x00\x00\x00\x01\x00\x00\x00'
2024-02-07 23:45:07,344 - PIL.TiffImagePlugin - DEBUG - tag: ResolutionUnit (296) - type: short (3) - value: b'\x02\x00'
2024-02-07 23:45:07,348 - PIL.TiffImagePlugin - DEBUG - tag: DateTime (306) - type: string (2) Tag Location: 94 - Data Location: 168 - value: b'2006:08:25 19:37:19\x00\x00'
2024-02-07 23:45:07,351 - PIL.TiffImagePlugin - DEBUG - tag: YCbCrPositioning (531) - type: short (3) - value: b'\x01\x00'
2024-02-07 23:45:07,354 - PIL.TiffImagePlugin - DEBUG - tag: ExifIFD (34665) - type: long (4) - value: b'\xbd\x00\x00\x00'
2024-02-07 23:45:07,363 - PIL.TiffImagePlugin - DEBUG - tag: ExposureTime (33434) - type: rational (5) Tag Location: 203 - Data Location: 567 - value: b'\x01\x00\x00\x00\xf4\x01\x00\x00'
2024-02-07 23:45:07,365 - PIL.TiffImagePlugin - DEBUG - tag: FNumber (33437) - type: rational (5) Tag Location: 215 - Data Location: 575 - value: b'8\x00\x00\x00\n\x00\x00\x00'
2024-02-07 23:45:07,366 - PIL.TiffImagePlugin - DEBUG - tag: ISOSpeedRatings (34855) - type: short (3) - value: b'd\x00'
2024-02-07 23:45:07,368 - PIL.TiffImagePlugin - DEBUG - tag: ExifVersion (36864) - type: undefined (7) - value: b'0221'
2024-02-07 23:45:07,369 - PIL.TiffImagePlugin - DEBUG - tag: DateTimeOriginal (36867) - type: string (2) Tag Location: 251 - Data Location: 583 - value: b'2006:08:25 14:00:59\x00\x00'
2024-02-07 23:45:07,370 - PIL.TiffImagePlugin - DEBUG - tag: DateTimeDigitized (36868) - type: string (2) Tag Location: 263 - Data Location: 604 - value: b'2006:08:25 14:00:59\x00\x00'
2024-02-07 23:45:07,371 - PIL.TiffImagePlugin - DEBUG - tag: ComponentsConfiguration (37121) - type: undefined (7) - value: b'\x01\x02\x03\x00'
2024-02-07 23:45:07,372 - PIL.TiffImagePlugin - DEBUG - tag: CompressedBitsPerPixel (37122) - type: rational (5) Tag Location: 287 - Data Location: 625 - value: b'\x03\x00\x00\x00\x01\x00\x00\x00'
2024-02-07 23:45:07,373 - PIL.TiffImagePlugin - DEBUG - tag: ShutterSpeedValue (37377) - type: signed rational (10) Tag Location: 299 - Data Location: 633 - value: b'>\xf7\x08\x00\x00\x00\x01\x00'
2024-02-07 23:45:07,374 - PIL.TiffImagePlugin - DEBUG - tag: ApertureValue (37378) - type: rational (5) Tag Location: 311 - Data Location: 641 - value: b'\x8a\xf8\x04\x00\x00\x00\x01\x00'
2024-02-07 23:45:07,376 - PIL.TiffImagePlugin - DEBUG - tag: ExposureBiasValue (37380) - type: signed rational (10) Tag Location: 323 - Data Location: 649 - value: b'\x03\x00\x00\x00\x03\x00\x00\x00'
2024-02-07 23:45:07,377 - PIL.TiffImagePlugin - DEBUG - tag: MaxApertureValue (37381) - type: rational (5) Tag Location: 335 - Data Location: 657 - value: b'\x8a\xf8\x04\x00\x00\x00\x01\x00'
2024-02-07 23:45:07,378 - PIL.TiffImagePlugin - DEBUG - tag: MeteringMode (37383) - type: short (3) - value: b'\x05\x00'
2024-02-07 23:45:07,379 - PIL.TiffImagePlugin - DEBUG - tag: Flash (37385) - type: short (3) - value: b'\x00\x00'
2024-02-07 23:45:07,379 - PIL.TiffImagePlugin - DEBUG - tag: FocalLength (37386) - type: rational (5) Tag Location: 371 - Data Location: 665 - value: b'P\x00\x00\x00\x01\x00\x00\x00'
2024-02-07 23:45:07,380 - PIL.TiffImagePlugin - DEBUG - tag: MakerNote (37500) - type: undefined (7) Tag Location: 383 - Data Location: 673 - value: <table: 1292 bytes>
2024-02-07 23:45:07,380 - PIL.TiffImagePlugin - DEBUG - tag: UserComment (37510) - type: undefined (7) Tag Location: 395 - Data Location: 1965 - value: <table: 264 bytes>
2024-02-07 23:45:07,381 - PIL.TiffImagePlugin - DEBUG - tag: FlashPixVersion (40960) - type: undefined (7) - value: b'0100'
2024-02-07 23:45:07,381 - PIL.TiffImagePlugin - DEBUG - tag: ColorSpace (40961) - type: short (3) - value: b'\x01\x00'
2024-02-07 23:45:07,382 - PIL.TiffImagePlugin - DEBUG - tag: PixelXDimension (40962) - type: long (4) - value: b'\x00\x08\x00\x00'
2024-02-07 23:45:07,382 - PIL.TiffImagePlugin - DEBUG - tag: PixelYDimension (40963) - type: long (4) - value: b'\x00\x03\x00\x00'
2024-02-07 23:45:07,383 - PIL.TiffImagePlugin - DEBUG - tag: InteroperabilityIFD (40965) - type: long (4) - value: b'\xc5\x08\x00\x00'
2024-02-07 23:45:07,383 - PIL.TiffImagePlugin - DEBUG - tag: FocalPlaneXResolution (41486) - type: rational (5) Tag Location: 467 - Data Location: 2229 - value: b'\x00\xe0.\x00|\x03\x00\x00'
2024-02-07 23:45:07,384 - PIL.TiffImagePlugin - DEBUG - tag: FocalPlaneYResolution (41487) - type: rational (5) Tag Location: 479 - Data Location: 2237 - value: b'\x00@\x1f\x00S\x02\x00\x00'
[... cut ...]
2024-02-07 23:45:08,814 - PIL.TiffImagePlugin - DEBUG - tag: Sharpness (41994) - type: short (3) - value: b'\x02\x00'
2024-02-07 23:45:08,814 - PIL.TiffImagePlugin - DEBUG - tag: SubjectDistanceRange (41996) - type: short (3) - value: b'\x00\x00'
```

I uploaded the offending picture here: https://www.dropbox.com/scl/fi/a5gnhkyfcw31r7e954j5d/CockPheasantUK.jpg.php.jpg?rlkey=0huj5or92ahu6f2ybympcffnu&dl=0

I hope this helps. I removed the Wallpapers from the folder and started again, they dont belong there anyways :)

jgodfrey, 2024-02-08:
Thanks for the info and the link. A few comments / questions:

- Did removing the unexpected file(s) from the folder allow the frame to run normally?
- 280 MB? As a JPG? Yikes. If so, I assume it's a very high resolution and very high quality JPG. If it's very high-res, you're probably best off to create a resized copy for display that's closer to your frames resolution, as it'll put much less stress on the raspberry pi...
- The image I'm able to download from your posted link is only 514 KB. While I'm not in a position to test with it ATM, it's likely that it won't exhibit the same issues as the original file.

Mettbrot, 2024-02-08:
Sorry I was not clear before. The log file is 280Mb. The image file is the one you downloaded. 
I remove the folder and picframe runs since this morning so that seems ok :)

The filtering with tags_filter in my config file does not work though, do I have to provide it as array or something?

jgodfrey, 2024-02-08:
> Sorry I was not clear before. The log file is 280Mb. 

Ah, right. I think I understood that at one point during my thought process, but somehow lost track of that detail... ;)

Good to know that things are working without these images and that the d/l image is good for me to test with. I'll try to take a look soon.

jgodfrey, 2024-02-09:
Update - I was able to install a fresh copy on a Linux laptop and I'm able to see the error now. I'll report back when I find out more.

jgodfrey, 2024-02-09:
OK - so I see the problem. In this image the "ISOSpeedRatings" value (which is expected to be a single integer value) is being returned as a tuple of `(50, 50)`. So, two values. And, that's not a valid format to be inserted by the constructed SQL statement.

Do you know what the ISO speed should be on this image. For example, is it potentially just `50`? I'm not sure why the value is recorded like it is in the image...

jgodfrey, 2024-02-09:
More info...

Examining your image with `exiftool`, it also reports the ISO value as `50, 50`, and doesn't complain about the data being invalid when asked to validate the file. So, that makes me think it might indeed be a legitimate value, but I haven't found any documentation to corroborate that - so, I'm not sure ATM.

Again, it's pretty easy to fix the underlying problem by simply calling this `50`, though I'm not sure if that's the best fix or not.

jgodfrey, 2024-02-09:
I've created a PR to fix this here:

https://github.com/helgeerbe/picframe/pull/371

@helgeerbe - I'm not sure whether this is the most appropriate fix not. Feel free to merge, ignore, or continue the discussion...

jgodfrey, 2024-02-09:
Thinking more about this, an alternative fix here would be to keep this value "as-is", but convert it to a CSV string instead of a tuple. In the case of the sample image, where the ISO value is `50,50`, the above PR will return that as `50`. Instead, we could return it as a CSV string like `"50,50"`. 

Looking at the DB schema, the ISO field is defined as a REAL data type, but that won't prevent the insertion of a string.

@helgeerbe - I'm happy to submit a different PR to convert this case to a CSV string if you'd prefer. I'm not sure which is best here...

jgodfrey, 2024-02-09:
More research and another comment...

Apparently, the 2 values found in this case are know as the "ISO Latitude". They are described in the `ISO 12232` spec which can be seen [here](https://cdn.standards.iteh.ai/samples/73758/7857d2b375984ab3877d2339881471cf/ISO-12232-2019.pdf).

Specific to this issue, the doc mentions this:

```
To prevent confusion, a single value is designated as the ISO speed, with the ISO speed
latitude upper and lower limits indicating the speed range.
```

So, this is apparently that "upper and lower limits" situation. My take is that simply pulling the first value out of the tuple is sufficient for picframe's trivial use of this data.

So, I think the current PR is fine IMO.

helgeerbe, 2024-02-09:
Hi @jgodfrey, Thanks for fixing this. I think also, that using the first value ist fine. Its just used for information purpose.

jgodfrey, 2024-02-10:
Closed the original PR and opened a new one targeting `main`. The previous PR inadvertently targeted `dev`...

https://github.com/helgeerbe/picframe/pull/372

helgeerbe, 2024-02-19:
@jgodfrey  dev was the correct branch ;) I already merged it.

helgeerbe, 2024-02-19:
should be fixed with 2024.02.19

closed 2024-02-19 by helgeerbe