helgeerbe/picframe #382: New Setup can't get it working

loucapo, 2024-03-21:
Hello! I am new to picframe and have not been able to get it to open once i point it to my photos folder (which is stored on a thumb drive). when it tries to run, the screen goes blank for a few seconds then it returns to a command prompt with no errors. Any idea what I could be doing wrong?

I'm running a fresh bullseye install on my Pi 4. LMK if you need any more info. 

Thanks in advance!

paddywwoof, 2024-03-21:
Hi, does it work OK if you put a few images in `/home/pi/Pictures` i.e. can you tell if the problem is related to the USB drive. It might be worth setting the logging to go to a file (in the configuration yaml file) and a lower threshold for message such as DEBUG, to see if any indicators are recorded.

loucapo, 2024-03-22:
perhaps thats my issue? i am not using the pi user, but my own. could that be it?

paddywwoof, 2024-03-22:
It should be fine with whatever user you want, but you need to make sure that all the file paths used in the config yaml file are correct. Did it run OK with the 'no pictures found' default image?

loucapo, 2024-03-25:
yes it did. i havent had a chance to try the local images thing yet but should tonight.

loucapo, 2024-03-26:
having the same issue when pointing to a local folder
```/home/MYUSERNAME/Pictures```

loucapo, 2024-04-03:
ok i am a tool and got it to work with a local folder. once its running as a service, is there a way to stop it? the keyboard commands seem disabled during this.

jgodfrey, 2024-04-03:
> is there a way to stop it?

To stop the service? If so, this should work:

`sudo systemctl stop picframe`

(where `picframe` is the name of your service)

loucapo, 2024-04-03:
yea but i dont have access to the terminal when its running (and i dont think i enabled SSH) 🤦

jgodfrey, 2024-04-03:
Ah, yeah. I just `ssh` in to fiddle with things...

loucapo, 2024-04-04:
there must be a way to stop it from booting without ssh, i would think. i havent found anything as of yet.

jgodfrey, 2024-04-04:
I hardly ever use it, but IIRC, the `http` interface provides a stop button (if you activated `http` in the frame's config). By default, that runs on port `9000`. Also, if you have a keyboard hooked up, I'd guess you could reboot and (quickly) disable the service prior to it starting up...

loucapo, 2024-04-04:
how would u do that last part? i'm not a terminal guy, generally but a UI dev 😬

jgodfrey, 2024-04-04:
After a reboot, as soon as you have keyboard control, enter:

`sudo systemctl disable picframe`

(again, where `picframe` should be the name of your service)

If you're quick, that should disable the service before it has a chance to start.

loucapo, 2024-04-04:
apparently i'm too slow or my pi is too fast 😢

paddywwoof, 2024-04-04:
I use http as I don't have home assistant running. Set this to True in the config file then you can control it with your phone browser with address`192.168.0.123:9000` Obviously changing `...0.123` to match the address of your frame.

closed 2024-04-08 by loucapo