helgeerbe/picframe #383: screen goes blank when MQTT activated

lagerbeer, 2024-03-31:
I have pictureframe running fine without MQTT but when I activate MQTT it makes the screen blank:

Here are my settings in the config file:

mqtt:
  use_mqtt: True
  server: "192.168.0.78"   
  port: 1883
  login: "xxxxxxx"               
  password: "xxxxxxx"         
  tls: ""               
  device_id: "picframe"

M11R05, 2024-04-04:
Hello! I'm struggling with the same issue. As I turn on MQTT the slideshow doesn't start any more and screen stays black (does not turn in standby). I'm also unable to detect the picframe as a device via Home Assistant. An MQTT broker is running on the same RasPi 4 and working. What is the device_url parameter in the configuration.yaml for?

mqtt:
  use_mqtt: False
  server: "localhost"
  port: 1883
  login: ""
  password: ""
  tls: ""
  device_id: "picframe"
  device_url: ""

helgeerbe, 2024-06-05:
Sorry, for the late response. Were you able to solve it?

bobburgers7, 2024-06-12:
I have the same issue as @lagerbeer. I ran a picframe debug log and found the below error in addition to the recommendation of updating picframe. 

```
interface_mqtt.InterfaceMQTT - INFO - creating an instance of InterfaceMQTT
interface_mqtt.InterfaceMQTT - ERROR - MQTT not set up because of: Client.__init__() missing 1 required positional argument: 'callback_api_version'
controller.Controller - ERROR - Can't initialize mqtt. Stopping picframe
```

So I installed picframe-2024.5.31-py3-none-any.whl with the below commands:

```
source venv_picframe/bin/activate
pip3 install --upgrade picframe
```
However, when I try to run picframe now, blank screen continues with or without MQTT enabled.  Logs show:

```
controller.Controller - INFO - creating an instance of Controller
image_cache.ImageCache - DEBUG - No unprocessed files in memory, checking disk
image_cache.ImageCache - DEBUG - Found 0 new files on disk
image_cache.ImageCache - DEBUG - Updating cache
```

and the image_cache lines just continually repeat.

bobburgers7, 2024-06-17:
I 'solved' this by rebuilding my image using these instructions:
https://www.thedigitalpictureframe.com/how-to-build-the-best-raspberry-pi-digital-picture-frame-with-bookworm-and-pi3d-picframe-2024-edition/

previously I was using Wayland rather than X11.  I now can run MQTT w/o issue.

closed 2024-08-05 by helgeerbe