helgeerbe/picframe #385: Error When Trying to Run/Start

mfranzel, 2024-04-18:
Hello,
I was following the directions here: https://www.thedigitalpictureframe.com/how-to-build-a-gorgeous-digital-photo-frame-with-a-raspberry-pi-based-on-2023-bookworm/ using a RPi4

But when I get to the part where "start_picframe.sh" has to run, it always just gives the error:

> "error: XDG_RUNTIME_DIR is invalid or not set in the environment. failed to connect to display"

Do you have any suggestions on how to fix this?

pippin88, 2024-04-26:
I also have not had any success with:
RPI4
Bookworm / Raspberry Pi OS Lite (actually Openhab distribution which is Pi OS Lite with Openhab scripts)
Following that guide and https://github.com/helgeerbe/picframe/wiki/Setup-guide-Bookworm-and-Wayland

When I try start_picframe.sh, I get errors:
`failed to connect to display`
`AssertionError: failed to init sdl2: b'wayland not available'`

My monitor displays the console fine.
I can show pictures with `feh`

mfranzel, 2024-04-26:
Yep, I get that too. I cannot display photos in ANY way. I also tried Orange Pi and that doesn't work either.

pippin88, 2024-04-27:
Progress.

I started again, this time using the Raspberry Pi OS Lite image (not openhabian, which is based on that).

Followed the github guide (https://github.com/helgeerbe/picframe/wiki/Setup-guide-Bookworm-and-Wayland)

I am able to get picframe working if:
SSH into RPI
I enter the virtual environment `source venv_picframe/bin/activate`
then `picframe`
Pictures display and work fine

Auto-starting on boot is not working for me.

(I have named my user openhabian because I plan to manually install openhab later, but this is the Raspberry Pi OS Lite image)

`journalctl -g picframe`
```
Apr 27 14:34:08 openhabian systemd[773]: picframe.service: Main process exited, code=exited, status=255/EXCEPTION
Apr 27 14:34:08 openhabian systemd[773]: picframe.service: Failed with result 'exit-code'.
Apr 27 14:34:09 openhabian systemd[773]: picframe.service: Scheduled restart job, restart counter is at 9.
Apr 27 14:34:09 openhabian systemd[773]: Stopped picframe.service - PictureFrame.
```

`systemctl --user status picframe.service`
```
Apr 27 14:42:34 openhabian wayfire[3418]: EE 27-04-24 14:42:34.665 - [libseat] [seatd/seat.c:61] Could not open tty0 to update VT: Permission denied
Apr 27 14:42:34 openhabian wayfire[3418]: II 27-04-24 14:42:34.665 - [libseat] [seatd/seat.c:170] Added client 0 to seat0
Apr 27 14:42:34 openhabian wayfire[3418]: EE 27-04-24 14:42:34.665 - [libseat] [common/terminal.c:162] Could not open target tty: Permission denied
Apr 27 14:42:34 openhabian wayfire[3418]: EE 27-04-24 14:42:34.665 - [libseat] [seatd/seat.c:72] Could not open terminal for VT 0: Permission denied
Apr 27 14:42:34 openhabian wayfire[3418]: EE 27-04-24 14:42:34.665 - [libseat] [seatd/seat.c:461] Could not open VT for client
Apr 27 14:42:34 openhabian wayfire[3418]: EE 27-04-24 14:42:34.665 - [libseat] [common/terminal.c:162] Could not open target tty: Permission denied
Apr 27 14:42:34 openhabian wayfire[3418]: EE 27-04-24 14:42:34.665 - [libseat] [seatd/seat.c:86] Could not open terminal to clean up VT 0: Permission denied
Apr 27 14:42:34 openhabian wayfire[3417]: II 27-04-24 14:42:34.665 - [libseat] [libseat/libseat.c:73] Seat opened with backend 'builtin'
Apr 27 14:42:34 openhabian wayfire[3417]: II 27-04-24 14:42:34.665 - [backend/session/session.c:109] Successfully loaded libseat session
Apr 27 14:42:34 openhabian wayfire[3417]: II 27-04-24 14:42:34.665 - [backend/backend.c:91] Waiting for a session to become active
```

pippin88, 2024-04-27:
I got mine working.

Not sure exactly how I ended up getting auto-start on boot to work, but have it working now.
I did add `export WAYLAND_DISPLAY=wayland-1` `export XDG_RUNTIME_DIR=/run/user/1000` to `start_picframe.sh`

But in the mean-time, I installed openhab/openhabian - doing so stopped picframe working at all (no picframe output to monitor). Eventually I worked out openhabian does tweaks for headless/no display.

So I needed to change /boot/firmware/config.txt back to:
```
# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

comment out gpu memory (openhabian adds gpu_mem=16)
```

and now I have picframe working, and openhab working

My `start_picframe.sh`
``` 
#!/bin/bash
wlr-randr --output HDMI-A-1 --off
export SDL_VIDEODRIVER=wayland # sdl has to use wayland
source /home/openhabian/venv_picframe/bin/activate # activate phyton virtual env
export WAYLAND_DISPLAY=wayland-1
export XDG_RUNTIME_DIR=/run/user/1000
picframe &  #start picframe
wlr-randr --output HDMI-A-1 --on
```

mfranzel, 2024-04-29:
@pippin88 I added

```
export WAYLAND_DISPLAY=wayland-1
export XDG_RUNTIME_DIR=/run/user/1000
```

And if I type `picframe` I can get it working, but how did you get it to run automatically?

pippin88, 2024-04-29:
Did you change your start_picframe?

See my post above for my version.

mfranzel, 2024-04-30:
> #!/bin/bash
> wlr-randr --output HDMI-A-1 --off
> export SDL_VIDEODRIVER=wayland # sdl has to use wayland
> source /home/openhabian/venv_picframe/bin/activate # activate phyton virtual env
> export [...]

@pippin88 I specifically used this from above that you posted.

mfranzel, 2024-05-07:
Absolutely nothing. I attached it here just in case.
[journalctl.txt](https://github.com/helgeerbe/picframe/files/15238098/journalctl.txt)

helgeerbe, 2024-06-05:
Using bookworm, I recommend actual still to run it under X and not wayland.
https://github.com/helgeerbe/picframe/wiki/Setup-guide-Bookworm-and-X

closed 2026-05-06 by github-actions[bot]