helgeerbe/picframe #386: picframe on Bookworm (current RPI OS) - screen/display on/off control -> possible solution

pippin88, 2024-04-27:
Not a *picframe* issue as such, rather due to major changes in Bookworm and depreciation of `vcgencmd display_power`. However, control of display on/off is a major issue for many using *picframe*.

The `wlr-randr --output HDMI-A-1 --on` (or off) command are used to control display on/off. However, I found when `wlr-randr --output HDMI-A-1 --on` is used when *picframe* is already running, I just got a blank screen - no pictures.
I think I have worked out a solution. 
My solution at present is to shutdown *picframe* when turning the display off, and start *picframe* again when turning the display on.
My photo frame is on for set hours - this solution may not be suitable for those using motion detection etc.

I did modify `./start_picframe.sh` (from https://github.com/helgeerbe/picframe/wiki/Setup-guide-Bookworm-and-Wayland) by adding lines `export WAYLAND_DISPLAY=wayland-1` `export XDG_RUNTIME_DIR=/run/user/1000`. This seemed to get auto-start on boot working correctly for me.

`./start_picframe.sh`
```                                                            
#!/bin/bash
wlr-randr --output HDMI-A-1 --off
export SDL_VIDEODRIVER=wayland # sdl has to use wayland
source /home/openhabian/venv_picframe/bin/activate # activate phyton virtual env
export WAYLAND_DISPLAY=wayland-1
export XDG_RUNTIME_DIR=/run/user/1000
picframe &  #start picframe
wlr-randr --output HDMI-A-1 --on
```
Then `chmod +x ./start_picframe.sh`

`nano ./stop_picframe.sh`
```
#!/bin/bash
export WAYLAND_DISPLAY=wayland-1
export XDG_RUNTIME_DIR=/run/user/1000
/usr/bin/wlr-randr --output HDMI-A-1 --off
pkill -x picframe #kill all instances of picframe, as we need to start picframe again when we turn the display on and would end up with multiple picframe processes running
```
Then `chmod +x ./stop_picframe.sh`

Then `crontab -e`
And something like:
```
# For more information see the manual pages of crontab(5) and cron(8)
#  +---------------- minute (0 - 59)
#  |  +------------- hour (0 - 23)
#  |  |  +---------- day of month (1 - 31)
#  |  |  |  +------- month (1 - 12)
#  |  |  |  |  +---- day of week (0 - 6) (Sunday=0 or 7)
#  |  |  |  |  |
#  *  *  *  *  *  command to be executed
#
# m h  dom mon dow   command

#Turn on display at 6.30am Mon-Fri
30 6 * * 1-5  ./start_picframe.sh

#Turn off display at 9am M-F
0 9 * * 1-5 ./stop_picframe.sh

#Turn on display at 3pm M-F
0 15 * * 1-5 ./start_picframe.sh

#Turn on display at 8am Sun
0 8 * * 0 ./start_picframe.sh

#Turn on display at 8am Sat
0 8 * * 6 ./start_picframe.sh

#Turn off display at 9pm every day
0 21 * * * ./stop_picframe.sh
```

Smarter minds than I can probably work out why using `wlr-randr --output HDMI-A-1 --on` after *picframe* is already running, and `wlr-randr --output HDMI-A-1 --off` has been used, results in a blank screen.

paddywwoof, 2024-04-29:
I had half concluded that stopping picframe then restarting it was the only way round the Wayland problem. One issue is that the image sequence would be lost each time it happened, and I thought of saving a sequence number so it could continue. The issue as it stands is that we use a sqlite randomiser that has no option to seed in order to give a reproducible random sequence so we would need to switch to a python randomised list system. Quite doable, I'll put it on the list ...

pippin88, 2024-05-04:
5 days in -> no issues with display off/on using my above solution.

No idea if it messes with the sequence -> I don't mind what sequence (so long as some degree of change / randomness)

sapnho, 2024-06-05:
Here is the revised long-form article based on Helge's instructions: https://www.thedigitalpictureframe.com/how-to-build-the-best-raspberry-pi-digital-picture-frame-with-bookworm-and-pi3d-picframe-2024-edition/

Notrial, 2024-07-25:
Isn't there a problem with 100% CPU usage in the new solution?

sapnho, 2024-11-03:
I guess this issue can be closed.

closed 2024-11-03 by helgeerbe

ylexus, 2025-02-21:
> Here is the revised long-form article based on Helge's instructions: https://www.thedigitalpictureframe.com/how-to-build-the-best-raspberry-pi-digital-picture-frame-with-bookworm-and-pi3d-picframe-2024-edition/ where screen on/off works like a charm.

Hi @sapnho, I have installed the typical/recommended RaspiOS on RPi 5 using https://www.thedigitalpictureframe.com/install-the-pi3d-pictureframe-software-with-one-click-2025-edition-raspberry-pi-2-3-4-5/. I am getting the same symptoms as described in this issue. Do you happen to know what exactly makes "screen on/off works like a charm" if one was to follow  https://github.com/helgeerbe/picframe/wiki/Setup-guide-Bookworm-and-Wayland ? By working like a charm do you mean normal MQTT on/off commands tuen the display off and on and the pic frame continues as normal? In my case when the display durns on it just shows a greyish screen.

sapnho, 2025-02-22:
Hi @ylexus, can you try this: https://www.thedigitalpictureframe.com/xrandr-the-new-command-to-turn-your-connected-monitor-on-and-off/

What this does it to turn the HMDI output off. It may be that your monitor does not shut down when no HDMI signal is received. Mine does it after like 30 seconds.

ylexus, 2025-02-22:
Hi @sapnho, so this kinda works. The only difference, as I understand, between what picframe does and those commands, is the presence of the `--mode` parameter. And indeed, when I simply invoke `wlr-randr --output HDMI-A-1 --on`, the monitor turns on in 640x480@60Hz mode with the screen being grey, just like when turning in on wth picframe. However, when I do `wlr-randr --output HDMI-A-1 --on --mode 2560x1440` (my display is 2560x1440), then it turns on at 2560x1440@60Hz, but the picframe photos are now scaled down and aligned to the lower (see image below). It's as if the picframe originally used a different, smaller resolution. However, when I restart picframe I can clearly see that the monitor switches to 2560x1440@144Hz, so the only difference is the refresh rate. I tried doing `wlr-randr --output HDMI-A-1 --on --mode 2560x1440@144Hz`, which, according to the man page is correct syntax, however it errors with `unknown mode: 2560x1440@144Hz`. Any ideas here?

[image]