helgeerbe/picframe #387: Low resolution on simple Bookworm Lite Picframe install

Alvb75, 2024-05-28:
I followed the instructions and have Picframe working on a Pi-3B with a HDMI 1920x1080 display.  The Systemctl enable, start, stop, and restart all work as they should.  The photos are shown at the correct size, but the aspect is a tad off, and the resolution is about HALF what 1920 x 1180 should look like.  The text is displayed at about twice the correct size.  I'm rather new the Pi world, and have tried dozens of different configurations, but so far no luck.   Note, I put the SD into a Pi-4 4GB, and it worked exactly the same.

helgeerbe, 2024-06-05:
can you share your configuration.yaml and a logfile (prefereable on debug level).

Alvb75, 2024-06-05:
After my post, I decided to take two steps backwards.  I started over with Buster, and I have it doing exactly what I wanted.  I'm ready to start soldering everything together.  I have the SD from the Bookworm install.  Would you still like to see the configuration.yaml and logfile.  Would I use (journalctl --user-unit picframe.service) to get the logfile?  As I originally said,  I'm rather new the Pi world,  I don't know anything about "using X".