helgeerbe/picframe #388: Filtering via Home Assistant stopped working

sapnho, 2024-05-30:
This one is for the master of Home Assistant, @helgeerbe ! :-)

I have always loved the filter possibilities of Pi3D via Home Assistant. However, some time ago, it stopped working. So, it has no effect if I now enter a keyword or modify the dates. HA is updated all the time, so I guess something broke along the way.

Was there anything that needed to be changed? I still use your old blueprint, which worked wonderfully before it didn't.

sapnho, 2024-05-30:
[image]

sapnho, 2024-05-30:
[image]

helgeerbe, 2024-05-30:
Hi Wolfgang,

I can confirm that this stopped working. I don't know why yet. But I hope I have time this weekend to fix it.

sapnho, 2024-05-30:
Ah, good to know! Thanks!

helgeerbe, 2024-05-31:
Home Assistant changed the way how it handles text sensors. Sensor devices are now limited to numbers. Text sensors have to use now the new text device entity.

I fixed this and created a new blueprint. This works as expected on my own devices. 

Will deploy a new versions of picframe and the home assistant blueprint this weekend.

sapnho, 2024-05-31:
Oh, wow, great. Many thanks!

sapnho, 2024-05-31:
@helgeerbe I just did a fresh install on a Pi 4 with Buster. The max version of picframe I get to is 2023.07.26.post1

During installation, I got the following text:

> Collecting picframe
  Downloading picframe-2024.5.31-py3-none-any.whl.metadata (3.9 kB)
INFO: pip is looking at multiple versions of picframe to determine which version is compatible with other requirements. This could take a while.
  Downloading https://www.piwheels.org/simple/picframe/picframe-2024.2.19-py3-none-any.whl (3.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.2/3.2 MB 3.0 MB/s eta 0:00:00
  Downloading https://www.piwheels.org/simple/picframe/picframe-2023.7.26.post1-py3-none-any.whl (3.2 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 3.2/3.2 MB 2.9 MB/s eta 0:00:00

So, could it be that I can't use your latest commit with Buster?

sapnho, 2024-05-31:
Which OS version are you using?

helgeerbe, 2024-05-31:
I use bullseye. 
````cmd
pi@picframe:~ $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
````

Is there a reason why you try buster?

sapnho, 2024-05-31:
I remember there was one issue with Bullseye back then which kept me using Buster. I'll do a fresh install with Buster and will check. Would you have the Bullseye specific instructions somewhere?

helgeerbe, 2024-05-31:
What happens if you try `pip install picframe==2024.5.31`

sapnho, 2024-05-31:
I am in the middle of installing Bullseye.. :-)

What I would need are the raspi config settings and the manual start command line..

sapnho, 2024-05-31:
I am also having troubles with all the commands for Buster that I documented here: https://www.thedigitalpictureframe.com/how-to-add-crossfading-slide-transitions-to-your-digital-picture-frame-using-pi3d/

Do we have a Bullseye installation wiki somewhere?

helgeerbe, 2024-05-31:
This ist still my fastest approach: https://gist.github.com/thomasedoff/3973846101da1e7496524be075f1c02f
And the best, ist doesn't require to run something with sudo as root.

sapnho, 2024-05-31:
Even this
`python3 -m pip install picframe
doesn't work.`

> /usr/bin/python3: No module named pip

helgeerbe, 2024-05-31:
`sudo apt install python3-pip`

sapnho, 2024-05-31:
Thanks. That solves one thing but `picframe -i .
 ` results in

`Traceback (most recent call last):
  File "/home/pi/.local/bin/picframe", line 5, in <module>
    from picframe.start import main
  File "/home/pi/.local/lib/python3.9/site-packages/picframe/start.py", line 8, in <module>
    from picframe import model, viewer_display, controller, __version__
  File "/home/pi/.local/lib/python3.9/site-packages/picframe/viewer_display.py", line 1, in <module>
    import pi3d
  File "/home/pi/.local/lib/python3.9/site-packages/pi3d/__init__.py", line 10, in <module>
    from pi3d.constants import (USE_PYGAME, PIL_OK,
  File "/home/pi/.local/lib/python3.9/site-packages/pi3d/constants/__init__.py", line 200, in <module>
    PLATFORM, bcm, openegl, opengles = _detect_platform_and_load_libraries()
  File "/home/pi/.local/lib/python3.9/site-packages/pi3d/constants/__init__.py", line 195, in _detect_platform_and_load_libraries
    set_gles_function_args(opengles) #function defined in constants/gl.py
  File "/home/pi/.local/lib/python3.9/site-packages/pi3d/constants/gl.py", line 387, in set_gles_function_args
    gles.glActiveTexture.argtypes = [GLenum] #
`

I guess, that's because I didn't install the right driver in raspi config. But which one do I choose?

sapnho, 2024-05-31:
I'll have to write a completely revised version for Bullseye once I get it to work.

sapnho, 2024-06-01:
> I use bullseye.
> 
> Is there a reason why you try buster?

Yes, the whole batch of problems we saw back then: https://github.com/helgeerbe/picframe/issues/195

I never got around to installing a running version with Bullseye with the same performance as Buster. But it seems you have found a way. How did you get it to work on a Pi 4?

sapnho, 2024-06-01:
> What happens if you try `pip install picframe==2024.5.31`

> pip install picframe==2024.5.31
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting picframe==2024.5.31
  Could not find a version that satisfies the requirement picframe==2024.5.31 (from versions: 0.1.1, 0.1.3, 0.1.5, 0.1.6, 0.1.9, 2021.2.27a1, 2021.2.27a2, 2021.2.27a3, 2021.2.27a4, 2021.2.27a5, 2021.2.27a6, 2021.3.10, 2021.3.20, 2021.4.2, 2021.5.2, 2021.5.5, 2021.6.26, 2021.7.5, 2021.8.30, 2021.9.22, 2021.10.6, 2021.11.4, 2022.1.20, 2022.1.20.post1, 2022.3.24, 2022.4.19, 2022.6.3, 2022.12.4, 2022.12.13)
No matching distribution found for picframe==2024.5.31

sapnho, 2024-06-02:
> I also updated the blueprint https://gist.github.com/helgeerbe/78f66ecf1f5206ce50ef9599892f202a .
> 
> On bullesye everythings works fine for me.

Great, thanks. I can't find your Wiki page where you originally described the integration, and mine is a bit outdated. Would you have the link again, please @helgeerbe ?

helgeerbe, 2024-06-02:
https://github.com/helgeerbe/picframe/wiki/Homeassistant

sapnho, 2024-06-02:
Thanks! I only now discovered the Wiki button on the right, I had always looked left... Everything works fine with Bookworm, will write a revised setup version shortly.

The only thing left is the Home Assistant integration with the filters. I replaced the old blueprint with the new one. However, I was wondering, what else needs to be changed in HA? So much has changed in HA since you first developed the integration and I am not sure if the Wiki page reflects this.

sapnho, 2024-06-02:
Thanks. I am a bit lost with the Home Assistant Update.

I replaced the Blueprint, removed the MQTT integration, which came back after restarting picframe and removed the helpers you specified. Everything works fine except for the text inputs, which don't work. Tags, Location, Dates. When I enter a tag, I can see the message sent via MQTT (in MQTT Explorer) but the image choice is not adapted. I went through the original installation but I am not sure where I missed something.

sapnho, 2024-06-02:
[image]

sapnho, 2024-06-02:
[image]

helgeerbe, 2024-06-02:
can you check in the blueprint the entity ids of your helpers. If you deleted the original ones, they may have a different name.
[image]

sapnho, 2024-06-02:
Hmm, not entirely sure what you mean, but the names are the same everywhere.
[image]
[image]

helgeerbe, 2024-06-02:
can you show me the topic, where the tags input is send from the mqtt explorer?

helgeerbe, 2024-06-02:
if you enter a tag in the web frontend of picframe, does it show up in home assistant?

sapnho, 2024-06-02:
> can you show me the topic, where the tags input is send from the mqtt explorer?

[image]
I entered "Bike" as a tag.

sapnho, 2024-06-02:
> if you enter a tag in the web frontend of picframe, does it show up in home assistant?

Hmm, not using this so far. I have activated it now in configuration.yaml but can't access it with http://192.168.178.67/home/pi/picframe_data/html:9000

But I wonder if that is the correct address?

sapnho, 2024-06-02:
[image]

helgeerbe, 2024-06-03:
http://192.168.178.67:9000

sapnho, 2024-06-03:
> > if you enter a tag in the web frontend of picframe, does it show up in home assistant?

Yes, it does.. but I can't remove it then. I can delete the Tags entry but the number of selected images stays the same.

helgeerbe, 2024-06-03:
What I don't understand that you have a topic cmd under picframe_tags_filter. I would expect the response in picframe/tags_filter

[image]

Can you send me the content of `picframe_input_text.yaml` Your will find it under `blueprints/automation/helgeerbe` starting from the directory where your `configuration.yaml` from home assistant is located.

sapnho, 2024-06-03:
Done. And one more observation: When I delete the tag filter value, and refresh HA, it comes back.

helgeerbe, 2024-06-03:
OK, found it.

I bet you didn't added in your Home Assistant UI not the helper `input_text.picframe_tags_filter` instead you added the mqtt element ` text.picframe_tags_filter'

Can you confirm that?

I wasn't aware that the ui element is created automatically. That might simplify the blueprint in the future.

sapnho, 2024-06-03:
> I wasn't aware that the ui element is created automatically. That might simplify the blueprint in the future.

Yes, they are all created automatically. Most convenient.

> I bet you didn't added in your Home Assistant UI not the helper input_text.picframe_tags_filter instead you added the mqtt element ` text.picframe_tags_filter'

I am not sure what you mean. These are my helpers:
[image]

helgeerbe, 2024-06-03:
[image]
Use the yellow, not the red element

sapnho, 2024-06-03:
And in the Frontend, I installed
[image]

sapnho, 2024-06-03:
But when I add values there, no filtering is applied.

helgeerbe, 2024-06-03:
your mqtt message says you are using `text.picframe_tags_filter`, the tag "Bike" is in the `cmd` element.

your entity card says you are using `input_text.picframe_tags_filter`

Now I'm confused.

Can you ad both elements into your entity card and check the mqtt messages.

sapnho, 2024-06-03:
Oh my, my update seems to have messed everything up! :-))
So, the MQTT integration gives this:
[image]

sapnho, 2024-06-03:
The Text Input card doesn't appear in MQTT Explorer-
I cannot see this entry of yours:
[image]

helgeerbe, 2024-06-03:
looks like the automation didn't run. Have you restarted home assistant aber configuring the blueprint?

sapnho, 2024-06-03:
Yes, I had but just in case, I just did a full restart of the HA system. This is how it looks:
[image]

helgeerbe, 2024-06-03:
Is the automation triggered, when you enter a tag?

Can you show the traces?

[image]

sapnho, 2024-06-03:
Can you tell me exactly what to do so that I can show you this result? Never worked with traces before.

sapnho, 2024-06-03:
The thing is, I don't see any automation called "picframe".

sapnho, 2024-06-03:
And when I do "show automations" under your Blueprint, I don't see any.

sapnho, 2024-06-04:
I guess, it's time to update this great article: https://www.thedigitalpictureframe.com/integrate-raspberry-pi-picture-frame-home-assistant/

Should be a lot shorter because HA creates the front end automatically.

helgeerbe, 2024-06-04:
Just to be sure. Did you create the automation from the blueprint?

[image]

sapnho, 2024-06-04:
Ah, a very good question: No, I hadn't! :.-)

Now, the tags and country filters work fine! 

The date filters reduce the number of selected photos but not the right date range...

sapnho, 2024-06-04:
Can Blueprints also create Helpers? This would make the instructions for users even easier.

helgeerbe, 2024-06-04:
Good to hear, that it was at the end an easy solution.

What do you mean about the right date range? Can you show what you entered and how the mqqt payload looks like? The automation converts the entered text, into a timestamp number.

> Can Blueprints also create Helpers? This would make the instructions for users even easier.

Unfortunately not. For `tags` and `location` we could get rid of the helpers, because the input text field elements are now created by mqtt discovery. Problem would be the date fields. The automation takes a text like dd.mm.yyyy and converts ist into a timestamp number.  I see 2 possible solution. 1. It might be possible to manipulate the text of the date input field before it's published over mqtt. 2. We extend picframe and convert the date text there into the timestamp. Advantage of the 2nd solution, we wouldn't need blueprint and automation anymore. Disadvatange performance depends on a full communication cyle, because every input has to be processed not in the webfrontend but on the picframe server.

sapnho, 2024-06-04:
Date from 01.01.2024 gives
[image]

helgeerbe, 2024-06-04:
Convertion seems to be correct: 
- 01.01.2023 -> 1704063600 -> Mon Jan 01 2024 00:00:00 GMT+0100 (Mitteleuropäische Normalzeit)
- 31.12.2023 -> 1703977200 -> Sun Dec 31 2023 00:00:00 GMT+0100 (Mitteleuropäische Normalzeit)

Problematic could be that images might be included in the start date, but excluded from the end date.

sapnho, 2024-06-04:
Which I do would do where?

helgeerbe, 2024-06-05:
I just updated the blueprint. `date_from`starts at midnight, `date_to` ends at 23:59:59

[12 more replies]