helgeerbe/picframe #389: Error message during Pi3D update

sapnho, 2024-05-31:
In preparation for Helge's Home Assistant update, I wanted to update my older installation.

I currently run picframe version: 2022.06.03. I ran `sudo pip3 install picframe --upgrade` to update and got the following:

> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting picframe
  Using cached https://files.pythonhosted.org/packages/d9/75/f14c375b7e48abe52b5b7c65d3782368b3598a4bcdf9448ea1ad13d97c23/picframe-2024.2.19-py3-none-any.whl
Requirement already satisfied, skipping upgrade: numpy in /usr/lib/python3/dist-packages (from picframe) (1.16.2)
Requirement already satisfied, skipping upgrade: IPTCInfo3 in /usr/local/lib/python3.7/dist-packages (from picframe) (2.1.4)
Collecting Pillow>=10.0.1 (from picframe)
  Could not find a version that satisfies the requirement Pillow>=10.0.1 (from picframe) (from versions: 1.0, 1.1, 1.2, 1.3, 1.4, 1.5, 1.6, 1.7.0, 1.7.1, 1.7.2, 1.7.3, 1.7.4, 1.7.5, 1.7.6, 1.7.7, 1.7.8, 2.0.0, 2.1.0, 2.2.0, 2.2.1, 2.2.2, 2.3.0, 2.3.1, 2.3.2, 2.4.0, 2.5.0, 2.5.1, 2.5.2, 2.5.3, 2.6.0, 2.6.1, 2.6.2, 2.7.0, 2.8.0, 2.8.1, 2.8.2, 2.9.0, 3.0.0, 3.1.0rc1, 3.1.0, 3.1.1, 3.1.2, 3.2.0, 3.3.0, 3.3.1, 3.3.2, 3.3.3, 3.4.0, 3.4.1, 3.4.2, 4.0.0, 4.1.0, 4.1.1, 4.2.0, 4.2.1, 4.3.0, 5.0.0, 5.1.0, 5.2.0, 5.3.0, 5.4.0, 5.4.1, 6.0.0, 6.1.0, 6.2.0, 6.2.1, 6.2.2, 7.0.0, 7.1.0, 7.1.1, 7.1.2, 7.2.0, 8.0.0, 8.0.1, 8.1.0, 8.1.1, 8.1.2, 8.2.0, 8.3.0, 8.3.1, 8.3.2, 8.4.0, 9.0.0, 9.0.1, 9.1.0, 9.1.1, 9.2.0, 9.3.0, 9.4.0, 9.5.0)
No matching distribution found for Pillow>=10.0.1 (from picframe)

sapnho, 2024-05-31:
I should add: OS Buster on a Pi 4.

helgeerbe, 2024-05-31:
Strange. https://pypi.org/project/pillow/ shows that 10.3.0 is the latest version of Pillow. Can you Pillow manually.

helgeerbe, 2024-06-05:
@sapnho can we close this issue?

closed 2024-06-05 by helgeerbe