helgeerbe/picframe #390: Installing picframe on OS Bullseye

sapnho, 2024-06-01:
I am opening this new thread so we don't mix the topics too much. I am trying to see if the problems we had some years ago when moving from Buster to Bullseye can be fixed. Thanks, @helgeerbe and @paddywwoof for your help. I will make a new blog post if everything works fine and the performance is comparable or better than Buster.

On a Pi 4.

So here is what I did:
1) Installed OS Bullseye 64-bit Lite
2) Ran `sudo apt update -y && sudo apt upgrade -y`
3) `sudo raspi-config` and changed D4 Screen blanking to off, GPU Memory to 512MB, an Boot Option to auto-login to console
4) `sudo apt install python3-pip` and then `python3 -m pip install picframe`
resulting in 

> Successfully installed IPTCInfo3-2.1.4 Pillow-10.3.0 PyYAML-6.0.1 defusedxml-0.7.1 ninepatch-0.2.0 paho-mqtt-2.1.0 pi-heif-0.16.0 pi3d-2.51 picframe-2024.5.31 pysdl2-0.9.16

5) `sudo python3 -m pip install -U pip` and `sudo python3 -m pip install pi-heif` and `sudo pip3 install picframe`
resulting in

> Installing collected packages: pysdl2, IPTCInfo3, PyYAML, pi3d, paho-mqtt, ninepatch, defusedxml, picframe
Successfully installed IPTCInfo3-2.1.4 PyYAML-6.0.1 defusedxml-0.7.1 ninepatch-0.2.0 paho-mqtt-2.1.0 pi3d-2.51 picframe-2024.5.31 pysdl2-0.9.16

6) `mkdir DeletedPictures`, reboot, `picframe -i .`

7) Then I wanted to start with `sudo xinit /usr/bin/python3 /home/pi/picframe_data/run_start.py /home/pi/picframe_data/config/configuration.yaml
`
which got me 

> sudo: xinit: command not found

What is missing or needs to be changed?

sapnho, 2024-06-01:
So adding `sudo apt install xinit` after which the start command gives me

> X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
Build Operating System: linux Debian
Current Operating System: Linux pi4bullseye 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=D8:3A:DD:80:D9:99 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=daaf1c0c-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=DE
Build Date: 29 April 2024  03:20:14PM
xorg-server 2:1.20.11-1+rpt3+deb11u13 (https://www.debian.org/support) 
Current version of pixman: 0.40.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Sat Jun  1 09:03:16 2024
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
error: XDG_RUNTIME_DIR not set in the environment.
DRM_IOCTL_MODE_CREATE_DUMB failed: Cannot allocate memory
Failed to create scanout resource
(EE) 
Fatal server error:
(EE) AddScreen/ScreenInit failed for driver 0
(EE) 
(EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
(EE) 
(EE) Server terminated with error (1). Closing log file.
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error

paddywwoof, 2024-06-01:
Hi Wolfgang, it looks like there's something missing from the X setup.  I've used @thomasedoff's setup https://gist.github.com/thomasedoff/3973846101da1e7496524be075f1c02f in the past as it installs the minimum needed X libraries and does away with the `sudo` for running xinit. The instructions in step are obfuscated version of just editing the required files, `sed` does a search and replace and `bash/echo/>>` pipes to the target file.

sapnho, 2024-06-01:
Thanks, Paddy. I looked at Thomas' Setup but he did not use a Pi 4 and following his instructions didn't work for me. Also, I am unclear which OS he actually used. I'll try again but so far no luck.

paddywwoof, 2024-06-01:
I might have used the instructions selectively... can't remember details. You could try just adding the X related sounding dependencies from step 6 and see if that gets the X server running.

Paddy

sapnho, 2024-06-01:
Ok, I added
`sudo apt-get install --no-install-recommends \
  xserver-xorg \
  xserver-xorg-legacy \
  xinit \
  python3-pip \
  libopenjp2-7 \
  libgles-dev \
  libatlas3-base \
  libxrender-dev
 `
and

`sudo sed -i 's/\(^allowed_users=\).*/\1anybody/' /etc/X11/Xwrapper.config
sudo bash -c 'echo "needs_root_rights=yes" >> /etc/X11/Xwrapper.config'`

and

`sed -i 's/\(^  use_glx: \).*/\1True/' /home/pi/picframe_data/config/configuration.yaml
`
But still when I do `xinit /usr/bin/python3 /home/pi/.local/bin/picframe /home/pi/picframe_data/config/configuration.yaml`

I get

> X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
Build Operating System: linux Debian
Current Operating System: Linux pi4bullseye 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=D8:3A:DD:80:D9:99 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=daaf1c0c-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=DE
Build Date: 29 April 2024  03:20:14PM
xorg-server 2:1.20.11-1+rpt3+deb11u13 (https://www.debian.org/support) 
Current version of pixman: 0.40.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Sat Jun  1 15:06:00 2024
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
error: XDG_RUNTIME_DIR not set in the environment.
DRM_IOCTL_MODE_CREATE_DUMB failed: Cannot allocate memory
Failed to create scanout resource
(EE) 
Fatal server error:
(EE) AddScreen/ScreenInit failed for driver 0
(EE) 
(EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
(EE) 
(EE) Server terminated with error (1). Closing log file.
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error

sapnho, 2024-06-01:
Okay, I did a completely fresh install, and lo and behold, it works! The difference from last time is that I used the 32-bit OS version instead of 64-bit.

Now, what I'd like to test is the frame rate to determine if Bullseye is at least as good as Buster.

sapnho, 2024-06-01:
Try to use @paddywwoof [RPI4 Test](https://github.com/paddywwoof/test_rpi4), but they seem outdated. 

However, I made an interesting observation, which I think is a good sign: 

On my Pi3D Buster Pi 4, I have set the transition time to 10 seconds. However, in reality, the fading time is about 22 seconds.
On my Pi3D Bullseye Pi 4, the same 10s setting results in a fading time of around 8 seconds. Fading also looks quite smooth but I haven't used it on the 4K yet.

sapnho, 2024-06-01:
So, I tested it on a 4 K monitor. The fading is a bit slower, I think (it's hard to measure crossfading transitions exactly because you don't know when they start) on a 4 K monitor with Bullseye Pi 4, but only a bit. Setting it to 7 seconds in the config results in about 20 seconds of fading time, which is great.

So, I guess it's time to move to Bullseye! I will test the system overnight, but I am really keen on having the filtering functionality again, which @helgeerbe has kindly updated.

@paddywwoof Is there any recent fps testing routine?

paddywwoof, 2024-06-01:
Hi, A few of the pi3d_demos do FPS monitoring but there are different factor that effect the speed of GPU rendering. The one that is probably nearest to picframe, ironically, is Minimal.py. You need to hack it a bit to get a meaningful result. Something like this
```python
import pi3d
import time
## set the height and width the same as the screen (or leave blank, there is a X setting to make if full screen, something like
## display_config=pi3d.DISPLAY_CONFIG_FULLSCREEN)
display = pi3d.Display.create(display_config=pi3d.DISPLAY_CONFIG_FULLSCREEN)#w=1920, h=1280)
camera = pi3d.Camera(is_3d=False) ## set default camera to orthographic with pixel width and height vals
shader = pi3d.Shader("uv_flat")
## read in different sized images to see how the fps varies
sprite = pi3d.ImageSprite("textures/PATRN.PNG", shader, w=1920.0, h=1280.0)
mykeys = pi3d.Keyboard()
start_tm = time.time()
n_frames = 0
while display.loop_running():
    n_frames += 1
    sprite.draw()
    k = mykeys.read()
    if k == 27:
        mykeys.close()
        display.destroy()
        break
## you might not get the print outputting to terminal when running with xinit so write to ramdisk i.e. /dev/shm/fps.txt say
with open("/dev/shm/fps.txt", "a") as f:
    f.write(f"FPS: {n_frames/(time.time() - start_tm):5.1f}\n") #append fps as new line
```
leave it running for a while to get a reasonable average then ESC to stop. If you're running remotely and ESC doesn't work (can't remember if it does or not) then you might need to change the while loop to a for loop to do 1000 frames or something.

helgeerbe, 2024-06-01:
@sapnho I wrote a setup guide for bookworm using X. So even screen blanking is working.

Don't know what the performance is. But this setup should work also on a pi 3/4/5

https://github.com/helgeerbe/picframe/wiki/Setup-guide-Bookworm-and-X

sapnho, 2024-06-02:
Thanks, @helgeerbe , I just did a fresh install on a Pi 4, and it works. I will try to get to works Paddy's code to compare the performance on the various OS.

I suppose in your instructions, you mean Screen Blanking is to be disabled, not enabled.

And how can I turn on/off the screen using a batch command in Bookworm or a separate Python script? Using `xset -display :0 s off` doesn't work here.

sapnho, 2024-06-02:
Found it!
`DISPLAY=:0.0 xrandr --output HDMI-1 --off`
and
`DISPLAY=:0.0 xrandr --output HDMI-1 --mode 1920x1080 --rate 60`

Didn't that not work before in Bookworm? Nice!

sapnho, 2024-06-02:
@paddywwoof So I did the following:

Using this code:
```
#!/usr/bin/python3
import pi3d
import time
# Create fullscreen display
display = pi3d.Display.create(display_config=pi3d.DISPLAY_CONFIG_FULLSCREEN)
camera = pi3d.Camera(is_3d=False)  # Set default camera to orthographic with pixel width and height values
shader = pi3d.Shader("uv_flat")
# Read in different sized images to see how the FPS varies
sprite = pi3d.ImageSprite("textures/PATRN.PNG", shader, w=1920.0, h=1080.0)
mykeys = pi3d.Keyboard()
start_tm = time.time()
n_frames = 0
# Run the loop 1000 times
while display.loop_running() and n_frames < 1000:
    n_frames += 1
    sprite.draw()
    print(f"Loop count: {n_frames}")
# Calculate and print FPS to the console
fps = n_frames / (time.time() - start_tm)
print(f"FPS: {fps:5.1f}")
# Clean up
mykeys.close()
display.destroy()
```

I get the error message

> > ************************
> X11 needs to be running
> ************************
> Traceback (most recent call last):
>   File "/home/pi/fps_test.py", line 7, in <module>
>     display = pi3d.Display.create(display_config=pi3d.DISPLAY_CONFIG_FULLSCREEN)
>               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/pi/venv_picframe/lib/python3.11/site-packages/pi3d/Display.py", line 591, in create
>     display = Display(tkwin, use_sdl2)
>               ^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/pi/venv_picframe/lib/python3.11/site-packages/pi3d/Display.py", line 152, in __init__
>     self.opengl = DisplayOpenGL()
>                   ^^^^^^^^^^^^^^^
>   File "/home/pi/venv_picframe/lib/python3.11/site-packages/pi3d/util/DisplayOpenGL.py", line 77, in __init__
>     assert False, 'Couldnt open DISPLAY {}'.format(display_name)
> AssertionError: Couldnt open DISPLAY None
> (venv_picframe) pi@pi4bookworm:~ $ python3 fps_test.py
> ************************
> X11 needs to be running
> ************************
> Traceback (most recent call last):
>   File "/home/pi/fps_test.py", line 7, in <module>
>     display = pi3d.Display.create(display_config=pi3d.DISPLAY_CONFIG_FULLSCREEN)
>               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/pi/venv_picframe/lib/python3.11/site-packages/pi3d/Display.py", line 591, in create
>     display = Display(tkwin, use_sdl2)
>               ^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/pi/venv_picframe/lib/python3.11/site-packages/pi3d/Display.py", line 152, in __init__
>     self.opengl = DisplayOpenGL()
>                   ^^^^^^^^^^^^^^^
>   File "/home/pi/venv_picframe/lib/python3.11/site-packages/pi3d/util/DisplayOpenGL.py", line 77, in __init__
>     assert False, 'Couldnt open DISPLAY {}'.format(display_name)
> AssertionError: Couldnt open DISPLAY None

that X11 is not running.

sapnho, 2024-06-02:
> @sapnho I wrote a setup guide for bookworm using X. So even screen blanking is working.
> 
> Don't know what the performance is. But this setup should work also on a pi 3/4/5
> 
> https://github.com/helgeerbe/picframe/wiki/Setup-guide-Bookworm-and-X

Works on a Pi 4 but couldn't get it to work on a Pi 5 yet.

helgeerbe, 2024-06-02:
> I suppose in your instructions, you mean Screen Blanking is to be disabled, not enabled.
> 
> 

No, you have to enable it. Otherwise the monitor power saving (dpms) functionality is disabled and xset will not work.

Run `xset -display :0 dpms force onˋ or `xset -display :0 dpms force offˋ

sapnho, 2024-06-02:
Installed it on a Pi3: Works great. Pi 4 already worked great.
Tried again on a Pi 5 like you described: Everything installs alright but picframe does not come up.

Were you able to try it on a Pi 5?

helgeerbe, 2024-06-03:
> Were you able to try it on a Pi 5?

Sorry I don't own a Pi 5. Do you have some logs?

sapnho, 2024-06-05:
Here is the revised long-form article based on Helge's instructions: https://www.thedigitalpictureframe.com/how-to-build-the-best-raspberry-pi-digital-picture-frame-with-bookworm-and-pi3d-picframe-2024-edition/ where the screen on/off works like a charm.

@helgeerbe One thing: I left the D4 Screenblanking at OFF and it still works fine. If I don't, my screen blanks after a while in Terminal.

sapnho, 2024-06-06:
> > Were you able to try it on a Pi 5?
> 
> Sorry I don't own a Pi 5. Do you have some logs?

Will try again this weekend. Which logs would you like to see?

helgeerbe, 2024-06-06:
set loglevel of picframe to debug and run picframe from the console. You can set up a log directory in the configuration.yaml

sapnho, 2024-06-06:
It looks my Pi 5 is broken. I changed SD cards three times, but I still get random freezes, which I have never seen with any Pi before. Have to get a new one.

sapnho, 2024-06-13:
I got a new Pi 5 today and David Wei sent me his solution to make it work with a Pi 5. I just amended the article with his solution.

sapnho, 2024-06-13:
One thing I noticed is that the Pi 5 gets superhot running Pi3D whereas the Pi 4 stays relatively cool. Does anybody else get that?

sapnho, 2024-06-14:
> set loglevel of picframe to debug and run picframe from the console. You can set up a log directory in the configuration.yaml

How do I start picframe manually these days? `xinit /home/pi/start_picframe.sh` results in a server error.

sapnho, 2024-06-14:
And this is what I changed in the config file:

log_level: "DEBUG"
log_file: "/home/pi/pi5log.txt"

sapnho, 2024-06-14:
This is the CPU load:
[image]

paddywwoof, 2024-06-14:
That does look like a lot of CPU for something that's supposed to use the GPU for all the heavy lifting. I wonder if the pi5 had to use software emulsion for GPU if it's not running Wayland.

I will try to have a look after the weekend.

Paddy

disruptivepatternmaterial, 2024-06-24:
I can chime in there and confirm I cannot get this working on a 5, I reverted back to python -m pip install git+https://github.com/helgeerbe/picframe@sdl2 and everything is fine. Not sure what I am missing out on, but just in case anyone else comes here, the setup listed for that version works fine.

Debian GNU/Linux 12 (bookworm) (GNU/Linux 6.6.31+rpt-rpi-v8 aarch64)
Linux Photo-Frame-1 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64

sapnho, 2024-06-24:
Can you recheck your link, please? It gives me a 404.

sapnho, 2024-11-03:
I think this issue can be closed.

closed 2024-11-03 by helgeerbe