helgeerbe/picframe #394: Matting seems at least strange

TorstenSt, 2024-07-21:
Hi, 

after finally setting up a new Pasi3 with Bookworm I see some issues with the matting.
The texture is visble only in the upper left corner with some colors based on the picture . . the rest of the mat remains without matting.
[image]
any idea ?

cheers
Torsten

sapnho, 2024-07-27:
@jgodfrey Hi Jeff, I just did a fresh install using my default method (https://www.thedigitalpictureframe.com/how-to-build-the-best-raspberry-pi-digital-picture-frame-with-bookworm-and-pi3d-picframe-2024-edition/) and indeed your wonderful matting is broken. 

I guess due to a recent Bookworm OS update. It works perfectly on my earlier Bookworm installation, but no longer on a fresh install.

Could you try a fresh install for yourself? Thanks much!

jgodfrey, 2024-07-28:
Hi Wolfgang,

I've got a spare pi4 here. I can try to do a fresh install using your mentioned article to see if I can reproduce the issue. I'll try to do that sometime "soon'ish".

jgodfrey, 2024-07-31:
Update... I've just completed a fresh rpi4 installation with Bookworm using the article linked by Wolfgang above. That worked flawlessly (kudos!). With that, I am able to reproduce the matting problem reported here, though I haven't started to debug it yet. I'll post back when I know more.

jgodfrey, 2024-08-01:
OK - I think I found the problem, and it's not OS-related. It seems to be related to a `numpy` change. The problem is here:

https://github.com/helgeerbe/picframe/blob/6e9fecec6241880df6d462c4f76a9ffebed12b7a/src/picframe/mat_image.py#L308

Changing that line to...

`return tuple(colors[0].tolist())`

...fixes the problem for me locally. Can one of you verify that fixes the issue on your local system?

Some quick research indicates that in versions of numpy prior to 1.20, when a numpy array is converted to a tuple, the individual elements are converted to native python types (as expected by the consuming code). However, from v1.20 on, the converted data retains the underlying types (`np.uint8` in this case) which apparently breaks the `colorize()` function of PIL (the consumer of this data) which is used to modify the colors in the background mat.

It seems that a new picframe install includes numpy v2.0.1, though I'm not sure when that changed exactly...

@helgeerbe / @paddywwoof - any thoughts?

TorstenSt, 2024-08-01:
just tried it . . . it worked right away . . . 
thanks 
Torsten

jgodfrey, 2024-08-01:
I just tried this change on my older picframe install (with a pre 1.20 version of `numpy`) and it doesn't seem to hurt anything. With that in mind, I'll try to submit a small PR with the update soon.

jgodfrey, 2024-08-01:
PR [here](https://github.com/helgeerbe/picframe/pull/396)...

sapnho, 2024-08-01:
Great, thanks so much, @jgodfrey .

So now @helgeerbe needs to put it in the latest release, and then we are back to all systems working again?

jgodfrey, 2024-08-01:
@sapnho - Correct. @helgeerbe will need to verify / merge the PR and release another build.

jgodfrey, 2024-08-01:
@sapnho - In the meantime, you can edit the source file to fix your local instance.

TorstenSt, 2024-08-02:
again thanks for the nice work and one more question . . . 
the texture file . . . does it have to be only one file with that specific name or is a kind of random pick possible whenever there are .jpg files in the respective folder ?

jgodfrey, 2024-08-02:
Hi @TorstenSt,

Currently, the texture file is just a single image, identified by the name `mat_texture.jpg`. While it certainly could be expanded into something more like you describe, it's not supported currently.

TorstenSt, 2024-08-02:
Hi,

at least it would be a nice feature ....

cheers
Torsten

paddywwoof, 2024-08-02:
@jgodfrey In both picframe and pi3d `pyproject.yaml` files we've left the dependency as just `numpy` though some of the others (ninepatch, Pillow) have `>= version`. It would be possible to bake in version number but I feel it would be better to get an error and fix it. That's because generally speaking later versions fix problems or tighten up on things that were underspecified. I can maybe see instances where it would be annoying for tuple() to automatically change the types of elements in the result.

In later numpy `list(x)` and `x.tolist()` behave differently where the dimension are greater than one. i.e. `tolist()` converts the whole thing whereas `list()` only converts the first axis to a python list and leaves the elements inside as they were.

Looking at the Pillow code for `ImageOps.colorize` I would expect it to behave fine with an array of int even though they specify tuple (or str) in the function definition... Though it's not clear why it can't cope with a tuple of np.uint8 values. Maybe that fails the `typing.cast()` against a `typing.Sequence[int]` that gets done at the start of the function.

Paddy

jgodfrey, 2024-08-02:
Thanks for the additional detail @paddywwoof. I'd agree that we probably don't want/need to lock the version of `numpy`. Regarding the current (numpy-based) K-means clustering algorithm... The intent there is to pick the "most dominant" color. IIRC, the 3 returned colors are ordered such that element 0 is (an approximation of) the most dominant color - which is why it's being explicitly picked. I'd think that randomly picking from the returned collection could give undesired results in some situations...

sapnho, 2024-08-03:
Just tried the manual correction and it works fine. Thanks so much for the quick fix, @jgodfrey.

What's the verdict on `numpy` now? It would be nice to have a fully functioning version up again. :-)

helgeerbe, 2024-08-05:
As the changes from @jgodfrey works fine with older numpy packages. I won't add any version dependencies to numpy. I leave it as it is.

I'll prepare a new release today.

closed 2024-08-16 by helgeerbe