helgeerbe/picframe #395: Updating Picframe to latest version on Pi5

Notrial, 2024-07-25:
Does anyone know the proper procedure to update from few versions back to the latest one through SSH on Pi5?
Will I have to reconfigure everything or does it maintain config files?

I am on MacOS terminal and typing sudo pip3 install picframe --upgrade just gives me this.

[image]

Also the commans like picframe -v do nothing:
[image]

paddywwoof, 2024-07-25:
Hi, did you originally install picframe with a virtual environment. There have been various versions of setting up picframe on the RPi5 and (some?) use `venv`. If you originally used venv then you will need to activate it before using pip. something like
```source /home/pi/venv_picframe/bin/activate```

Notrial, 2024-07-26:
that is it! Thank you so much.

Notrial, 2024-07-26:
@paddywwoof btw. should I backup anything before I run the upgrade command? Will it overwrite the config files?

Cheers!

paddywwoof, 2024-07-26:
Well I would *never* suggest not backing up... however pip should only write over the installed files and I think that has a file `.../config/confguration_example.yaml` that you would have copied to `configuration.yaml`, which should remain unscathed. Just to be on the safe side I would probably copy that latter file to your mac at the very least!

closed 2024-08-05 by helgeerbe

Notrial, 2024-08-29:
I finally got around to do this, but it seems I spoke too soon...
Although picframe -v works after source /home/pi/venv_picframe/bin/activate, unfortunately sudo pip3 install picframe --upgrade gives me the same error as before (This environment is externally managed...).

Do I have to type out this upgrade command in a different way?