helgeerbe/picframe #401: Portrait Mode With Bookworm and Raspberry pi4 ?

michaelpaar, 2024-09-01:
helgerbe,

Is it possible to invoke portrait mode using Bookworm and a pi4?

Regards,
mike

sapnho, 2024-09-13:
I get this question quite often these days. 

The old instruction set doesn't seem to do anymore: https://www.thedigitalpictureframe.com/raspberry-pi-digital-picture-frame-portrait-orientation/

Any thoughts on how to achieve this with Bookworm?

sapnho, 2024-11-09:
Now that we have solved Bookworm + Wayland, does Portrait mode work again?

adammhaile, 2024-11-09:
> Now that we have solved Bookworm + Wayland, does Portrait mode work again?

Sadly no, @sapnho - I just came here to try and find a solution. I followed the directions at the link you posted above, but of course that doesn't work as you noted.
I have yet to find an actual solution. 

I'm running a Pi 4 with the latest and updated Bookworm Lite OS and latest picframe.

paddywwoof, 2024-11-09:
With the latest setup we use `wlr-randr` (for turning screen on and off) which should allow `--mode` and `--transform` arguments. See https://manpages.debian.org/testing/wlr-randr/wlr-randr.1.en.html 

What happens if you run that on the command line while picframe is running?

paddywwoof, 2024-11-10:
@adammhaile I'm away at the moment so can't check up on this properly, but I seem to remember that I put a couple of variables into the `configuration. yaml` file to force the correct dimensions on rotated screens. Try setting `display_w` and `display_h`, see what that does. 

Paddy

adammhaile, 2024-11-11:
@paddywwoof Closer still!
So, if I set:
`display_w=1440`
`display_h=2560`

It does work but only if I run the following command AFTER the `picframe` service has started:
`wlr-randr --output HDMI-A-1 --mode 2560x1440 --transform 270`

It does nothing still when it's in `run_start.py`
I'm not familiar enough with the code to know where/how to make it run that command after the fact...

adammhaile, 2024-11-11:
In the meantime, I added an option to rotate images on load, which should solve the problem at least for my specific needs:

https://github.com/helgeerbe/picframe/pull/414

paddywwoof, 2024-11-12:
Adam, that's a tidy solution to your problem. It might be even easier, and save adding niche code to picframe, to write a five line python script to just rotate images and save them as a one-off job after copying them to the raspberry pi.

I will look into the issue of when wlr-randr should be run to rotate the screen, once I'm home.

Paddy

adammhaile, 2024-11-12:
> write a five line python script to just rotate images and save them as a one-off job after copying them to the raspberry pi

I did consider it... probably what I'll actually end up doing. But I was curious how the code worked and thought maybe it might be generally useful beyond my use case.

paddywwoof, 2024-11-12:
Yes definitely worth putting up anything that solves a problem for you as it could well help someone else down the line.

I'll test it all out when I'm home but I think I would try putting the wlr-randr line at the end of `start_picframe.sh` possibly after a pause to let picframe settle. Probably leave `run_start.py` unchanged.

sapnho, 2024-11-15:
Updated it in the [article](https://www.thedigitalpictureframe.com/raspberry-pi-digital-picture-frame-portrait-orientation/). Thanks, @adammhaile

michaelpaar, 2024-11-28:
All you guys,

sapnho, adammhaile, paddywwoof, helgerbe - YOU ARE GEMS !

I made a digital picture frame from an old computer monitor, and used the instructions for Bookworm and a pi4, and had the crazy thing function, which I thought was amazing.  Then I asked the question about 'portrait' mode.  And lo and behold, you guys figured it out.  I now can do either landscape or portrait.  Amazing!  Thank you!  All of you!

And Happy Thanksgiving and Happy Holidays.

You guys are great!

Thank You,
Mike

blee-blee, 2024-12-26:
Hi,

I previously used the 2021 version of Picframe on a Pi4B board with the portrait setting described in the 2022 publication. In that version, the modifications for the portrait setting worked perfectly.

Recently, I updated Picframe to the December 2024 version on the same Pi4B board. However, despite applying the changes suggested by adammhaile (in the Nov 14 post) to configuration.yaml and start_picframe.sh, the images remain in landscape mode and do not rotate to portrait.

Has anyone verified these modifications on a Pi4B board with this version of Picframe? Are there any additional changes or steps required to enable portrait mode?

Thank you very much for your help!

paddywwoof, 2024-12-26:
Hi, the latest system should work OK but it will need the latest OS with Wayland etc. Did you do that or just `pip upgrade...`

Paddy

blee-blee, 2024-12-26:
Some additional info for reference:
1. SD card was prepared on Dec 25, 2024, loaded with Bookworm Lite (64-bit), on booting the Pi4B displays: raspberrypi 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64.
2. The SD card with Picframe loaded was tested on a 4K monitor (display_w: 2160, display_h: 3840) and a 1080 monitor (display_w: 1080, display_h: 1920). 
3. Observation: both monitors display portrait images, but without needed rotation.  The image display starts at the top-left corner of the monitor. Since the image did not rotate, the image is placed at the left part of the monitor, and only the top portion of the image is displayed.  [Note, without the modifications for portrait display, the portrait image will be scaled and be placed at the center of a landscape positioned monitor]
For the 2021 version of the Picframe, images are rotated and the whole portrait image would be shown on a vertically placed monitor. 

Thanks in advance for the comments.

paddywwoof, 2024-12-26:
Hmm, sounds like the `wlr-randr --output HDMI-A-1 --mode 2560x1440 --transform 270
` in start_picframe.sh isn't working. What happens if you enter that on an SSH terminal?

blee-blee, 2024-12-26:
Hi Paddy, thanks a lot. There was a problem in this line. I confirm the modification for the portrait display works for both 1080p monitor and 4k monitor, either 90-degree and 270-degree rotation, works on Pi4B board.
Again, thank you very much.

paddywwoof, 2024-12-26:
Brilliant. Glad you sorted it out. Hope you continue enjoying the picture frame.

Paddy